linux-nvme/nvme-cli #1905: fabrics.c total: 6 errors, 19 warnings, 1468 lines checked

johnmeneghini, 2023-04-14:
```
jmeneghi:nvme-cli(master) > ~/data/repos/CENTOS9/centos-stream-9/scripts/checkpatch.pl --no-tree fabrics.c
WARNING: Improper SPDX comment style for 'fabrics.c', please use '//' instead
#1: FILE: fabrics.c:1:
+/* SPDX-License-Identifier: GPL-2.0-only */

WARNING: Missing or malformed SPDX-License-Identifier tag in line 1
#1: FILE: fabrics.c:1:
+/* SPDX-License-Identifier: GPL-2.0-only */

ERROR: Macros with complex values should be enclosed in parentheses
#87: FILE: fabrics.c:87:
+#define NVMF_OPTS(c)                                                                   \
+       OPT_STRING("transport",       't', "STR", &transport,   nvmf_tport), \
+       OPT_STRING("nqn",             'n', "STR", &subsysnqn,   nvmf_nqn), \
+       OPT_STRING("traddr",          'a', "STR", &traddr,      nvmf_traddr), \
+       OPT_STRING("trsvcid",         's', "STR", &trsvcid,     nvmf_trsvcid), \
+       OPT_STRING("host-traddr",     'w', "STR", &c.host_traddr,       nvmf_htraddr), \
+       OPT_STRING("host-iface",      'f', "STR", &c.host_iface,        nvmf_hiface), \
+       OPT_STRING("hostnqn",         'q', "STR", &hostnqn,     nvmf_hostnqn), \
+       OPT_STRING("hostid",          'I', "STR", &hostid,      nvmf_hostid), \
+       OPT_STRING("dhchap-secret",   'S', "STR", &hostkey,     nvmf_hostkey), \
+       OPT_INT("nr-io-queues",       'i', &c.nr_io_queues,       nvmf_nr_io_queues),   \
+       OPT_INT("nr-write-queues",    'W', &c.nr_write_queues,    nvmf_nr_write_queues),\
+       OPT_INT("nr-poll-queues",     'P', &c.nr_poll_queues,     nvmf_nr_poll_queues), \
+       OPT_INT("queue-size",         'Q', &c.queue_size,         nvmf_queue_size),     \
+       OPT_INT("keep-alive-tmo",     'k', &c.keep_alive_tmo,     nvmf_keep_alive_tmo), \
+       OPT_INT("reconnect-delay",    'c', &c.reconnect_delay,    nvmf_reconnect_delay),\
+       OPT_INT("ctrl-loss-tmo",      'l', &c.ctrl_loss_tmo,      nvmf_ctrl_loss_tmo),  \
+       OPT_INT("tos",                'T', &c.tos,                nvmf_tos),            \
+       OPT_INT("keyring",              0, &c.keyring,            nvmf_keyring),        \
+       OPT_INT("tls_key",              0, &c.tls_key,            nvmf_tls_key),        \
+       OPT_FLAG("duplicate-connect", 'D', &c.duplicate_connect,  nvmf_dup_connect),    \
+       OPT_FLAG("disable-sqflow",    'd', &c.disable_sqflow,     nvmf_disable_sqflow), \
+       OPT_FLAG("hdr-digest",        'g', &c.hdr_digest,         nvmf_hdr_digest),     \
+       OPT_FLAG("data-digest",       'G', &c.data_digest,        nvmf_data_digest), \
+       OPT_FLAG("tls",                 0, &c.tls,                nvmf_tls)     \
+

WARNING: Missing a blank line after declarations
#269: FILE: fabrics.c:269:
+       struct nvme_id_ctrl id = { 0 };
+       if (nvme_ctrl_identify(c, &id)) {

WARNING: quoted string split across lines
#296: FILE: fabrics.c:296:
+       printf("\nDiscovery Log Number of Records %d, "
+              "Generation counter %"PRIu64"\n",

ERROR: spaces required around that ':' (ctx:VxW)
#306: FILE: fabrics.c:306:
+                       nvmf_adrfam_str(e->adrfam): "");
                                                  ^

WARNING: Symbolic permissions 'S_IRUSR|S_IWUSR' are not preferred. Consider using octal permissions '0600'.
#340: FILE: fabrics.c:340:
+       fd = open(raw, O_CREAT|O_RDWR|O_TRUNC, S_IRUSR|S_IWUSR);

WARNING: Block comments use a trailing */ on a separate line
#450: FILE: fabrics.c:450:
+                                * controller around? */

ERROR: code indent should use tabs where possible
#500: FILE: fabrics.c:500:
+^I^I^I         bool discovery_ctrl)$

WARNING: else is not generally useful after a break or return
#508: FILE: fabrics.c:508:
+                       return stringify(NVME_DISC_IP_PORT);
+               } else {

WARNING: quoted string split across lines
#830: FILE: fabrics.c:830:
+                                       "ctrl device %s found, ignoring "
+                                       "non matching command-line options\n",

WARNING: quoted string split across lines
#837: FILE: fabrics.c:837:
+                                       "ctrl device %s found, ignoring "
+                                       "non discovery controller\n",

ERROR: code indent should use tabs where possible
#884: FILE: fabrics.c:884:
+^I        if (!c) {$

WARNING: return of an errno should typically be negative (ie: return -EINVAL)
#948: FILE: fabrics.c:948:
+               return EINVAL;

WARNING: return of an errno should typically be negative (ie: return -EINVAL)
#953: FILE: fabrics.c:953:
+               return EINVAL;

WARNING: return of an errno should typically be negative (ie: return -EINVAL)
#959: FILE: fabrics.c:959:
+               return EINVAL;

WARNING: return of an errno should typically be negative (ie: return -EINVAL)
#967: FILE: fabrics.c:967:
+                       return EINVAL;

WARNING: return of an errno should typically be negative (ie: return -EINVAL)
#1101: FILE: fabrics.c:1101:
+               return EINVAL;

WARNING: Too many leading tabs - consider code refactoring
#1130: FILE: fabrics.c:1130:
+                                               if (!nvme_disconnect_ctrl(c))

WARNING: return of an errno should typically be negative (ie: return -EINVAL)
#1292: FILE: fabrics.c:1292:
+                       return EINVAL;

ERROR: "foo * bar" should be "foo *bar"
#1341: FILE: fabrics.c:1341:
+static void dim_operation(nvme_ctrl_t c, enum nvmf_dim_tas tas, const char * name)

ERROR: "foo * bar" should be "foo *bar"
#1347: FILE: fabrics.c:1347:
+       const char * t;

WARNING: return of an errno should typically be negative (ie: return -EINVAL)
#1394: FILE: fabrics.c:1394:
+               return EINVAL;

WARNING: return of an errno should typically be negative (ie: return -EINVAL)
#1400: FILE: fabrics.c:1400:
+               return EINVAL;

WARNING: return of an errno should typically be negative (ie: return -EINVAL)
#1410: FILE: fabrics.c:1410:
+               return EINVAL;

total: 6 errors, 19 warnings, 1468 lines checked

NOTE: For some of the reported defects, checkpatch may be able to
      mechanically convert to the typical style using --fix or --fix-inplace.

NOTE: Whitespace errors detected.
      You may wish to use scripts/cleanpatch or scripts/cleanfile

fabrics.c has style problems, please review.

NOTE: If any of the errors are false positives, please report
      them to the maintainer, see CHECKPATCH in MAINTAINERS.
```

ikegami-t, 2023-04-17:
Alrmost the fabrics check patch errors and warnings have been fixed but still remaining the error below.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ ~/centos-stream-9/scripts/checkpatch.pl --no-tree fabrics.c
~/centos-stream-9/scripts/checkpatch.pl --no-tree fabrics.c
ERROR: Macros with complex values should be enclosed in parentheses
#89: FILE: fabrics.c:89:
+#define NVMF_OPTS(c)                                                                   \
+       OPT_STRING("transport",       't', "STR", &transport,     nvmf_tport),          \
+       OPT_STRING("nqn",             'n', "STR", &subsysnqn,     nvmf_nqn),            \
+       OPT_STRING("traddr",          'a', "STR", &traddr,        nvmf_traddr),         \
+       OPT_STRING("trsvcid",         's', "STR", &trsvcid,       nvmf_trsvcid),        \
+       OPT_STRING("host-traddr",     'w', "STR", &c.host_traddr, nvmf_htraddr),        \
+       OPT_STRING("host-iface",      'f', "STR", &c.host_iface,  nvmf_hiface),         \
+       OPT_STRING("hostnqn",         'q', "STR", &hostnqn,       nvmf_hostnqn),        \
+       OPT_STRING("hostid",          'I', "STR", &hostid,        nvmf_hostid),         \
+       OPT_STRING("dhchap-secret",   'S', "STR", &hostkey,       nvmf_hostkey),        \
+       OPT_INT("nr-io-queues",       'i', &c.nr_io_queues,       nvmf_nr_io_queues),   \
+       OPT_INT("nr-write-queues",    'W', &c.nr_write_queues,    nvmf_nr_write_queues),\
+       OPT_INT("nr-poll-queues",     'P', &c.nr_poll_queues,     nvmf_nr_poll_queues), \
+       OPT_INT("queue-size",         'Q', &c.queue_size,         nvmf_queue_size),     \
+       OPT_INT("keep-alive-tmo",     'k', &c.keep_alive_tmo,     nvmf_keep_alive_tmo), \
+       OPT_INT("reconnect-delay",    'c', &c.reconnect_delay,    nvmf_reconnect_delay),\
+       OPT_INT("ctrl-loss-tmo",      'l', &c.ctrl_loss_tmo,      nvmf_ctrl_loss_tmo),  \
+       OPT_INT("tos",                'T', &c.tos,                nvmf_tos),            \
+       OPT_INT("keyring",              0, &c.keyring,            nvmf_keyring),        \
+       OPT_INT("tls_key",              0, &c.tls_key,            nvmf_tls_key),        \
+       OPT_FLAG("duplicate-connect", 'D', &c.duplicate_connect,  nvmf_dup_connect),    \
+       OPT_FLAG("disable-sqflow",    'd', &c.disable_sqflow,     nvmf_disable_sqflow), \
+       OPT_FLAG("hdr-digest",        'g', &c.hdr_digest,         nvmf_hdr_digest),     \
+       OPT_FLAG("data-digest",       'G', &c.data_digest,        nvmf_data_digest),    \
+       OPT_FLAG("tls",                 0, &c.tls,                nvmf_tls)             \
+

total: 1 errors, 0 warnings, 1477 lines checked

NOTE: For some of the reported defects, checkpatch may be able to
      mechanically convert to the typical style using --fix or --fix-inplace.

fabrics.c has style problems, please review.

NOTE: If any of the errors are false positives, please report
      them to the maintainer, see CHECKPATCH in MAINTAINERS.
```

igaw, 2023-04-18:
I see two options here:
  - just ignore `checkpath.pl`'s opinion 
  - refactor the macro approach into functions approach.

I would leave it as it is right now. It doesn't hurt as it is and except `checkpatch.pl` no one really cared so far. But if you want to work on it, please don't let me stop you :)

ikegami-t, 2023-04-18:
Thanks for your advice. Just pushed the PR #1913 to refacor the macro approach.

ikegami-t, 2023-04-28:
The remained the error below mentioned by https://github.com/linux-nvme/nvme-cli/issues/1905#issuecomment-1511333706 is not easy to fix simply as far as tried by the PR #1913.
```sh
ERROR: Macros with complex values should be enclosed in parentheses
```

ikegami-t, 2023-05-19:
Now all the fabric.c errors and warning have been resolved.

igaw, 2023-05-24:
Thanks for you effort. The core code is in good shape at this point, I wonder if we should also look at the plugins eventually. This would be even a bigger effort though.