linux-nvme/nvme-cli #1917: Newly flashed firmware is not showed up in "nvme list" command after controller reset

wxiong38, 2023-04-20:
With previous nvme-cli(v1.0), new flashed firmware is showed up in "nvme list" command after controller reset. But it doesn't work with the current nvme-cli.

Live Firmware Update should work after controller  reset: 

   1. nvme reset /dev/nvme#

From e8deee397e8c965ee7735a83ad50703d7c3eb7ba Mon Sep 17 00:00:00 2001
From: Wen Xiong <wenxiong@linux.ibm.com>
Date: Wed, 19 Apr 2023 22:30:21 -0400
Subject: [PATCH 1/1] nvme-cli: Patch for firmware update

---
 nvme-print.c | 12 ++++++++++--
 1 file changed, 10 insertions(+), 2 deletions(-)

diff --git a/nvme-print.c b/nvme-print.c
index 11746993..87e202dc 100644
--- a/nvme-print.c
+++ b/nvme-print.c
@@ -5142,6 +5142,8 @@ void nvme_show_list_item(nvme_ns_t n)
 {
    char usage[128] = { 0 }, format[128] = { 0 };
    char devname[128] = { 0 }; char genname[128] = { 0 };
+    struct nvme_id_ctrl ctrl;
+    int fd;

    long long lba = nvme_ns_get_lba_size(n);
    double nsze = nvme_ns_get_lba_count(n) * lba;
@@ -5159,10 +5161,16 @@ void nvme_show_list_item(nvme_ns_t n)
    nvme_dev_full_path(n, devname, sizeof(devname));
    nvme_generic_full_path(n, genname, sizeof(genname));

-   printf("%-21s %-21s %-20s %-40s %#-10x %-26s %-16s %-8s\n",
+    fd = open(devname, O_RDONLY);
+    if (fd < 0)
+        fprintf(stderr, "smart log: %s\n", nvme_strerror(errno));
+    nvme_identify_ctrl(fd, &ctrl);
+    close(fd);
+
+   printf("%-21s %-21s %-20s %-40s %#-10x %-26s %-16s %-.*s\n",
        devname, genname, nvme_ns_get_serial(n),
        nvme_ns_get_model(n), nvme_ns_get_nsid(n), usage, format,
-       nvme_ns_get_firmware(n));
+       (int)sizeof(ctrl.fr), ctrl.fr);
 }

 static void nvme_show_simple_list(nvme_root_t r)
--
2.35.3

Above is suggested patch for the fix. Please review it.

Thanks,
Wendy

igaw, 2023-04-21:
Just to rule out any misunderstandings, first the firmware on the target side. Then on th host side
```shell
# nvme reset /dev/nvme$ID
```
is executed and
```
# nvme list
```
doesn't show the new version of the firmware instead you still see the old one?

Currently, the firmware version is discovered by the kernel and exported via sysfs. nvme-cli uses this as input. I would have expected that a `nvme rest` should trigger a rediscover of the device also inside in the kernel, which then should update the version information. Maybe I missing something here, need to check what the `nvme reset` command does and what the kernel does in this scenario.

wxiong38, 2023-04-21:
(1) on host side
(2)#nvme list still see the old firmware level
(3)Saw the issue in all the kernel(all linux distros).

The following steps we use to update firmware since we start to use nvme devices. After nvme-cli redesigned from 1.0 to 2.0, we started to see this issue:

·         Download firmware
·         Activate firmware to all supported slots

nvme fw-log /dev/nvme#

nvme fw-download /dev/nvme# --fw=<fw_image>.img

nvme fw-activate /dev/nvme# -a 0 -s 1  # This step and next two steps copies the downloaded firmware to the specified FW slot

nvme fw-download /dev/nvme# --fw=<fw_image>.img

nvme fw-activate /dev/nvme# -a 0 -s 2

nvme fw-download /dev/nvme# --fw=<fw_image>.img

nvme fw-activate /dev/nvme# -a 0 -s 3

nvme fw-activate /dev/nvme# -a 3 -s 3  # This step activates the firmware in slot #3 immediately a.k.a Live Firmware Update

If the Live Firmware Update step reports reset is required use one of the following methods:

   # nvme reset /dev/nvme#

igaw, 2023-04-21:
Okay, so it's a PCI device and not a fabric one.

igaw, 2023-04-21:
Ah, we don't ship this fix in SLE so far. Thanks for the pointer!

wxiong38, 2023-04-21:
Thanks! Let me try the patch Keith pointed out.

wxiong38, 2023-04-21:
Hi Keith, 

Thanks for the patch!

The patch works. But saw this warning when running: nvme fw-activate /dev/nvme0 -a 3 -s 3

Is this a warning messahe?

[ 1659.908272] nvme0: Identify(0x6), Host Pathing Error (sct 0x3 / sc 0x70)

#dmesg

[ 1659.908272] nvme0: Identify(0x6), Host Pathing Error (sct 0x3 / sc 0x70)
[ 1752.656585] nvme nvme0: resetting controller
[ 1752.774427] nvme nvme0: Shutdown timeout set to 10 seconds
[ 1752.790134] nvme nvme0: 8/0/0 default/read/poll queues

keithbusch, 2023-04-21:
You can safely ignore that warning message.

wxiong38, 2023-04-23:
Thanks for your help!

closed 2023-04-26 by igaw