linux-nvme/nvme-cli #1931: nvme-cli make process is trying to contact github

pyite, 2023-05-05:
I am trying to build on a system with no outbound internet access.  This should be possible with the official release zip or tar.gz files, but it is trying to phone home:
...
Cloning into 'libnvme'...
fatal: unable to access 'https://github.com/linux-nvme/libnvme.git/': OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to github.com:443 

meson.build:48:0: ERROR: Git command failed: ['/usr/bin/git', 'clone', 'https://github.com/linux-nvme/libnvme.git', 'libnvme']

Thanks

martin-belanger, 2023-05-05:
Hey @pyite 

Have you tried to configure meson with the `--wrap-mode` option? Check [README.md](https://github.com/linux-nvme/nvme-cli/blob/master/README.md) for details.

pyite, 2023-05-05:
Not yet, but I'll try when I can.  There are a whole ton of dependencies like openssl, sphinx, snowballstemmer, etc that I have to copy over by hand.

igaw, 2023-05-07:
I recommend not to use the Makefile wrapper, instead use meson directly. The tool knows a lot of options how you can configure your build. I think you want `meson setup --wrap-mode=nodownload .build`. We changed the defaults for the download behavior recently. So the next release should not try to download anymore per default.

closed 2023-05-07 by igaw