linux-nvme/nvme-cli #1934: create a namesapce with  entering just MB or GB capacity

wxiong38, 2023-05-12:
Hi Daniel

Can we pass MB or GB capacity when creating a nvmespace in nvme-cli? If not, do we want to implement one for this?

like: nvme create-ns -n 800G

Thanks,
Wendy

ikegami-t, 2023-05-14:
It can be passed MB or GB capacity by using the '--ncap-si' or '-C' options as below.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme create-ns -h
...
  [  --nsze-si=<STRING>, -S <STRING> ]  --- size of ns (NSZE) in standard SI
                                            units
  [  --ncap-si=<STRING>, -C <STRING> ]  --- capacity of ns (NCAP) in standard
                                            SI units
...
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme create-ns /dev/nvme0 -C 800.0G -f 0
```
But currently the value `800G` is not parsed correctly but parsed as `0`.  (I will do investigate the casue detail to fix it.) To avoid this issue it can be set the value as `800.0G` and correctly set.
Note: The value is devided by the formatted LBA size to set as for example the value `800.0G` is calcurated to `1,562,500,000` if the LBA size is 512 bytes.

igaw, 2023-05-15:
Hi Wendy, as @ikegami-t said, there are already these option available. Just merged the fix for parsing the units correctly. If there is still an issue just open the bug again.

closed 2023-05-15 by igaw