linux-nvme/nvme-cli #1938: When using passthru read cmd, it is hoped that the output works separately from the data and metadata

soojung56, 2023-05-17:
I want the output to work separately from the data and metadata when using the passthru read cmd.
However, the latest version of nvme-cli provides for saving output files only when the "--input-file" and "--property" options exist.

The summary is shown in the table below.
[image]

**[original code]**
```
if (err < 0) {
    nvme_show_error("%s: %s", __func__, nvme_strerror(errno));
} else if (err) {
    nvme_show_status(err);
} else  {
    printf("%s Command %s is Success and result: 0x%08x\n", admin ? "Admin" : "IO",
            strcmp(cmd_name, "Unknown") ? cmd_name : "Vendor Specific", result);
    if (cfg.read && strlen(cfg.input_file)) {
        if (write(dfd, (void *)data, cfg.data_len) < 0)
            perror("failed to write data buffer");
        if (cfg.metadata_len && cfg.metadata) {
            if (write(mfd, (void *)mdata, cfg.metadata_len) < 0)
                perror("failed to write metadata buffer");
        }
    } else if (!cfg.raw_binary) {
        if (data && cfg.read && !err)
            d((unsigned char *)data, cfg.data_len, 16, 1);
    } else if (data && cfg.read) {
        d_raw((unsigned char *)data, cfg.data_len);
    }
}
```

**[proposal code]**
```
if (err < 0) {
    nvme_show_error("%s: %s", __func__, nvme_strerror(errno));
} else if (err) {
    nvme_show_status(err);
} else  {
    printf("%s Command %s is Success and result: 0x%08x\n", admin ? "Admin" : "IO",
            strcmp(cmd_name, "Unknown") ? cmd_name : "Vendor Specific", result);
    if (cfg.read) {
        if (data) {
            if (strlen(cfg.input_file)) {
                if (write(dfd, (void *)data, cfg.data_len) < 0) {
                    perror("failed to write data buffer");
                }
            } else {
                if (!cfg.raw_binary) {
                    d((unsigned char *)data, cfg.data_len, 16, 1);
                } else {
                    d_raw((unsigned char *)data, cfg.data_len);
                }
            }
        }
        if (mdata) {
            if (strlen(cfg.metadata)) {
                if (write(mfd, (void *)mdata, cfg.metadata_len) < 0) {
                    perror("failed to write metadata buffer");
                }
            } else {
                if (!cfg.raw_binary) {
                    d((unsigned char *)mdata, cfg.metadata_len, 16, 1);
                } else {
                    d_raw((unsigned char *)mdata, cfg.metadata_len);
                }
            }
        }
    }
}

```

ikegami-t, 2023-05-17:
To make sure let me confirm below. Is it really metadata is output as stdout on NVMe-CLI ver 2.4 with the passthru cmd option only "input-file"? Since it seems that the original code does not output the metadata for the case. Also have checked it a little on my local environment.

soojung56, 2023-05-18:
Sorry, I checked again metadata no output when I only '--input-file' option. 

And I checked what you reflected. Thanks you.

igaw, 2023-05-19:
I've merged @ikegami-t fix. Closing the report. Thanks!

closed 2023-05-19 by igaw