linux-nvme/nvme-cli #1941: [Question] when I try to use nvme show-regs, report Get-property opertion not permit

wiketool, 2023-05-18:
[image]

wiketool, 2023-05-18:
The Chinese [不允许的操作] means [ Operation not allowed ] in English

keithbusch, 2023-05-18:
I think only fabrics targets support the get property operations. Reading pci registers need to map the BAR, which doesn't work on most kernels anymore.

igaw, 2023-05-18:
show-regs depends on the kernel configuration CONFIG_IO_STRICT_DEVMEM.

See also https://github.com/linux-nvme/nvme-cli/issues/1846

wiketool, 2023-05-19:
> show-regs depends on the kernel configuration CONFIG_IO_STRICT_DEVMEM.
> 
> See also #1846

Thanks you. But after I add iomem=relaxed in /etc/default/grub, I am still unable to use `show-regs`. My linux distribution is Ubuntu20.04

igaw, 2023-05-19:
Do you happen to have selinux or any other LSM running?

igaw, 2023-05-19:
Hmm, so the question is where does the ENOTPERM come from. Could you do a strace and post
the result here?

`sudo strace -o /tmp/nvme.log nvme show-regs /dev/nvme0 -H`

?

wiketool, 2023-05-20:
> Hmm, so the question is where does the ENOTPERM come from. Could you do a strace and post the result here?
> 
> `sudo strace -o /tmp/nvme.log nvme show-regs /dev/nvme0 -H`
> 
> ?

Thanks. This is the log
```
execve("/usr/local/sbin/nvme", ["nvme", "show-regs", "/dev/nvme0", "-H"], 0x7fffa7a78a38 /* 26 vars */) = 0
brk(NULL)                               = 0x561c4282b000
arch_prctl(0x3001 /* ARCH_??? */, 0x7fff2c2c3de0) = -1 EINVAL (Invalid argument)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=62755, ...}) = 0
mmap(NULL, 62755, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f63bedc2000
close(3)                                = 0
openat(AT_FDCWD, "/usr/local/lib/x86_64-linux-gnu/libnvme.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\230\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=389376, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f63bedc0000
mmap(NULL, 190328, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f63bed91000
mmap(0x7f63bed99000, 94208, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8000) = 0x7f63bed99000
mmap(0x7f63bedb0000, 40960, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1f000) = 0x7f63bedb0000
mmap(0x7f63bedba000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x28000) = 0x7f63bedba000
close(3)                                = 0
openat(AT_FDCWD, "/usr/local/lib/x86_64-linux-gnu/libnvme-mi.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\240$\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=102944, ...}) = 0
mmap(NULL, 45456, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f63bed85000
mmap(0x7f63bed87000, 24576, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f63bed87000
mmap(0x7f63bed8d000, 8192, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8000) = 0x7f63bed8d000
mmap(0x7f63bed8f000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9000) = 0x7f63bed8f000
close(3)                                = 0
openat(AT_FDCWD, "/usr/local/lib/x86_64-linux-gnu/libjson-c.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\360N\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=181640, ...}) = 0
mmap(NULL, 96424, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f63bed6d000
mmap(0x7f63bed71000, 57344, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0x7f63bed71000
mmap(0x7f63bed7f000, 16384, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x12000) = 0x7f63bed7f000
mmap(0x7f63bed83000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15000) = 0x7f63bed83000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0 \22\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=18848, ...}) = 0
mmap(NULL, 20752, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f63bed67000
mmap(0x7f63bed68000, 8192, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7f63bed68000
mmap(0x7f63bed6a000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f63bed6a000
mmap(0x7f63bed6b000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f63bed6b000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\220q\0\0\0\0\0\0"..., 832) = 832
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0{E6\364\34\332\245\210\204\10\350-\0106\343="..., 68, 824) = 68
fstat(3, {st_mode=S_IFREG|0755, st_size=157224, ...}) = 0
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0{E6\364\34\332\245\210\204\10\350-\0106\343="..., 68, 824) = 68
mmap(NULL, 140408, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f63bed44000
mmap(0x7f63bed4a000, 69632, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7f63bed4a000
mmap(0x7f63bed5b000, 24576, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x17000) = 0x7f63bed5b000
mmap(0x7f63bed61000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1c000) = 0x7f63bed61000
mmap(0x7f63bed63000, 13432, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f63bed63000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\300A\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\30x\346\264ur\f|Q\226\236i\253-'o"..., 68, 880) = 68
fstat(3, {st_mode=S_IFREG|0755, st_size=2029592, ...}) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\30x\346\264ur\f|Q\226\236i\253-'o"..., 68, 880) = 68
mmap(NULL, 2037344, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f63beb52000
[... cut ...]
openat(AT_FDCWD, "/usr/share/locale/en/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale-langpack/en_US/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale-langpack/en/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
write(2, "get-property: Operation not perm"..., 38) = 38
close(3)                                = 0
close(5)                                = 0
exit_group(1)                           = ?
+++ exited with 1 +++

```

wiketool, 2023-05-24:
When I turn off ` Secure Boot ` options in BIOS settings, it seems works normally

igaw, 2023-05-24:
```
openat(AT_FDCWD, "/sys/class/nvme/nvme0/device/resource0", O_RDONLY) = 4
mmap(NULL, 4096, PROT_READ, MAP_SHARED, 4, 0) = -1 EPERM (Operation not permitted)
```

As expected the `EPERM` is from the mmap call. 

I've checked what the default settings for LSM are in 22.04.2 and it is:

```
# grep CONFIG_DEFAULT_SECURITY /boot/config-5.15.0-72-generic 
# CONFIG_DEFAULT_SECURITY_SELINUX is not set
# CONFIG_DEFAULT_SECURITY_SMACK is not set
# CONFIG_DEFAULT_SECURITY_TOMOYO is not set
CONFIG_DEFAULT_SECURITY_APPARMOR=y
# CONFIG_DEFAULT_SECURITY_DAC is not set
```

I suggest to disable AppAmor and see if you still get EPERM. 

As this is a distribution/kernel configuration issue and not a bug in nvme-cli I am closing this bug report. There is really nothing we can do.

imthenachoman, 2023-10-18:
Can someone please help me understand what I need to do. I'm not understanding any of what is discussed above. I opened #2092 because nvme show-regs is giving invalid argument error. Someone sent me here but I'm not able to figure out what I need to do.

I'm on Debian 12 if it matters.