linux-nvme/nvme-cli #1947: Compilation fails if json-c is not enabled

yhr, 2023-05-22:
If I set up meson with "--force-fallback-for=libnvme .build" i get a build error:

```
[67/93] Compiling C object nvme.p/nvme-print.c.o
FAILED: nvme.p/nvme-print.c.o 
cc -Invme.p -I. -I.. -Iccan -I../ccan -Isubprojects/libnvme/src -I../subprojects/libnvme/src -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -g -fomit-frame-pointer -D_GNU_SOURCE -include config.h -DCCAN_LIST_DEBUG=1 -DCCAN_STR_DEBUG=1 -MD -MQ nvme.p/nvme-print.c.o -MF nvme.p/nvme-print.c.o.d -o nvme.p/nvme-print.c.o -c ../nvme-print.c
In file included from ../nvme-print.c:13:
../nvme-print-json.h:143:33: error: expected ',' or ')', found "char"
  143 | #define json_output_error(const char *msg, va_list ap)
      |                                 ^~~~
```

Worked around this by adding json-c to the force fallback list: (meson setup --force-fallback-for=libnvme,json-c .build)
Maybe json-c should be added to the required dependencies in the README?

ikegami-t, 2023-05-23:
If possible please do double checking the fix PR #1952 changes.

yhr, 2023-05-24:
> If possible please do double checking the fix PR #1952 changes.

Tried it, compilation finishes but linking fails. See #1952 for the details.

PR #1957 (2023-05-25): build: Do not use json-c fallback for muon minimal build
Configure the muon minimal build without using the json-c fallback. This ensures that the case where json-c is not available is also compile tested.

Fixes #1947
closed 2023-05-25 by igaw