linux-nvme/nvme-cli #1950: In the 'handle_plugin' function, why do we use 'strncmp' function to parse input command string?

leowashington, 2023-05-23:
This is a function 'handle_plugin' in the file 'plugin.c' which handles input command string, and it may call the function 'strncmp' to decide which program to execute next.
The function 'strncmp' accepts 3 input parameters: Str1, Str2, length. It is like 'strcmp' but it only compares the given length of chars in the string.
So if I input a command 'nvme id-ct /dev/nvme0n1', it will be the same as 'nvme id-ctrl /dev/nvme0n1'.
I think it just looks really weird. What is the purpose of using 'strncmp' function? Can we be stricter and expect the input command string is fully typed?

ikegami-t, 2023-05-23:
The implementation was introduced by the PR #1081 changes b1f7700 as I think it is useful but is there any problem or concern about this behavior?

leowashington, 2023-05-24:
It's just that I think this kind of behavior is barely seen and used in CLI. It's a subjective matter anyway.
A more important and useful feature would be the tab key completion. However, I've found that in some of my server, 'nvme abc cmd' **cannot** be auto-filled, where 'abc' is my DIY plugin name, while in other servers the 'cmd' **can** be auto-filled... But for 'nvme <cmd>', the command name can always be able to auto-filled, **except for** 'nvme version' and 'nvme help'.
Can you suggest a way of finding the problem that some servers cannot tab auto-fill the plugin command name? I have no idea how it works and it always frustrates me that I can't auto-fill the name.

igaw, 2023-05-24:
Which command do you want to expand in which shell? It just might be that the tab completion file isn't complete or buggy.

igaw, 2023-09-21:
@ikegami-t worked on the zsh/bash completion scripts. I did give it a quick test and the tab completion for help and version on zsh works fine but not for bash.

PR #2058 (2023-09-21): completion: add 'version' and 'help' to list
The variable containing the commands is call _cmds and not cmds, thus the tab completion didn't work for 'version' and 'help'.

Fixes: #1950
closed 2023-09-21 by igaw