linux-nvme/nvme-cli #1967: When using 'nvme read', output form is different from data and metadata.

soojung56, 2023-06-02:
When using 'nvme read', output form is different from data and metadata.
why data output size is 'cfg.data_size'?

cfg.block_count is considered when setting the buffer_size.
but when it is output as the current operation, 'cfg.block_count'  for data is not reflected.

In addtion metadata setting output size is mbuffer_size.

- data
`write(dfd, (void *)buffer, cfg.data_size)`
- metadata
` write(mfd, (void *)mbuffer, mbuffer_size)`

**data/metadata setting code in submit_io, nvme.c +7547**
```
buffer_size = ((long long)cfg.block_count + 1) * logical_block_size;
if (cfg.data_size < buffer_size)
    nvme_show_error("Rounding data size to fit block count (%lld bytes)", buffer_size);
else
    buffer_size = cfg.data_size;

/* Get the required block count. Note this is a zeroes based value. */
nblocks = ((buffer_size + (logical_block_size - 1)) / logical_block_size) - 1;

/* Update the data size based on the required block count */
buffer_size = (nblocks + 1) * logical_block_size;

buffer = nvme_alloc(buffer_size, &huge);

...

mbuffer_size = ((unsigned long long)cfg.block_count + 1) * ms;
if (ms && cfg.metadata_size < mbuffer_size)
    nvme_show_error("Rounding metadata size to fit block count (%lld bytes)",
            mbuffer_size);
else
    mbuffer_size = cfg.metadata_size;

mbuffer = malloc(mbuffer_size);
```

**data/metadata output code in submit_io, nvme.c +7668**
```
if (err < 0) {
	nvme_show_error("submit-io: %s", nvme_strerror(errno));
} else if (err) {
	nvme_show_status(err);
} else {
	if (!(opcode & 1) && write(dfd, (void *)buffer, cfg.data_size) < 0) {
		nvme_show_error("write: %s: failed to write buffer to output file",
			strerror(errno));
		err = -EINVAL;
	} else if (!(opcode & 1) && cfg.metadata_size &&
		   write(mfd, (void *)mbuffer, mbuffer_size) < 0) {
		nvme_show_error(
		    "write: %s: failed to write meta-data buffer to output file",
		    strerror(errno));
		err = -EINVAL;
	} else {
		printf("%s: Success\n", command);
	}
}
```

igaw, 2023-06-02:
Not sure if I understood your question. Is the question about the `args` struct we are preparing and not using `mbuffer_size`? This looks indeed a bit suspicious:

```c
	struct nvme_io_args args = {
		.args_size	= sizeof(args),
		.fd		= dev_fd(dev),
		.nsid		= cfg.namespace_id,
		.slba		= cfg.start_block,
		.nlb		= nblocks,
		.control	= control,
		.dsm		= cfg.dsmgmt,
		.sts		= sts,
		.pif		= pif,
		.dspec		= cfg.dspec,
		.reftag_u64	= cfg.ref_tag,
		.apptag		= cfg.app_tag,
		.appmask	= cfg.app_tag_mask,
		.storage_tag	= cfg.storage_tag,
		.data_len	= buffer_size,
		.data		= buffer,
		.metadata_len	= cfg.metadata_size,   // XXX mbuffer_size?
		.metadata	= mbuffer,
		.timeout	= NVME_DEFAULT_IOCTL_TIMEOUT,
		.result		= NULL,
	};
```

soojung56, 2023-06-02:
Yes, right. The ultimate issue seems to be the difference in handling data and metadata.

In the case of metadata, the 'cfg.metadata_size' argument is received to set mbuffer_size, and the 'cfg.metadata_size' is passed to the structure. After executing the command, the output size is set to mbuffer_size.

In the case of data, the 'cfg.data_size' argument is received to set buffer_size, and the buffer_size is passed to the structure. However, when setting the output size after executing the command, 'cfg.data_size' is used instead of buffer_size.

ikegami-t, 2023-06-04:
The implementation part mentioned was added by the change d10feb0.
```c
static int submit_io(int opcode, char *command, const char *desc, int argc, char **argv)
{
...
	struct nvme_io_args args = {
		.data_len	= buffer_size,
...
		.metadata_len	= cfg.metadata_size,
...
```
Also the implementation below was also changed by the comit 68cb097.
```c
	if (err < 0) {
...
	} else {
		if (!(opcode & 1) && write(dfd, (void *)buffer, cfg.data_size) < 0) {
...
		} else if (!(opcode & 1) && cfg.metadata_size &&
			   write(mfd, (void *)mbuffer, mbuffer_size) < 0) {
...
```
As mentioned it seems that it can be fixed to use same `cfg.datasize` / `cfg.metadata_size` or `buffer_size` / `mbuffer_size` values for data and metadata to pass and output.

soojung56, 2023-06-04:
What I'm trying to say is why you get the buffer size(mbuffer_size) of the metadata and use a different value (cfg.metasize) for the structure.
Also, I wonder why the output for data uses "cfg.data_size" instead of buffer_size for data.

[image]

[image]

Is there anything I'm misunderstanding?
Can you explain to me if there is a specific reason why the code is designed to work this way? If there's anything I don't know, I'd appreciate it if you could let me know about that.

igaw, 2023-06-05:
> Is there anything I'm misunderstanding?

No, don't think so.  The `cfg.metadata_size` argument was added in 18de3a6d61a7 ("Convert to libnvme") and the change is pretty massive so it is likely just an oversight and wasn't spotted during the review. Let's fix this one.

soojung56, 2023-06-05:
Thank you very much for your reply. I'll be waiting for the correction.

ikegami-t, 2023-06-05:
FYI: submit_io related changes commits are below and just fixed pass metadata value by the PR #1970 changes c1abbd7 but still seems the output data and metadata values are not same to use cfg or buffer sizes.
| Commit | Date | Pass | Output | Changes |
--- | --- | --- | --- | ---
| bf5eb4e | Jul 13, 2016 | cfg.block_count | cfg.data_size, cfg.metadata_size | None for pass and output values |
| 2151da2 | Feb 6, 2021 | cfg.block_count | cfg.data_size, cfg.metadata_size | None for pass and output values |
| 68cb097 | Feb 7, 2021 | cfg.block_count | cfg.data_size, ***mbuffer_size*** | Changed output: cfg.metadata_size to mbuffer_size |
| 18de3a6 | Jun 17, 2021 | ***buffer_size***, ***cfg.metadata*** | cfg.data_size, mbuffer_size | Changed pass: cfg.block_count to buffer_size and cfg.metadata_size |
| 0e1edbd | Nov 15, 2021 | buffer_size, cfg.metadata_size | cfg.data_size, mbuffer_size | None for pass and output values |
| 89bddbd | Nov 15, 2021 | buffer_size, cfg.metadata_size | cfg.data_size, mbuffer_size | None for pass and output values |
| b2f2c53 | Nov 18, 2021 | buffer_size, cfg.metadata_size | cfg.data_size, mbuffer_size | None for pass and output values |
| d10feb0 | Jan 11, 2022 | buffer_size, cfg.metadata_size | cfg.data_size, mbuffer_size | None for pass and output values |
| 74c8f79 | Jan 11, 2022 | buffer_size, cfg.metadata_size | cfg.data_size, mbuffer_size | None for pass and output values |
| c1abbd7 | Jun 6, 2023 | buffer_size, ***mbuffer_size*** | cfg.data_size, mbuffer_size | Changed pass: cfg.metadata_size to mbuffer_size |

igaw, 2023-06-05:
Wow! Nice digging!

So if I understand your table correctly, we also need to update the `cfg.data_size` to `buffer_size` in the output path?

soojung56, 2023-06-05:
Right. Regarding 'cfg.data_size' in the output, it hasn't been modified to buffer_size!
I checked that the output of the metadata was modified to mbuffer_size.

soojung56, 2023-06-05:
I think it needs to be modified as shown in the picture below.
[image]

igaw, 2023-06-06:
Makes sense. Let me spin another patch...

closed 2023-06-06 by igaw

soojung56, 2023-06-06:
@igaw Thank you for your reflection.

PR #1970 (2023-06-05): nvme: Use correct metadata buffer size
The submit_io command is using the metadata buffer size provided by the user. But this size argument might be updated and the correct length for the mbuffer is mbuffer_size. Thus use this value and not the initial length.

Fixes: 18de3a6d61a7 ("Convert to libnvme")

Fixes: #1967
PR #1973 (2023-06-06): nvme: Use correct data buffer size
The submit_io command is using the buffer size provided by the user. But this size argument might be updated and the correct length for the buffer is buffer_size. Thus use this value and not the initial length.

Fixes: #1967