linux-nvme/nvme-cli #1974: plugins/micron: Compiler warning on Fedora 38

martin-belanger, 2023-06-06:
I get this warning when I compile nvme-cli on Fedora 38, using gcc (GCC) 13.1.1 20230511 (Red Hat 13.1.1-2):
```
ninja -C .build
ninja: Entering directory `.build'
[48/82] Compiling C object nvme.p/plugins_micron_micron-nvme.c.o
../plugins/micron/micron-nvme.c: In function ‘micron_internal_logs’:
../plugins/micron/micron-nvme.c:3263:33: warning: argument 1 null where non-null expected [-Wnonnull]
 3263 |                                 memset(dataBuffer, 0, bSize);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../plugins/micron/micron-nvme.c:3:
/usr/include/string.h:61:14: note: in a call to function ‘memset’ declared ‘nonnull’
   61 | extern void *memset (void *__s, int __c, size_t __n) __THROW __nonnull ((1));
      |              ^~~~~~
[82/82] Linking target nvme
```

ikegami-t, 2023-06-06:
Sorry for the error. Fixed this by the PR #1975.

ikegami-t, 2023-06-09:
I tried to confirm the build warning with the Fedora and gcc versions mentioned but the warning was not happend. Can you advice any about this?

martin-belanger, 2023-06-09:
Hi @ikegami-t - I just confirmed that the fix https://github.com/linux-nvme/nvme-cli/pull/1975 solves the issue.

ikegami-t, 2023-06-10:
Thanks for the confirmation. I thought to check if any other warning outputs more to fix.

igaw, 2023-06-12:
I've merged #1975.

closed 2023-06-12 by igaw