linux-nvme/nvme-cli #1981: When doing a fw-commit with action 3, the fw does not activate without a reset

anemitz-intc, 2023-06-14:
Trying to udpate an NVMe drive FW using nvme cli. When using "nvme fw-commit /dev/nvme* -s 0 -a 3" where * is the specific nvme controller, the FW activation without reset does not work. The NVMe drives being update do support FW activation without reset.

keithbusch, 2023-06-14:
Could you explain "does not work?" I'm assuming you're either getting an error or not seeing the firmware version change change as expected in the firmware log.

anemitz-intc, 2023-06-14:
The firmware log is showing the new FW version, but when you run "nvme list" after the fw-commit the output does not show the new FW version. In order for "nvme list" output to show the new FW version, have to reset the controller.

keithbusch, 2023-06-14:
I think that's because you're running an older kernel with a big. That problem was fixed in 6.1 (I think stable got it too, but not sure how far back)

anemitz-intc, 2023-06-14:
I am running kernel 6.2 with CentOS 9 streaming.

igaw, 2023-06-16:
Verify that you have the linux commit a8eb6c1ba48b ("nvme: copy firmware_rev on each init") in your kernel

closed 2023-06-16 by igaw

keithbusch, 2023-06-16:
That commit is in 6.2, but this is slightly different than what that fixed. We need driver to refresh the controller properties, but that behavior was removed recently because of some obscure error condition. I think we can fix this, but it'll have to be a driver side change.