linux-nvme/nvme-cli #1982: nvme create-ns doesn't work after delete ns

yizhanglinux, 2023-06-15:
Hello
create-ns seems doesn't work on my nvme ssd, could you help check what's wrong with it?

```
# nvme list 
STDOUT:
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme1n1          /dev/ng1n1            S64FNE0R802879       SAMSUNG MZQL2960HCJR-00A07               0x1        586.32  GB / 960.20  GB    512   B +  0 B   GDC5302Q

# nvme --version
nvme version 2.4 (git 2.4-176-gaed2abe)
libnvme version 1.4 (git 1.4-46-g42ac453)

# nvme delete-ns /dev/nvme1n1
delete-ns: Success, deleted nsid:1

# nvme create-ns /dev/nvme1 -f 0 -C 80G
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)
```

keithbusch, 2023-06-15:
This was closed as completed, but what was the resolution?

igaw, 2023-06-15:
nvme-cli is just reporting that the operation was not successful by the firmware. We need more logs to help here.

@yizhanglinux do you have any additional logs from the kernel? Also `nvme id-ctrl` output would be useful.

keithbusch, 2023-06-15:
I was guessing you need to specify the nsze field in addition to ncap, but not sure (it's been a while since I had a ns-mgmt capable nvme).

igaw, 2023-06-15:
No idea, haven't played with this stuff yet :)

yizhanglinux, 2023-06-16:
Sorry, I forgot to update here: 
I finally created the ns after add --nsze and --ncap, but --ncap-si seems still did not work, add @maurizio-lombardi here, he is looking this issue:

```
# nvme id-ctrl /dev/nvme0 | grep -E "^nn|^cntlid|mcap"
cntlid    : 0x41
tnvmcap   : 1,920,383,410,176
unvmcap   : 1,920,383,352,832
nn        : 32

# nvme create-ns /dev/nvme0 -f 0 -s 101 -c 101
create-ns: Success, created nsid:1

# nvme attach-ns /dev/nvme0 -n 1 -c 0x41
attach-ns: Success, nsid:1

# nvme create-ns /dev/nvme0 -f 0 -C 80G
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)

# nvme create-ns /dev/nvme0 -f 0 -C 80M
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)

```

igaw, 2023-06-16:
Let's reopen this issue in this case.

MatiasBjorling, 2023-06-16:
Having a nice way to specify the size in G, M, K units would be great!

maurizio-lombardi, 2023-06-16:
Looks like the firmware doesn't like the "20G" parameter because of an alignment problem of some sort, if I use "20Gi" instead, it works.

# nvme create-ns /dev/nvme1 -C 20Gi -S 20Gi -f 0 
create-ns: Success, created nsid:1

Note that I patched nvme-cli to accept binary suffixes, it will not work with the vanilla nvme-cli

maurizio-lombardi, 2023-06-16:
This is the patch I used to do the test

http://bsdbackstore.eu/misc/nvme_cli_suffix.patch

igaw, 2023-06-16:
Can't we make nvme-cli clever enough to figure this out itself? I mean, asking the user to figure out to use either `20G` or `20Gi` will cause some troubles...

maurizio-lombardi, 2023-06-30:
First of all, the -c and -s options (nsze and ncap) should be of type "OPT_UINT" and not "OPT_SUFFIX", otherwise you could execute commands like the following which are really confusing from the user's point of view.

`[root@storageqe-40 ~]# nvme create-ns -c 10Mi -s 10Mi -f 0 /dev/nvme1
create-ns: Success, created nsid:1
[root@storageqe-40 ~]# nvme attach-ns /dev/nvme1 --namespace-id=1 --controllers=0x1
attach-ns: Success, nsid:1
[root@storageqe-40 ~]# nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme1n1          /dev/ng1n1            2135312ADFD1         Micron_9300_MTFDHAL3T8TDP                0x1          5.37  GB /   5.37  GB    512   B +  0 B   11300DY0
`

maurizio-lombardi, 2023-06-30:
To fix the original issue, we might replace those two options:

                OPT_STR("nsze-si",       'S', &cfg.nsze_si,  nsze_si),
                OPT_STR("ncap-si",       'C', &cfg.ncap_si,  ncap_si),
with:
                OPT_STR("nsze-bi",       'S', &cfg.nsze_bi,  nsze_si),
                OPT_STR("ncap-bi",       'C', &cfg.ncap_bi,  ncap_si),

And use the binary suffixes instead of the SI ones.

Personally, I don't really like to write "Gi" or "Mi" for "gigabytes" and "megabytes". I would really prefer to just use "G" or "M" like other Linux commands do (fallocate, for example), but I don't see an easy way to do that without creating a mess in nvme-cli

igaw, 2023-06-30:
I agree using `M` & friends is more natural on the command line. When adding the `SI` units we tried to avoid any breaking. But given the feedback I'd say we should consider trying to get it right even if it breaks the existing command line. At least I get the feeling most are not so happy with this interface anyway. Thoughts?

igaw, 2024-04-16:
This has been fixed but I forgot to close this issue. The parser understands 'G' etc.

closed 2024-04-16 by igaw