linux-nvme/nvme-cli #1983: nvme list: missing namespace info?

martin-belanger, 2023-06-16:
I'm not sure whether this is normal behavior or a bug. Hoping someone will be able the shed light.

One of our testers reported that he's not seeing all the namespaces associated with each controller in the output of `nvme list -v`. The output is shown below (I removed extra spaces and sorted the output to make it more readable).

For example, the namespaces `/dev/nvme0n2` to `/dev/nvme0n6` are associated with several controllers `nvme0, nvme1, nvme2, nvme3, nvme4, nvme5, nvme6, nvme7, nvme8, nvme9, nvme10, nvme11`). However, in the `Namespaces` column for each of these controllers we only see `nvme0n1` listed and none of the other namespaces. Is this expected?

While I was removing extra spaces and sorting the output, I noticed that there are 2 subsystems labelled `nvme-subsys0`. That seems odd to me. Thoughts?

Thanks!

```
# nvme list -v
Subsystem     Subsystem-NQN                                           Controllers
------------- ------------------------------------------------------- ----------------
nvme-subsys0  nqn.1988-11.com.dell:powerstore:00:a2ee06317d1845EE5227 nvme54, nvme53, nvme52, nvme51, nvme50, nvme49, nvme48, nvme47, nvme46, nvme45, nvme44, nvme43, nvme42, nvme41, nvme40, nvme39
nvme-subsys0  nqn.1988-11.com.dell:powerstore:00:a2ee06317d1845EE5227 nvme9, nvme8, nvme7, nvme6, nvme5, nvme4, nvme3, nvme2, nvme11, nvme10, nvme1, nvme0
nvme-subsys12 nqn.1988-11.com.dell:SFSS:5:20230412055241iF            nvme30, nvme19, nvme18, nvme14
nvme-subsys13 nqn.1988-11.com.dell:SFSS:2:20230412055241iF            nvme34, nvme33, nvme29, nvme20
nvme-subsys15 nqn.1988-11.com.dell:SFSS:1:20230524091904gP            nvme35, nvme32, nvme22, nvme21
nvme-subsys17 nqn.1988-11.com.dell:SFSS:4:20230412055241iF            nvme36, nvme23, nvme17, nvme16
nvme-subsys25 nqn.1988-11.com.dell:SFSS:3:20230412055241iF            nvme38, nvme31, nvme25, nvme24
nvme-subsys26 nqn.1988-11.com.dell:SFSS:1:20230412055241iF            nvme37, nvme28, nvme27, nvme26

Device SN               MN                 FR      TxP Address        Subsystem    Namespaces
------ ---------------- ------------------ ------- --- -------------- ------------ ----------------
nvme0  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.11,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 nvme0n1
nvme1  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.11,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 nvme0n1
nvme2  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.251.12,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 nvme0n1
nvme3  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.12,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 nvme0n1
nvme4  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.12,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 nvme0n1
nvme5  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.251.11,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 nvme0n1
nvme6  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.11,trsvcid=4420,host_traddr=100.102.190.31 nvme-subsys0 nvme0n1
nvme7  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.11,trsvcid=4420,host_traddr=100.102.190.31 nvme-subsys0 nvme0n1
nvme8  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.251.12,trsvcid=4420,host_traddr=100.102.190.31 nvme-subsys0 nvme0n1
nvme9  6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.12,trsvcid=4420,host_traddr=100.102.190.31 nvme-subsys0 nvme0n1
nvme10 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.12,trsvcid=4420,host_traddr=100.102.190.31 nvme-subsys0 nvme0n1
nvme11 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.251.11,trsvcid=4420,host_traddr=100.102.190.31 nvme-subsys0 nvme0n1
nvme14 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.225,trsvcid=8009,host_iface=em4  nvme-subsys12
nvme16 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.14,trsvcid=8009,host_iface=em4   nvme-subsys17
nvme17 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.14,trsvcid=8009,host_iface=p5p1  nvme-subsys17
nvme18 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.225,trsvcid=8009,host_iface=p5p1 nvme-subsys12
nvme19 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.225,trsvcid=8009,host_iface=p5p2 nvme-subsys12
nvme20 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.59,trsvcid=8009,host_iface=p5p2  nvme-subsys13
nvme21 20230524091904gP DELL-SFSS          (null)  tcp traddr=100.102.188.91,trsvcid=8009,host_iface=p5p2  nvme-subsys15
nvme22 20230524091904gP DELL-SFSS          (null)  tcp traddr=100.102.188.91,trsvcid=8009,host_iface=p5p1  nvme-subsys15
nvme23 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.14,trsvcid=8009,host_iface=p5p2  nvme-subsys17
nvme24 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.16,trsvcid=8009,host_iface=em4   nvme-subsys25
nvme25 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.16,trsvcid=8009,host_iface=p5p2  nvme-subsys25
nvme26 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.50,trsvcid=8009,host_iface=p5p1  nvme-subsys26
nvme27 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.50,trsvcid=8009,host_iface=p5p2  nvme-subsys26
nvme28 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.50,trsvcid=8009,host_iface=em4   nvme-subsys26
nvme29 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.59,trsvcid=8009,host_iface=em3   nvme-subsys13
nvme30 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.225,trsvcid=8009,host_iface=em3  nvme-subsys12
nvme31 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.16,trsvcid=8009,host_iface=p5p1  nvme-subsys25
nvme32 20230524091904gP DELL-SFSS          (null)  tcp traddr=100.102.188.91,trsvcid=8009,host_iface=em4   nvme-subsys15
nvme33 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.59,trsvcid=8009,host_iface=p5p1  nvme-subsys13
nvme34 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.59,trsvcid=8009,host_iface=em4   nvme-subsys13
nvme35 20230524091904gP DELL-SFSS          (null)  tcp traddr=100.102.188.91,trsvcid=8009,host_iface=em3   nvme-subsys15
nvme36 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.14,trsvcid=8009,host_iface=em3   nvme-subsys17
nvme37 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.50,trsvcid=8009,host_iface=em3   nvme-subsys26
nvme38 20230412055241iF DELL-SFSS          (null)  tcp traddr=100.102.188.16,trsvcid=8009,host_iface=em3   nvme-subsys25
nvme39 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.11,trsvcid=4420,host_iface=p5p1  nvme-subsys0
nvme40 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.11,trsvcid=4420,host_iface=p5p2  nvme-subsys0
nvme41 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.11,trsvcid=4420,host_iface=em4   nvme-subsys0
nvme42 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.12,trsvcid=4420,host_iface=em3   nvme-subsys0
nvme43 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.12,trsvcid=4420,host_iface=em3   nvme-subsys0
nvme44 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.12,trsvcid=4420,host_iface=p5p2  nvme-subsys0
nvme45 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.11,trsvcid=4420,host_iface=em3   nvme-subsys0
nvme46 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.11,trsvcid=4420,host_iface=em4   nvme-subsys0
nvme47 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.12,trsvcid=4420,host_iface=p5p1  nvme-subsys0
nvme48 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.11,trsvcid=4420,host_iface=em3   nvme-subsys0
nvme49 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.252.12,trsvcid=4420,host_iface=em4   nvme-subsys0
nvme50 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.11,trsvcid=4420,host_iface=p5p2  nvme-subsys0
nvme51 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.12,trsvcid=4420,host_iface=p5p1  nvme-subsys0
nvme52 6ML4L83          dellemc-powerstore 3.5.0.0 tcp traddr=100.102.253.12,trsvcid=4420,host_iface=p5p2  nvme-subsys0
[... cut ...]
/dev/nvme0n3 /dev/ng0n3 0x92 951.16  GB /   1.40  TB 512   B +  0 B nvme0, nvme1, nvme2, nvme3, nvme4, nvme5, nvme6, nvme7, nvme8, nvme9, nvme10, nvme11
/dev/nvme0n4 /dev/ng0n4 0x94 839.09  GB /   1.10  TB 512   B +  0 B nvme0, nvme1, nvme2, nvme3, nvme4, nvme5, nvme6, nvme7, nvme8, nvme9, nvme10, nvme11
/dev/nvme0n5 /dev/ng0n5 0x95   1.83  TB /   2.20  TB 512   B +  0 B nvme0, nvme1, nvme2, nvme3, nvme4, nvme5, nvme6, nvme7, nvme8, nvme9, nvme10, nvme11
/dev/nvme0n6 /dev/ng0n6 0x9d  33.76  MB /   2.20  TB 512   B +  0 B nvme0, nvme1, nvme2, nvme3, nvme4, nvme5, nvme6, nvme7, nvme8, nvme9, nvme10, nvme11

```

igaw, 2023-06-16:
Looks indeed a bit broken. Could you collect the sys fs tree for this configuration?

martin-belanger, 2023-06-16:
I'll see if the system is still in the same state and collect the data. 

BTW, when you say the sysfs tree, do you mean to actually run the `tree` command as follows:

```
# tree /sys/class/nvme
/sys/class/nvme
└── nvme0 ->
...
...
```

igaw, 2023-06-16:
I think you can collect the data with `tar` so you get the contents and the links.

ikegami-t, 2023-06-20:
From the implementation below seems the nvme_ctrl_for_each_ns() does not work to output the controller NS list but only nvme_ctrl_for_each_path() and nvme_subsystem_for_each_ns() work for it.
```c
static void stdout_detailed_list(nvme_root_t r)
...
	printf("%-8s %-20s %-40s %-8s %-6s %-14s %-12s %-16s\n", "Device",
		"SN", "MN", "FR", "TxPort", "Address", "Subsystem", "Namespaces");
...
	nvme_for_each_host(r, h) {
		nvme_for_each_subsystem(h, s) {
			nvme_subsystem_for_each_ctrl(s, c) {
...
				printf("%-8s %-20s %-40s %-8s %-6s %-14s %-12s ",
				       nvme_ctrl_get_name(c),
...
				       nvme_subsystem_get_name(s));

				nvme_ctrl_for_each_ns(c, n) {
					printf("%s%s", first ? "": ", ",
					       nvme_ns_get_name(n));
...
				nvme_ctrl_for_each_path(c, p) {
					n = nvme_path_get_ns(p);
...
					printf("%s%s", first ? "": ", ",
					       nvme_ns_get_name(n));
...
	printf("%-12s %-12s %-10s %-26s %-16s %-16s\n", "Device", "Generic",
		"NSID", "Usage", "Format", "Controllers");
...
	nvme_for_each_host(r, h) {
		nvme_for_each_subsystem(h, s) {
			nvme_subsystem_for_each_ctrl(s, c) {
				nvme_ctrl_for_each_ns(c, n) {
					stdout_ns_details(n);
					printf("%s\n", nvme_ctrl_get_name(c));
...
			nvme_subsystem_for_each_ns(s, n) {
...
				nvme_subsystem_for_each_ctrl(s, c) {
					printf("%s%s", first ? "" : ", ",
					       nvme_ctrl_get_name(c));
...
```
By the way does 'nvme list' command work but not 'nvme list -v' command since implemented as below?
```c
static void stdout_simple_list(nvme_root_t r)
...

	printf("%-21s %-21s %-20s %-40s %-10s %-26s %-16s %-8s\n",
	    "Node", "Generic", "SN", "Model", "Namespace", "Usage", "Format", "FW Rev");
...
	nvme_for_each_host(r, h) {
		nvme_for_each_subsystem(h, s) {
			nvme_subsystem_for_each_ns(s, n)
				stdout_list_item(n);

			nvme_subsystem_for_each_ctrl(s, c)
				nvme_ctrl_for_each_ns(c, n)
				stdout_list_item(n);
...
```

Prabhuguru21, 2023-06-21:
Here is  nvme list and nvme list -v output from failing system. 

 # nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n2          /dev/ng0n2            6ML4L83              dellemc-powerstore                       0x91       845.47  GB /   1.61  TB    512   B +  0 B   3.5.0.0
/dev/nvme0n1          /dev/ng0n1            6ML4L83              dellemc-powerstore                       0x9b        11.79  GB / 387.62  GB    512   B +  0 B   3.5.0.0

# nvme list -v
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys0     nqn.1988-11.com.dell:powerstore:00:a2ee06317d1845EE5227                                          nvme9, nvme8, nvme7, nvme12, nvme11, nvme10
nvme-subsys0     nqn.1988-11.com.dell:powerstore:00:a2ee06317d1845EE5227                                          nvme5, nvme4, nvme3, nvme2, nvme1, nvme0

Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
**nvme9**    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.251.12,trsvcid=4420,host_traddr=100.102.188.109 nvme-subsys0
**nvme8**    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.252.11,trsvcid=4420,host_traddr=100.102.188.109 nvme-subsys0
**nvme7**    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.253.11,trsvcid=4420,host_traddr=100.102.188.109 nvme-subsys0
**nvme12**   6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.251.11,trsvcid=4420,host_traddr=100.102.188.109 nvme-subsys0
**nvme11**   6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.252.12,trsvcid=4420,host_traddr=100.102.188.109 nvme-subsys0
**nvme10**   6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.253.12,trsvcid=4420,host_traddr=100.102.188.109 nvme-subsys0
nvme5    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.251.11,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 **nvme0n1**
nvme4    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.252.12,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 **nvme0n1**
nvme3    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.253.12,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 **nvme0n1**
nvme2    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.251.12,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 **nvme0n1**
nvme1    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.252.11,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 **nvme0n1**
nvme0    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.253.11,trsvcid=4420,host_traddr=100.102.184.15 nvme-subsys0 **nvme0n1**

Device       Generic      NSID       Usage                      Format           Controllers
------------ ------------ ---------- -------------------------- ---------------- ----------------
/dev/**nvme0n2** /dev/ng0n2   0x91       845.47  GB /   1.61  TB    512   B +  0 B   **nvme5, nvme4, nvme3, nvme2, nvme1, nvme0**
/dev/nvme0n1 /dev/ng0n1   0x9b        11.79  GB / 387.62  GB    512   B +  0 B   nvme5, nvme4, nvme3, nvme2, nvme1, nvme0

For controllers from nvme0 to nvme5 - Namespace column in "nvme list -v" shows only nvme0n1 
None of controllers shows nvme0n2 namespace.

ikegami-t, 2023-06-21:
Thanks for the `nvme list` log information. Both `nvme list` and `nvme list -v` command do not output nvme[1-5]n[1-2] correctly but only output nvme0n[1-2] as `Node` and `Device`. Still seems nvme_ctrl_for_each_ns() does not work correctly. It is based on the sys fs tree information so the sys fs tree information required as mentioned at first.

Prabhuguru21, 2023-06-21:
As suggested by Martin, I ran tar -cvpz -f sysfs.tar.gz --one-file-system /sys to collect sysfs data. TAR ball is of 40MB and Git has attachment limitation to 25MB. Is there any other way I can share this file here ?

ikegami-t, 2023-06-22:
About the file size issue itself can be avoided by using the split command. Also seems the file type should be changed to x**.tar.gz, x**.tgz or something from x**. But failed to attach the x**.tar.gz or x**.tgz files on local environment as far as tried. (I am not sure the failed reason.)
By the way can you attach the tree command output text file instead?

igaw, 2023-06-22:
There are several options how to share larger files on the internet, DropBox, Google Drive, Microsoft OneDrive.... and if this is not option you can upload it to your companies b2b share and send me the download link per email to my company address.

martin-belanger, 2023-06-28:
Sorry for the delay (just got back from vacation). Here's the link to the 40 MB file. You should be able to download it from DropBox (let me know if it doesn't work).

https://www.dropbox.com/s/174ja06tq6rzads/sysfs.tar.gz?dl=0

ikegami-t, 2023-06-28:
Extracted the sys fs tree and tried to reproduce the issue behavior but only the namespace 1 device /dev/nvme0n1 is output and not output other namespace devices.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            6ML4L83              dellemc-powerstore                       0x1          0.00   B / 960.20  GB      4 KiB +  0 B   3.5.0.0
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme list -v
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys8     nqn.1988-11.com.dell:SFSS:1:20230524091904gP                                                     nvme8
nvme-subsys7     nqn.1988-11.com.dell:SFSS:1:20230412055241iF                                                     nvme7
nvme-subsys6     nqn.1988-11.com.dell:SFSS:2:20230412055241iF                                                     nvme6
nvme-subsys0     nqn.1988-11.com.dell:powerstore:00:a2ee06317d1845EE5227                                          nvme9, nvme12, nvme11, nvme10
nvme-subsys0     nqn.1988-11.com.dell:powerstore:00:a2ee06317d1845EE5227                                          nvme5, nvme4, nvme3, nvme2, nvme1, nvme0

Device   SN                   MN                                       FR       TxPort Address        Slot   Subsystem    Namespaces
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------ ------------ ----------------
nvme8    20230524091904gP     DELL-SFSS                                (null)   tcp    traddr=100.102.188.91,trsvcid=8009,host_iface=em3        nvme-subsys8
nvme7    20230412055241iF     DELL-SFSS                                (null)   tcp    traddr=100.102.188.50,trsvcid=8009,host_iface=em3        nvme-subsys7
nvme6    20230412055241iF     DELL-SFSS                                (null)   tcp    traddr=100.102.188.59,trsvcid=8009,host_iface=em3        nvme-subsys6
nvme9    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.252.11,trsvcid=4420,host_iface=em3        nvme-subsys0
nvme12   6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.252.12,trsvcid=4420,host_iface=em3        nvme-subsys0
nvme11   6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.253.11,trsvcid=4420,host_iface=em3        nvme-subsys0
nvme10   6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.253.12,trsvcid=4420,host_iface=em3        nvme-subsys0
nvme5    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.251.11,trsvcid=4420,host_traddr=100.102.184.15        nvme-subsys0 nvme0n1
nvme4    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.252.12,trsvcid=4420,host_traddr=100.102.184.15        nvme-subsys0 nvme0n1
nvme3    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.253.12,trsvcid=4420,host_traddr=100.102.184.15        nvme-subsys0 nvme0n1
nvme2    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.251.12,trsvcid=4420,host_traddr=100.102.184.15        nvme-subsys0 nvme0n1
nvme1    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.252.11,trsvcid=4420,host_traddr=100.102.184.15        nvme-subsys0 nvme0n1
nvme0    6ML4L83              dellemc-powerstore                       3.5.0.0  tcp    traddr=100.102.253.11,trsvcid=4420,host_traddr=100.102.184.15        nvme-subsys0 nvme0n1

Device       Generic      NSID       Usage                      Format           Controllers
------------ ------------ ---------- -------------------------- ---------------- ----------------
/dev/nvme0n1 /dev/ng0n1   0x1          0.00   B / 960.20  GB      4 KiB +  0 B   nvme5, nvme4, nvme3, nvme2, nvme1, nvme0
```

calebsander, 2023-07-07:
This looks like expected output if NVMe multipath is being used:
```
$ cat sys/module/nvme_core/parameters/multipath
Y
```
The kernel combines the paths to the same namespaces across all controllers into a single block device ("0" refers to subsystem 0, not controller 0).

As far as the same subsystem being listed multiple times, I think that's expected if you have multiple hosts (i.e. different host NQNs/IDs) connected to the same subsystem NQN. The kernel will only create one `nvme-subsys*` entry for this subsystem, but the `nvme-cli` in-memory tree representation will have a separate subsystem entry under each host.

igaw, 2023-07-11:
TBH, I am still struggling to really understand the situation. Though one thing which looks really odd is that not all controller directories are populated with a 'channel' controller node. I don't think this is right.

```shell
$ find sys/devices/virtual/nvme-fabrics/ctl -name nvme\*  
sys/devices/virtual/nvme-fabrics/ctl/nvme7
sys/devices/virtual/nvme-fabrics/ctl/nvme5
sys/devices/virtual/nvme-fabrics/ctl/nvme5/nvme0c5n1
sys/devices/virtual/nvme-fabrics/ctl/nvme11
sys/devices/virtual/nvme-fabrics/ctl/nvme11/nvme0c11n2
sys/devices/virtual/nvme-fabrics/ctl/nvme3
sys/devices/virtual/nvme-fabrics/ctl/nvme3/nvme0c3n1
sys/devices/virtual/nvme-fabrics/ctl/nvme1
sys/devices/virtual/nvme-fabrics/ctl/nvme1/nvme0c1n1
sys/devices/virtual/nvme-fabrics/ctl/nvme8
sys/devices/virtual/nvme-fabrics/ctl/nvme6
sys/devices/virtual/nvme-fabrics/ctl/nvme12
sys/devices/virtual/nvme-fabrics/ctl/nvme12/nvme0c12n2
sys/devices/virtual/nvme-fabrics/ctl/nvme4
sys/devices/virtual/nvme-fabrics/ctl/nvme4/nvme0c4n1
sys/devices/virtual/nvme-fabrics/ctl/nvme10
sys/devices/virtual/nvme-fabrics/ctl/nvme10/nvme0c10n2
sys/devices/virtual/nvme-fabrics/ctl/nvme2
sys/devices/virtual/nvme-fabrics/ctl/nvme2/nvme0c2n1
sys/devices/virtual/nvme-fabrics/ctl/nvme0
sys/devices/virtual/nvme-fabrics/ctl/nvme0/nvme0c0n1
sys/devices/virtual/nvme-fabrics/ctl/nvme9
sys/devices/virtual/nvme-fabrics/ctl/nvme9/nvme0c9n2
```

This means nvme-cli is showing what sysfs exposes and it doesn't look like the issue is in nvme-cli.

martin-belanger, 2023-07-11:
@igaw  - Thanks for looking into this.

Just a bit more info:
I'm trying to find out if multipath was used during this test (I will let you know once I hear back). One thing that I know is that the system creates boot devices and later (once booted) it creates more devices. I asked the testers to confirm whether the same Host NQN/ID were used during boot and after boot. I suspect this has something to do with booting using NBFT configuration.

igaw, 2023-07-11:
I might be confused as the other info I got via the SUSE bugzilla also lists pcie devices which are missing here. As I said, still trying to figure out what's happening.

igaw, 2023-07-17:
So the critical missing info was that two different hostnqn are used.

igaw, 2023-07-20:
I've merged the fixes. Should work now.

closed 2023-07-20 by igaw