linux-nvme/nvme-cli #1991: nvme list: can not display correct value after changing the ssd's

jinhuahuang2021, 2023-06-29:
I am using nvme-cli 2.4 version on CentOS 9.

Here's my steps:
1. "nvme list" shows the correct SN.
2. use another tool nvmemgr and change ssd's SN.
3. "nvme list" again but display the old SN value, but "nvme id-ctrl" shows the correct SN value.
4. nvmemgr shows the correct SN value.
[image]

nvme 1.1x on CentOS 8 "nvme list" works well as nvmemgr.

Comparing with nvme-cli 1.x source code:
Seems 1.x version using the different method to list items:
[image]

jinhuahuang2021, 2023-06-29:
We need change the "nvme list" to get the id-ctrl value synchronized

igaw, 2023-06-29:
Can you please verify that you have the linux commit a8eb6c1ba48b ("nvme: copy firmware_rev on each init") in your kernel?

jinhuahuang2021, 2023-06-30:
Linux 5.14.0-307.el9.x86_64 x86_64 on CentOS9 has not include this commit:
[image]

It's a very newer commit, so "nvme list" can not work for such scenario?

keithbusch, 2023-06-30:
The problem is the 'list' implementation is reading sysfs attributes and the kernel isn't refreshing them automatically. We may have to dispatch identify commands instead of relying on these attributes if the kernel can't refresh automatically. I think I can fix the kernel though, I'll send something tomorrow.

jinhuahuang2021, 2023-06-30:
Thank you for kindly response.

igaw, 2023-07-20:
@keithbusch Did you send something out? I didn't see anything related on the mailing list, or I just missed it :)

Nitheesh1609, 2023-11-02:
@igaw @keithbusch, Is there any update on this issue? Will this be resolved by cli side or by kernel refresh?

igaw, 2023-11-03:
Yep, there is an update.  There is a kernel fix for this:

https://lore.kernel.org/linux-nvme/20231030160044.20355-1-dwagner@suse.de/

The fix is on the way to mainline.

igaw, 2023-11-17:
The fix has been merged into the kernel:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=983a338b96c8a25b81e773b643f80634358e81bc

v6.7 will ship this fix.

closed 2023-11-17 by igaw