linux-nvme/nvme-cli #1992: nvme discover --device with invalid device returns log pages

martin-belanger, 2023-06-29:
When using `nvme discover --device [device-id]` with an invalid `[device-id]` the command properly prints that the device does not exist, but still (somehow) retrieves and prints discovery log pages. 

Here's an example where I told nvme-discover to retrieve discovery log pages from `nvme68`, which does not exist on the system. The command properly prints `ctrl device nvme68 not found`, but then proceeds to print log pages anyway. Where it gets the discovery log pages from is a mystery.

```
# nvme discover -d nvme68
ctrl device nvme68 not found

Discovery Log Number of Records 17, Generation counter 2
=====Discovery Log Entry 0======
trtype:  tcp
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  64
trsvcid: 4420
subnqn:  nqn.1988-11.com.dell:powerstore:00:fd9777fd65a990B242B6
traddr:  172.16.10.54
eflags:  none
sectype: none
=====Discovery Log Entry 1======
...
...
...
```

igaw, 2023-06-30:
When the device is not found, nvme-cli falls back trying to use a discover controller by first trying to lookup a discovery controller from the sysfs (fabrics.c:858). I suppose it finds one.

martin-belanger, 2023-07-06:
I found that the option "`-d device`" is broken for the `discover` command. Change [18de3a6](https://github.com/linux-nvme/nvme-cli/commit/18de3a6d61a7f090ecea9f6afa52c2d6591d1700) (6/16/21) introduces a set of common options (`NVMF_ARGS` - [link](https://github.com/linux-nvme/nvme-cli/blob/master/fabrics.c#L90)) that are now applied to the `discover` command. `NVMF_ARGS` contains a `-d` option that refers to the `disable-sqflow` attribute. This option used to only apply to the `connect` command, but now it is also applied to the `discover` command, which conflicts with the `-d device` option.

Furthermore, the function `nvme_read_config()` ignores the case where the config file (`/etc/nvme/config.json`) is missing and simply returns 0 when the file is missing. This is then interpreted by `nvmf_discover()` as "the json config file is valid" and it proceeds to set `json_config = true;`. This later result in `discover_from_json_config_file()` being used to retrieve the configuration and I believe that this will return the first controller it can find, which explains why we're still getting log pages displayed.

My question is: Does `disable-sqflow` apply to discovery controllers? If not, it should be moved to the specific commands where it applies to eliminate the conflict with the `discover` command.

martin-belanger, 2023-07-06:
Looking more closely at `NVMF_ARGS` I found another conflict with the `-S` option. 

Here's the list of options and the conflicting parameters they refer to:
- `-d`: `device` and `disable-sqflow`
- `-S`: `quiet` and `dhchap-secret`

igaw, 2023-07-10:
> Does disable-sqflow apply to discovery controllers? 

Don't know, need to dig into the spec for this. Though we should really avoid having conflicts options. 

My take is that for more specialized commands we could just have the long format `--option` and drop the `-o` style. At least these conflicting one we could do this. It's not that we break anything what is not working right now :)

> simply returns 0 when the file is missing. This is then interpreted by nvmf_discover() as "the json config file is valid"

This is obviously a buggy and we should fix this.

PR #2414 (2024-08-01): Don't trigger discover from json file if it doesn't exist
fabrics: check if json config is existing

nvme_read_config will return success even if the file doesn't exist.
This leads to the situation that we always execute
discover_from_json_config_file even though there is no json config
file.

One side effect of this is that when executing 'nvme discover -d
/dev/nvmeXXX' we will always do a discover for the first controller we
find.
 
While at it also resolve duplicate alias  command line options.

Fixes: #1992
closed 2024-08-01 by igaw