linux-nvme/nvme-cli #1995: /nvme id-ctrl /dev/nvme0|grep tnvmcap: report string

wxiong38, 2023-07-06:
With the latest nvme-cli, 

[root@ltcrain119-lp2 .build]# ./nvme id-ctrl /dev/nvme0|grep tnvmcap
tnvmcap   : 3,200,631,791,616

This "tnvmcap" format broken some tester's test script by using tnvmcap for calculation. 

Before it was:

[root@ltcrain119-lp2 .build]# nvme id-ctrl /dev/nvme0|grep tnvmcap
tnvmcap   : 3200631791616

Is it possible to change it back without "," or keep it in format with ","?

Thanks,
Wendy

keithbusch, 2023-07-07:
Just to be clear, I agree the format should go back to the way it was. The new way is easier to read though so I hope it's safe to tie it to the -H human readable option.

ikegami-t, 2023-07-07:
The behavior was changed by the commit 2a8bc94 and also discussed by the PR #1749 comment https://github.com/linux-nvme/nvme-cli/pull/1749#pullrequestreview-1234770082 as changed the default behavior. But if needed I will do change to remain the previous default behavior.

wxiong38, 2023-07-07:
Like Keith's suggestion:
The new way is easier to read though so I hope it's safe to tie it to the -H human readable option:
[root@ltcrain119-lp2 .build]# ./nvme id-ctrl /dev/nvme0  |grep tnvmcap
tnvmcap   : 3,200,631,791,616
[root@ltcrain119-lp2 .build]# ./nvme id-ctrl /dev/nvme0  -H |grep tnvmcap
tnvmcap   : 3,200,631,791,616

If you add readable format with -H, it is good for us.

Thanks for your help!
Wendy

ikegami-t, 2023-07-07:
Just remembered the output can be changed with the LC_ALL setting as mentioend by the issue commnet https://github.com/linux-nvme/nvme-cli/issues/1745#issuecomment-1398485936. Is it still required the default behavior change?
```sh
tokunori@tokunori-desktop:~/nvme-cli$ locale
LANG=ja_JP.UTF-8
LANGUAGE=
...
LC_NUMERIC="ja_JP.UTF-8"
LC_TIME="ja_JP.UTF-8"
...
LC_IDENTIFICATION="ja_JP.UTF-8"
LC_ALL=
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme id-ctrl /dev/nvme0 | grep tnvmcap
tnvmcap   : 960,197,124,096
tokunori@tokunori-desktop:~/nvme-cli$ export LC_ALL=C
tokunori@tokunori-desktop:~/nvme-cli$ locale
LANG=ja_JP.UTF-8
LANGUAGE=
...
LC_NUMERIC="C"
LC_TIME="C"
...
LC_IDENTIFICATION="C"
LC_ALL=C
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme id-ctrl /dev/nvme0 | grep tnvmcap
tnvmcap   : 960197124096
```

igaw, 2023-07-10:
> The new way is easier to read though so I hope it's safe to tie it to the -H human readable option.

What we could do is if `-H` is not used we set `LC_ALL=C` as default. If `-H` is used we use the `AC_ALL`  environment variable. What that work for everyone?

ikegami-t, 2023-07-10:
The suggestion looks good. Seems needed to check if there is any other command output depended on the LC_ALL setting since it is possible to be changed the behavior.

ikegami-t, 2023-07-18:
The suggested changes merged into master so looks the issue can be closed for now if no concern.