linux-nvme/nvme-cli #1997: create namespace missing Endurance Group Identifier (ENDGID)

yalanliufio, 2023-07-10:
At nvme command set specification 1.0a 
Figure 105: Namespace Management – Host Software Specified Fields

bytes 103:102   Endurance Group Identifier (ENDGID)
is added, but it is not in the nvme cli create namespace command interface, the help function shows this:

nvme create-ns --help
Usage: nvme create-ns <device> [OPTIONS]

Send a namespace management command to the specified device to create a
namespace with the given parameters. The next available namespace ID is used
for the create operation. Note that create-ns does not attach the namespace
to a controller, the attach-ns command is needed.

Options:
  [  --nsze=<IONUM>, -s <IONUM> ]       --- size of ns (NSZE)
  [  --ncap=<IONUM>, -c <IONUM> ]       --- capacity of ns (NCAP)
  [  --flbas=<NUM>, -f <NUM> ]          --- Formatted LBA size (FLBAS), if
                                            entering this value ignore
                                            'block-size' field
  [  --dps=<NUM>, -d <NUM> ]            --- data protection settings (DPS)
  [  --nmic=<NUM>, -m <NUM> ]           --- multipath and sharing
                                            capabilities (NMIC)
  [  --anagrp-id=<NUM>, -a <NUM> ]      --- ANA Group Identifier (ANAGRPID)
  [  --nvmset-id=<NUM>, -i <NUM> ]      --- NVM Set Identifier (NVMSETID)
  [  --block-size=<IONUM>, -b <IONUM> ] --- target block size, specify only
                                            if 'FLBAS' value not entered
  [  --timeout=<NUM>, -t <NUM> ]        --- timeout value, in milliseconds
  [  --csi=<NUM>, -y <NUM> ]            --- command set identifier (CSI)
  [  --lbstm=<IONUM>, -l <IONUM> ]      --- logical block storage tag mask
                                            (LBSTM)

Could you please take a look and added in future? so we can test nvme 2.0 SSD drives.

Thanks,
Yalan

ikegami-t, 2023-07-11:
If possible could you please test the PR #1998 changes?

yalanliufio, 2023-07-11:
Thanks for the quick change! I would love to do the test, but I don't have a drive support endurance group at this moment. Will do the test when I have one. Thank you!

igaw, 2023-07-11:
I've merged the change. So it should be ready for testing. If it's not working just report back.

closed 2023-07-11 by igaw