linux-nvme/nvme-cli #1999: nvme ana-log show grpid in int32 format when output is json format

bwang1011, 2023-07-11:
Hello, in nvme cli version 2.2.1, the grpid is shown in int32 format in nvme ana-log:

# json 
nvme ana-log /dev/nvme0 -o json
{
  "Asymmetric Namespace Access Log for NVMe device":"nvme0",
  "chgcnt":0,
  "ngrps":1,
  "ANA DESC LIST ":[
    {
      "grpid":-1911209053,
      "nnsids":1,
      "chgcnt":0,
      "state":"optimized",
      "NSIDS":[
        {
          "nsid":9
        }
      ]
    }
  ]
}

# normal
nvme ana-log /dev/nvme0        
Asymmetric Namespace Access Log for NVMe device: nvme0
ANA LOG HEADER :-
chgcnt	:	0
ngrps	:	1
ANA Log Desc :-
grpid	:	2383758243
nnsids	:	1
chgcnt	:	0
state	:	optimized
	nsid	:	9

igaw, 2023-07-12:
I think this one is fixed with the latest version. Could you retest with v2.5?

bwang1011, 2023-07-12:
Yes, it is fixed in the latest version.

# nvme ana-log /dev/nvme0 -o json
Asymmetric Namespace Access Log for NVMe device: nvme0
ANA LOG HEADER :-
chgcnt	:	0
ngrps	:	1
ANA Log Desc :-
grpid	:	4208779832
nnsids	:	1
chgcnt	:	0
state	:	optimized
	nsid	:	519

# nvme ana-log /dev/nvme0
Asymmetric Namespace Access Log for NVMe device: nvme0
ANA LOG HEADER :-
chgcnt	:	0
ngrps	:	1
ANA Log Desc :-
grpid	:	4208779832
nnsids	:	1
chgcnt	:	0
state	:	optimized
	nsid	:	519

closed 2023-07-12 by bwang1011

calebsander, 2023-07-13:
(Specifically by #1849)