linux-nvme/nvme-cli #2002: What do "double free or corruption" and "Invalid Field in Command" errors mean?

imthenachoman, 2023-07-15:
I'll start by saying I'm not that technical that I understand how NVMe architecture works.

I have an NVMe drive on my daily driver (Debian + KDE).

Every now and then I ger errors from smart:

> The following warning/error was logged by the smartd daemon:
> 
> Device: /dev/nvme0, number of Error Log entries increased from 2230 to 2232
> 
> Device info:
> KBG30ZMV256G TOSHIBA, S/N:X8OPD1PGP12P, FW:ADHA0101

I see these errors when I run various check commands.

```
$ sudo nvme error-log /dev/nvme0 | more
Error Log Entries for device:nvme0 entries:64
.................
 Entry[ 0]   
.................
error_count     : 2235
sqid            : 0
cmdid           : 0x11
status_field    : 0x6002(Invalid Field in Command: A reserved coded value or an unsupported value in a defined field)
phase_tag       : 0x1
parm_err_loc    : 0x28
lba             : 0xffffffffffffffff
nsid            : 0x1
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
cs              : 0
trtype_spec_info: 0
.................
 Entry[ 1]   
.................
error_count     : 2234
sqid            : 0
cmdid           : 0x5008
status_field    : 0x6002(Invalid Field in Command: A reserved coded value or an unsupported value in a defined field)
phase_tag       : 0x1
parm_err_loc    : 0x28
lba             : 0xffffffffffffffff
nsid            : 0
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
cs              : 0
trtype_spec_info: 0
.................
 Entry[ 2]   
.................
error_count     : 0
sqid            : 0
cmdid           : 0
status_field    : 0(Successful Completion: The command completed without error)
phase_tag       : 0
parm_err_loc    : 0
lba             : 0
nsid            : 0
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
cs              : 0
trtype_spec_info: 0
.................
 Entry[ 3]   
.................
(...)
```

And:

```
$ sudo nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            X8OPD1PGP12P         KBG30ZMV256G TOSHIBA                     1         256.06  GB / 256.06  GB    512   B +  0 B   ADHA0101
```

And:

```
$ sudo nvme show-regs -H /dev/nvme0n1
get-property: Operation not permitted
cap     : 55b3a4dd5480
        Controller Ready With Media Support (CRWMS): Not Supported
        Controller Ready Independent of Media Support (CRIMS): Not Supported
        Controller Memory Buffer Supported (CMBS): The Controller Memory Buffer is Not Supported
        Persistent Memory Region Supported (PMRS): The Persistent Memory Region is Not Supported
        Memory Page Size Maximum         (MPSMAX): 4096 bytes
        Memory Page Size Minimum         (MPSMIN): 4096 bytes
        Boot Partition Support              (BPS): No
        Command Sets Supported              (CSS): NVM command set is Supported
                                                   One or more I/O Command Sets are Not Supported
                                                   Only Admin Command Set Supported
        NVM Subsystem Reset Supported     (NSSRS): Yes
        Doorbell Stride                   (DSTRD): 32 bytes
        Timeout                              (TO): 82000 ms
        Arbitration Mechanism Supported     (AMS): Weighted Round Robin with Urgent Priority Class is not supported
        Contiguous Queues Required          (CQR): Yes
        Maximum Queue Entries Supported    (MQES): 21633

version : 1f81acc0
        NVMe specification 8065.172

cc      : 0
        Controller Ready Independent of Media Enable (CRIME): Disabled
        I/O Completion Queue Entry Size (IOCQES): 1 bytes
        I/O Submission Queue Entry Size (IOSQES): 1 bytes
        Shutdown Notification              (SHN): No notification; no effect
        Arbitration Mechanism Selected     (AMS): Round Robin
        Memory Page Size                   (MPS): 4096 bytes
        I/O Command Set Selected           (CSS): NVM Command Set
        Enable                              (EN): No

csts    : 0
        Processing Paused               (PP): No
        NVM Subsystem Reset Occurred (NSSRO): No
        Shutdown Status               (SHST): Normal operation (no shutdown has been requested)
        Controller Fatal Status        (CFS): False
        Ready                          (RDY): No

double free or corruption (!prev)
Aborted
```

I'm not sure what "double free or corruption" and "Invalid Field in Command" mean? Is my NVMe going bad? Should I start shopping for a new one?

ikegami-t, 2023-07-15:
About the show-regs command `free` error seems a same issue with the issue #1903 so could you pelase try the latest version nvme-cli and which is used currently by the way?

imthenachoman, 2023-07-15:
I'm using what is packaged for Debian 12.

> nvme version 2.3 (git 2.3)
> libnvme version 1.3 (git 1.3)

Let me see if I can figure out how to install a newer version.

igaw, 2023-07-20:
There is a AppImage available for the current HEAD produced from our CI.

https://monom.org/linux-nvme/upload/AppImage/nvme-cli-latest-x86_64.AppImage

igaw, 2023-09-20:
Closing this for now. If there is still a problem, feel free to reopen.

closed 2023-09-20 by igaw

imthenachoman, 2023-10-14:
@igaw I finally got around to trying the AppImage. I'm not sure how to interpret this. Wasn't sure if you wanted me to open a new issue.

```
$ sudo ./nvme-cli-latest-x86_64.AppImage error-log /dev/nvme0 | more

Error Log Entries for device:nvme0 entries:64
.................
 Entry[ 0]   
.................
error_count     : 2514
sqid            : 0
cmdid           : 0x2011
status_field    : 0x6001(Invalid Command Opcode: A reserved coded value or an unsupported value in the command opcode field)
phase_tag       : 0
parm_err_loc    : 0
lba             : 0xffffffffffffffff
nsid            : 0x4
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
csi             : 0
opcode          : 0
cs              : 0
trtype_spec_info: 0
log_page_version: 0
.................
 Entry[ 1]   
.................
error_count     : 2513
sqid            : 0
cmdid           : 0x2010
status_field    : 0x6002(Invalid Field in Command: A reserved coded value or an unsupported value in a defined field)
phase_tag       : 0
parm_err_loc    : 0x28
lba             : 0xffffffffffffffff
nsid            : 0x1
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
csi             : 0
opcode          : 0
cs              : 0
trtype_spec_info: 0
log_page_version: 0
.................
 Entry[ 2]   
.................
error_count     : 2512
sqid            : 0
cmdid           : 0x1012
status_field    : 0x6002(Invalid Field in Command: A reserved coded value or an unsupported value in a defined field)
phase_tag       : 0
parm_err_loc    : 0x28
lba             : 0xffffffffffffffff
nsid            : 0x1
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
csi             : 0
opcode          : 0
cs              : 0
trtype_spec_info: 0
log_page_version: 0
.................
 Entry[ 3]   
.................
...
```

```
$ sudo ./nvme-cli-latest-x86_64.AppImage list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            X8OPD1PGP12P         KBG30ZMV256G TOSHIBA                     0x1        256.06  GB / 256.06  GB    512   B +  0 B   ADHA0101
```

```
$ sudo ./nvme-cli-latest-x86_64.AppImage show-regs -H /dev/nvme0n1
get-property: Invalid argument
```

ikegami-t, 2023-10-14:
Possibly related to the issue #1941 but the get-property error type is different as https://github.com/linux-nvme/nvme-cli/issues/1941#issuecomment-1553937039 so not sure. Anyway the current issue behavior is defferent from the original free error so seems better to open a new issue.

imthenachoman, 2023-10-15:
Opened #2092

igaw, 2023-11-02:
Thanks for testing. The `double free or corruption` bug is fixed (which is this issue about). Yep, let's discuss the other problem in the new issue and leave this closed.