linux-nvme/nvme-cli #2009: Unhelpful/inconsistent output of --help to stderr

scfc, 2023-07-22:
With nvme 2.2.1 on Fedora 37, I wanted to check if there are options related to trimming:

````
$ nvme --help | fgrep -i trim
The '<device>' may be either an NVMe character device (ex: /dev/nvme0), an
nvme block device (ex: /dev/nvme0n1), or a mctp address in the form
mctp:<net>,<eid>[:ctrl-id]$
````

Try again:

````
$ nvme --help > /dev/null 
The '<device>' may be either an NVMe character device (ex: /dev/nvme0), an
nvme block device (ex: /dev/nvme0n1), or a mctp address in the form
mctp:<net>,<eid>[:ctrl-id]$
````

So:

1. All output of `nvme --help` should go to stdout, not (partly) to stderr, so that for example one can do `nvme --help | less`.
2. The newline after the line `mctp:<net>,<eid>[:ctrl-id]` should be output to the same destination as the line itself :-).

ikegami-t, 2023-07-23:
The behavior was fixed by the PR #1772 changes b186f34 for the issue #1766.

scfc, 2023-07-24:
Thanks for the pointer.  I haven't verified it, but it appears that this is a true duplicate of issue #1766 and thus should be fixed by your patch.

closed 2023-07-24 by scfc