linux-nvme/nvme-cli #2010: ERROR: Invalid sub-command 'zns'

OGSD, 2023-07-24:
My nvme-cli version is 2.5 and linux kernel is 5.15. When i use sudo nvme zns help, it prints  ERROR: Invalid sub-command 'zns'.
I don't know how to solve.

igaw, 2023-07-24:
Did you compile the cli yourself? The plugins are currently disabled when the json-c library is missing (usually called somehting like `libjson-c-dev`)

OGSD, 2023-07-25:
yes i download the code from this github and do 
$ meson setup --force-fallback-for=libnvme .build
$ meson compile -C .build
# meson install -C .build
$ make
# make install
$ make rpm
$ man nvme
$ nvme help

so you mean i need to download jason-c library this library?

OGSD, 2023-07-25:
> Did you compile the cli yourself? The plugins are currently disabled when the json-c library is missing (usually called somehting like `libjson-c-dev`)

yes i download the code from this gihub and do 
meson setup --force-fallback-for=libnvme .build
meson setup .build
meson compile -C .build
meson install -C .build
make
make install
make rpm
man nvme
nvme help
done 

so you mean i need to download jason-c libarary ?

OGSD, 2023-07-25:
The Meson build system
Version: 1.2.0
Source dir: /home/ogsd/Downloads/nvme-cli-master
Build dir: /home/ogsd/Downloads/nvme-cli-master/libnvme
Build type: native build
Project name: nvme-cli
Project version: 2.5
C compiler for the host machine: cc (gcc 9.4.0 "cc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C linker for the host machine: cc ld.bfd 2.34
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Run-time dependency libnvme found: YES 1.5
Run-time dependency libnvme-mi found: YES 1.5
Run-time dependency json-c found: YES 0.13.1
Has header "hugetlbfs.h" : NO 
Checking if "__builtin_type_compatible_p" compiles: YES 
Checking if "typeof" compiles: YES 
Checking if "byteswap.h" compiles: YES 
Checking if "bswap64" : links: YES 
Checking if "isblank" : links: YES 
Checking if "sys/random.h" compiles: YES 
Compiler for C supports function attribute fallthrough: YES 
Configuring config.h using configuration
Configuring nvme.spec using configuration
Configuring discovery.conf using configuration
Configuring 70-nvmf-autoconnect.conf using configuration
Configuring nvmefc-boot-connections.service using configuration
Configuring nvmf-autoconnect.service using configuration
Configuring nvmf-connect.target using configuration
Configuring nvmf-connect@.service using configuration
Configuring 70-nvmf-autoconnect.rules using configuration
Configuring 71-nvmf-iopolicy-netapp.rules using configuration
Build targets in project: 6

nvme-cli 2.5

  Paths
    prefixdir     : /usr/local
    sysconfdir    : /usr/local/etc
    sbindir       : /usr/local/sbin
    datadir       : /usr/local/share
    mandir        : /usr/local/share/man
    udevrulesdir  : /usr/local/lib/udev/rules.d
    dracutrulesdir: /usr/local/lib/dracut/dracut.conf.d/
    rundir        : /usr/local/run
    systemddir    : /usr/local/lib/systemd/system
    build location: /home/ogsd/Downloads/nvme-cli-master/libnvme

  Dependencies
    json-c        : True
    libhugetlbfs  : False

  Configuration
    pdc enabled   : False

Found ninja-1.10.0 at /usr/bin/ninja

Dependencies json-c is true, but i still cannot use zns command

igaw, 2023-08-11:
Feel free to reopen if you still have a problem.

closed 2023-08-11 by igaw