linux-nvme/nvme-cli #2011: Unable to update firmware - INVALID_FIELD

therokh, 2023-07-24:
Hi,

I am trying to update the firmware of a SAMSUNG MZVLQ256HBJD-00BH1 M.2 SSD from a Proxmox 7.4.3 host.
The nvme fw-download command returns an INVALID_FIELD error

`uname -a`

> Linux pve003 5.15.107-2-pve #1 SMP PVE 5.15.107-2 (2023-05-10T09:10Z) x86_64 GNU/Linux

`nvme version`

> nvme version 1.12

`nvme fw-download --fw HPS4NFAV_20211127.bin /dev/nvme0`

> NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x2)

`nvme id-ctrl -H /dev/nvme0`
/
> NVME Identify Controller:
vid       : 0x144d
ssvid     : 0x144d
sn        : [REMOVED]
mn        : SAMSUNG MZVLQ256HBJD-00BH1
fr        : HPS1NFAV
rab       : 2
ieee      : 002538
cmic      : 0
  [3:3] : 0     ANA not supported
  [2:2] : 0     PCI
  [1:1] : 0     Single Controller
  [0:0] : 0     Single Port
mdts      : 9
cntlid    : 0x5
ver       : 0x10400
rtd3r     : 0x7a120
rtd3e     : 0x7a1200
oaes      : 0x200
[14:14] : 0     Endurance Group Event Aggregate Log Page Change Notice Not Supported
[13:13] : 0     LBA Status Information Notices Not Supported
[12:12] : 0     Predictable Latency Event Aggregate Log Change Notices Not Supported
[11:11] : 0     Asymmetric Namespace Access Change Notices Not Supported
  [9:9] : 0x1   Firmware Activation Notices Supported
  [8:8] : 0     Namespace Attribute Changed Event Not Supported
ctratt    : 0x10
  [9:9] : 0     UUID List Not Supported
  [7:7] : 0     Namespace Granularity Not Supported
  [5:5] : 0     Predictable Latency Mode Not Supported
  [4:4] : 0x1   Endurance Groups Supported
  [3:3] : 0     Read Recovery Levels Not Supported
  [2:2] : 0     NVM Sets Not Supported
  [1:1] : 0     Non-Operational Power State Permissive Not Supported
  [0:0] : 0     128-bit Host Identifier Not Supported
rrls      : 0
cntrltype : 1
  [7:2] : 0     Reserved
  [1:0] : 0x1   I/O Controller
fguid     :
crdt1     : 0
crdt2     : 0
crdt3     : 0
oacs      : 0x17
  [9:9] : 0     Get LBA Status Capability Not Supported
  [8:8] : 0     Doorbell Buffer Config Not Supported
  [7:7] : 0     Virtualization Management Not Supported
  [6:6] : 0     NVMe-MI Send and Receive Not Supported
  [5:5] : 0     Directives Not Supported
  [4:4] : 0x1   Device Self-test Supported
  [3:3] : 0     NS Management and Attachment Not Supported
  [2:2] : 0x1   FW Commit and Download Supported
  [1:1] : 0x1   Format NVM Supported
  [0:0] : 0x1   Security Send and Receive Supported
acl       : 7
aerl      : 7
frmw      : 0x16
  [4:4] : 0x1   Firmware Activate Without Reset Supported
  [3:1] : 0x3   Number of Firmware Slots
  [0:0] : 0     Firmware Slot 1 Read/Write
lpa       : 0xf
  [4:4] : 0     Persistent Event log Not Supported
  [3:3] : 0x1   Telemetry host/controller initiated log page Supported
  [2:2] : 0x1   Extended data for Get Log Page Supported
  [1:1] : 0x1   Command Effects Log Page Supported
  [0:0] : 0x1   SMART/Health Log Page per NS Supported
elpe      : 63
npss      : 4
avscc     : 0x1
  [0:0] : 0x1   Admin Vendor Specific Commands uses NVMe Format
apsta     : 0x1
  [0:0] : 0x1   Autonomous Power State Transitions Supported
wctemp    : 354
cctemp    : 358
mtfa      : 0
hmpre     : 16384
hmmin     : 4096
tnvmcap   : 256060514304
unvmcap   : 0
rpmbs     : 0
 [31:24]: 0     Access Size
 [23:16]: 0     Total Size
  [5:3] : 0     Authentication Method
  [2:0] : 0     Number of RPMB Units
edstt     : 35
dsto      : 0
fwug      : 4
kas       : 0
hctma     : 0x1
  [0:0] : 0x1   Host Controlled Thermal Management Supported
mntmt     : 318
mxtmt     : 356
sanicap   : 0x60000002
  [31:30] : 0x1 Media is not additionally modified after sanitize operation completes successfully
  [29:29] : 0x1 No-Deallocate After Sanitize bit in Sanitize command Not Supported
    [2:2] : 0   Overwrite Sanitize Operation Not Supported
    [1:1] : 0x1 Block Erase Sanitize Operation Supported
    [0:0] : 0   Crypto Erase Sanitize Operation Not Supported
hmminds   : 16
hmmaxd    : 1024
nsetidmax : 0
endgidmax : 1
anatt     : 0
anacap    : 0
  [7:7] : 0     Non-zero group ID Not Supported
  [6:6] : 0     Group ID does not change
  [4:4] : 0     ANA Change state Not Supported
  [3:3] : 0     ANA Persistent Loss state Not Supported
  [2:2] : 0     ANA Inaccessible state Not Supported
  [1:1] : 0     ANA Non-optimized state Not Supported
  [0:0] : 0     ANA Optimized state Not Supported
anagrpmax : 0
nanagrpid : 0
pels      : 0
sqes      : 0x66
  [7:4] : 0x6   Max SQ Entry Size (64)
  [3:0] : 0x6   Min SQ Entry Size (64)
cqes      : 0x44
  [7:4] : 0x4   Max CQ Entry Size (16)
  [3:0] : 0x4   Min CQ Entry Size (16)
maxcmd    : 0
nn        : 1
oncs      : 0x5f
  [7:7] : 0     Verify Not Supported
  [6:6] : 0x1   Timestamp Supported
  [5:5] : 0     Reservations Not Supported
  [4:4] : 0x1   Save and Select Supported
  [3:3] : 0x1   Write Zeroes Supported
  [2:2] : 0x1   Data Set Management Supported
  [1:1] : 0x1   Write Uncorrectable Supported
  [0:0] : 0x1   Compare Supported
fuses     : 0
  [0:0] : 0     Fused Compare and Write Not Supported
fna       : 0
  [2:2] : 0     Crypto Erase Not Supported as part of Secure Erase
  [1:1] : 0     Crypto Erase Applies to Single Namespace(s)
  [0:0] : 0     Format Applies to Single Namespace(s)
vwc       : 0x7
  [2:1] : 0x3   The Flush command supports NSID set to FFFFFFFFh
  [0:0] : 0x1   Volatile Write Cache Present
awun      : 1023
awupf     : 0
nvscc     : 1
  [0:0] : 0x1   NVM Vendor Specific Commands uses NVMe Format
nwpc      : 0
  [2:2] : 0     Permanent Write Protect Not Supported
  [1:1] : 0     Write Protect Until Power Supply Not Supported
  [0:0] : 0     No Write Protect and Write Protect Namespace Not Supported
acwu      : 0
sgls      : 0
 [1:0]  : 0     Scatter-Gather Lists Not Supported
mnan      : 0
subnqn    : nqn.1994-11.com.samsung:nvme:PM991aM.2:S672NA1T752504
ioccsz    : 0
iorcsz    : 0
icdoff    : 0
ctrattr   : 0
  [0:0] : 0     Dynamic Controller Model
msdbd     : 0
ps    0 : mp:5.36W operational enlat:0 exlat:0 rrt:0 rrl:0
          rwt:0 rwl:0 idle_power:- active_power:-
ps    1 : mp:4.47W operational enlat:0 exlat:0 rrt:1 rrl:1
          rwt:1 rwl:1 idle_power:- active_power:-
ps    2 : mp:2.23W operational enlat:0 exlat:500 rrt:2 rrl:2
          rwt:2 rwl:2 idle_power:- active_power:-
ps    3 : mp:0.0500W non-operational enlat:210 exlat:1200 rrt:3 rrl:3
          rwt:3 rwl:3 idle_power:- active_power:-
ps    4 : mp:0.0050W non-operational enlat:1000 exlat:9000 rrt:4 rrl:4
          rwt:4 rwl:4 idle_power:- active_power:-

igaw, 2023-07-24:
It didn't work with `nvme version 1.12` nor `nvme version 2.3`. That means it's not a regression.

```
fw-download: error on offset 0x00000000/0x00200000
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)
retrying offset 0 (1/3)
```

This indicates that the firmware rejects the new binary, right at the beginning. I suppose the firmware has some
sort of magic header or is even signed and the firmware just rejects it.  Are you sure the binary you got is the correct one?

As far I can tell, we have several positive feedback that the firmware upload code works for several different SSDs. This doesn't mean there no bugs, but I would first check if the firmware you want to flash is matching.

therokh, 2023-07-28:
Thanks, it's entirely possible that this firmware is not compatible or not signed correctly.

igaw, 2023-08-08:
Okay, let's close this issue then. You could try to ask Samsung if they got a firmware update for your device though. Not sure how likely they have one. The best thing they could do is to start uploading their firmware updates to https://fwupd.org/ This would be awesome.

closed 2023-08-08 by igaw