linux-nvme/nvme-cli #2013: Invalid FW image error (0x2107) with split files

tkenbo, 2023-07-25:
When we try downloading split firmware file with offset, it reported an error 2107 from fw_commit command with version 2.4.
Tried it with v1.16 and it did not report an error.

here is the command I executed.
```
nvme fw-download /dev/nvme2 -f file_0.bin -x 262144 -o 0
nvme fw-download /dev/nvme2 -f file_1.bin -x 262144 -o 50176
nvme fw-download /dev/nvme2 -f file_2.bin -x 262144 -o 161792
nvme fw-download /dev/nvme2 -f file_3.bin -x 262144 -o 256000
nvme fw-download /dev/nvme2 -f file_4.bin -x 262144 -o 385024
nvme fw-download /dev/nvme2 -f file_5.bin -x 262144 -o 481280
nvme fw-download /dev/nvme2 -f file_6.bin -x 262144 -o 560128
nvme fw-commit /dev/nvme2 -s 1 -a 1
NVMe status: Invalid Firmware Image: The firmware image specified for activation is invalid and not loaded by the controller(0x2107)
```
fw_commit completes successfully with v1.16.
```
nvme fw-commit /dev/nvme2 -s 1 -a 1
Success committing firmware action:1 slot:1
Multiple Update Detected (MUD) Value: 0
```
Looking at the trace in v2.4, we only see a few download commands sent to the drive.
```
            nvme-249925  [027] ..... 424627.900917: nvme_setup_cmd: nvme2: qid=0, cmdid=49176, nsid=0, flags=0x0, meta=0x0, cmd=(nvme_admin_download_fw cdw10=ff c3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
          <idle>-0       [047] d.h1. 424627.901595: nvme_sq: nvme2: qid=0, head=22, tail=22
          <idle>-0       [027] .Ns1. 424627.901608: nvme_complete_rq: nvme2: qid=0, cmdid=49176, res=0x0, retries=0, flags=0x2, status=0x0
            nvme-249949  [024] ..... 424747.717830: nvme_setup_cmd: nvme2: qid=0, cmdid=20493, nsid=0, flags=0x0, meta=0x0, cmd=(nvme_admin_download_fw cdw10=ff ff 00 00 00 c4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
          <idle>-0       [047] d.h1. 424747.718531: nvme_sq: nvme2: qid=0, head=23, tail=23
          <idle>-0       [024] .Ns1. 424747.718543: nvme_complete_rq: nvme2: qid=0, cmdid=20493, res=0x0, retries=0, flags=0x2, status=0x0
            nvme-249973  [025] ..... 424861.603381: nvme_setup_cmd: nvme2: qid=0, cmdid=16400, nsid=0, flags=0x0, meta=0x0, cmd=(nvme_admin_activate_fw cdw10=09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
          <idle>-0       [047] d.h1. 424861.638943: nvme_sq: nvme2: qid=0, head=25, tail=25
          <idle>-0       [025] ..s1. 424861.638966: nvme_complete_rq: nvme2: qid=0, cmdid=16400, res=0x0, retries=0, flags=0x2, status=0x2107

```
looking at the source code, noticed some changes in this change https://github.com/linux-nvme/nvme-cli/commit/e7ca3bcbdba36861409d91c74ce49be489878112.
`while (fw_size > 0)  -> while (cfg.offset < fw_size)`
This makes the offset must be smaller than size of split file but it's larger in our case and that makes this function to not working as expected.

igaw, 2023-07-26:
@jk-ozlabs

jk-ozlabs, 2023-07-26:
Just to clarify: you're wanting to perform a transfer of each file's first byte to be transferred at the offset provided on the command line?

At present, using `--offset n` will transfer from offset `n` of the input file to offset `n` of the destination firmware image; which may mean you don't have to split the file up. I'm happy to revert the the previous behaviour if that's the preference though.

jk-ozlabs, 2023-07-27:
OK, shall do! Will send a patch shortly.

igaw, 2023-09-20:
@jk-ozlabs not sure if I missed a PR for this?

jk-ozlabs, 2023-09-20:
> @jk-ozlabs not sure if I missed a PR for this?

Nope, just me being busy, sorry! I'm traveling at the moment, but will get a fix done when I'm back with access to hardware to test against.

igaw, 2023-09-21:
No worries. Just wrapping this release cycle up and stumbled across this report.

jk-ozlabs, 2023-09-23:
PR sent, but I'm not able to test right now. @tkenbo, are you able to give that a try?

closed 2023-09-27 by igaw