linux-nvme/nvme-cli #2016: Write command error: NVMe status: PRP Offset Invalid: The Offset field for a PRP entry is invalid(0x13)

yalanliufio, 2023-07-26:
With NVMe 2.4, write command abort with PRP Offset Invalid. This happen on format 512B+ 8B, 512B is data size, 8B is metadata size,  metdata Contiguous with LBA Data, Forming Extended LBA

Please refer NVMe Command Set Specification 1.0c
Figure 149: Metadata – Contiguous with LBA Data, Forming Extended LBA

Steps to repeat the problem.
1. Identify controller data structure command:
nvme id-ns /dev/nvme1 -n 1 
output format:
lbaf  0 : ms:0   lbads:9  rp:0 
lbaf  1 : ms:8   lbads:9  rp:0 (in use)
lbaf  2 : ms:0   lbads:12 rp:0 
lbaf  3 : ms:8   lbads:12 rp:0 
lbaf  4 : ms:64  lbads:12 rp:0

2. Create and attach namespace command:
nvme create-ns /dev/nvme1 -s 1000000 -c 1000000 -f 17 -d 0 -m 0
create-ns: Success, created nsid:1
nvme attach-ns /dev/nvme1 -n 1 -c 1
attach-ns: Success, nsid:1

3. issue write command:
nvme write /dev/nvme1n1 -s 0 -c 14 -z 7800 -d write_7800.data
NVMe status: PRP Offset Invalid: The Offset field for a PRP entry is invalid(0x2013)

Repeat previous step with nvme cli 2.2.1, write command finish successfully and has no issue

igaw, 2023-07-26:
Is this with the same kernel? Or did you also update the kernel?

keithbusch, 2023-07-26:
Hm, that's strange. The kernel is supposed to enforce the alignment regardless of what user space provides. The way the code is currently calculating number of blocks and buffer size looks suspicious, though. Could you run the same 'nvme write' command with the '--dry-run' option?

igaw, 2023-07-26:
I was going through the commits, one which stands out is this one:

8a2bfa5be4d3 ("nvme: fix block count and data size logic")

ikegami-t, 2023-07-26:
The write command is not specified the metadata parameters below but the data file size including metadata size is this correct and also is it resolved the issue if specified the parameters?
```
  [  --metadata-size=<IONUM>, -y <IONUM> ] --- size of metadata in bytes
  [  --metadata=<FILE>, -M <FILE> ]     --- metadata file
```

yalanliufio, 2023-07-26:
When metadata is transfer as extended LBA, -z option includes the metadata size. If metadata is transfer as separate buffer, then option -y and -M is used, it has been like this in version 1.16 and 2.2.1

yalanliufio, 2023-07-26:
nvme write /dev/nvme1n1 -s 0 -c 14 -z 7800 -d /tmp/write_7800.data --dry-run
opcode       : 01
nsid         : 01
flags        : 00
control      : 0000
nblocks      : 000f
metadata     : 0
addr         : c96000
slba         : 0
dsmgmt       : 00000000
reftag       : 0
apptag       : 0000
appmask      : 0000
storagetagcheck : 0000
storagetag      : 0
pif             : 00
sts             : 00

keithbusch, 2023-07-26:
It's definitely rounding incorrectly for extended metadata format. I am not sure how it results in the error you're seeing in this particular test, but definitely needs to be fixed.

keithbusch, 2023-07-26:
Wait, I see how it's happening. We're rounding the blocks from the requested 15 to 16, and with extended metadata, the implied size is 8320 bytes. That requires a PRP list, but we lied to the driver about the buffer size to DMA map and called in 8192 instead, so the kernel sets up a regular PRP entry instead, and the controller misinterprets the structure. Getting an Invalid PRP error is the best possible outcome for this mistake; the alternative could be data corruption.

keithbusch, 2023-07-26:
I have a fix, but damnit, github is unusable for me at the moment. I need to find another computer outside my corporate firewall that has the keys. In the meantime, here's the patch...

```
From 1fa535161436d2cc9eea21957fe2d5cb4a831737 Mon Sep 17 00:00:00 2001
From: Keith Busch <kbusch@kernel.org>
Date: Wed, 26 Jul 2023 10:30:47 -0700
Subject: [PATCH] nvme-cli: fix extended metadata size calculation

Link: https://github.com/linux-nvme/nvme-cli/issues/2016

---
 nvme.c | 26 ++++++++++++++------------
 1 file changed, 14 insertions(+), 12 deletions(-)

diff --git a/nvme.c b/nvme.c
index b35f9003..01dbad26 100644
--- a/nvme.c
+++ b/nvme.c
@@ -7549,6 +7549,20 @@ static int submit_io(int opcode, char *command, const char *desc, int argc, char
 	if (nvme_get_logical_block_size(dev_fd(dev), cfg.namespace_id, &logical_block_size) < 0)
 		goto close_mfd;
 
+	err = nvme_cli_identify_ns(dev, cfg.namespace_id, &ns);
+	if (err > 0) {
+		nvme_show_status(err);
+		goto close_mfd;
+	} else if (err < 0) {
+		fprintf(stderr, "identify namespace: %s\n", nvme_strerror(errno));
+		goto close_mfd;
+	}
+
+	nvme_id_ns_flbas_to_lbaf_inuse(ns.flbas, &lba_index);
+	ms = ns.lbaf[lba_index].ms;
+	if (ns.flbas & NVME_NS_FLBAS_META_EXT)
+		logical_block_size += ms;
+
 	buffer_size = ((long long)cfg.block_count + 1) * logical_block_size;
 	if (cfg.data_size < buffer_size)
 		nvme_show_error("Rounding data size to fit block count (%lld bytes)", buffer_size);
@@ -7568,18 +7582,6 @@ static int submit_io(int opcode, char *command, const char *desc, int argc, char
 	}
 
 	if (cfg.metadata_size) {
-		err = nvme_cli_identify_ns(dev, cfg.namespace_id, &ns);
-		if (err > 0) {
-			nvme_show_status(err);
-			goto free_buffer;
-		} else if (err < 0) {
-			nvme_show_error("identify namespace: %s", nvme_strerror(errno));
-			goto free_buffer;
-		}
-
-		nvme_id_ns_flbas_to_lbaf_inuse(ns.flbas, &lba_index);
-		ms = ns.lbaf[lba_index].ms;
-
 		err = nvme_identify_ns_csi(dev_fd(dev), 1, 0, NVME_CSI_NVM, &nvm_ns);
 		if (!err) {
 			sts = nvm_ns.elbaf[lba_index] & NVME_NVM_ELBAF_STS_MASK;
-- 
2.34.1

```

yalanliufio, 2023-07-26:
Thanks for the quick fix, could you let me know how to get the fix code so I can test it. 
I am using the nvme cli 2.4 now, shall I get the tips code

> I have a fix, but damnit, github is unusable for me at the moment. I need to find another computer outside my corporate firewall that has the keys. In the meantime, here's the patch...
> 
> ```
> From 1fa535161436d2cc9eea21957fe2d5cb4a831737 Mon Sep 17 00:00:00 2001
> From: Keith Busch <kbusch@kernel.org>
> Date: Wed, 26 Jul 2023 10:30:47 -0700
> Subject: [PATCH] nvme-cli: fix extended metadata size calculation
> 
> Link: https://github.com/linux-nvme/nvme-cli/issues/2016
> Signed-off-by: Keith Busch <kbusch@kernel.org>
> ---
>  nvme.c | 26 ++++++++++++++------------
>  1 file changed, 14 insertions(+), 12 deletions(-)
> 
> diff --git a/nvme.c b/nvme.c
> index b35f9003..01dbad26 100644
> --- a/nvme.c
> +++ b/nvme.c
> @@ -7549,6 +7549,20 @@ static int submit_io(int opcode, char *command, const char *desc, int argc, char
>  	if (nvme_get_logical_block_size(dev_fd(dev), cfg.namespace_id, &logical_block_size) < 0)
>  		goto close_mfd;
>  
> +	err = nvme_cli_identify_ns(dev, cfg.namespace_id, &ns);
> +	if (err > 0) {
> +		nvme_show_status(err);
> +		goto close_mfd;
> +	} else if (err < 0) {
> +		fprintf(stderr, "identify namespace: %s\n", nvme_strerror(errno));
> +		goto close_mfd;
> +	}
> +
> +	nvme_id_ns_flbas_to_lbaf_inuse(ns.flbas, &lba_index);
> +	ms = ns.lbaf[lba_index].ms;
> +	if (ns.flbas & NVME_NS_FLBAS_META_EXT)
> +		logical_block_size += ms;
> +
>  	buffer_size = ((long long)cfg.block_count + 1) * logical_block_size;
>  	if (cfg.data_size < buffer_size)
>  		nvme_show_error("Rounding data size to fit block count (%lld bytes)", buffer_size);
> @@ -7568,18 +7582,6 @@ static int submit_io(int opcode, char *command, const char *desc, int argc, char
>  	}
>  
>  	if (cfg.metadata_size) {
> -		err = nvme_cli_identify_ns(dev, cfg.namespace_id, &ns);
> -		if (err > 0) {
> -			nvme_show_status(err);
> -			goto free_buffer;
> -		} else if (err < 0) {
> -			nvme_show_error("identify namespace: %s", nvme_strerror(errno));
> -			goto free_buffer;
> -		}
> -
> -		nvme_id_ns_flbas_to_lbaf_inuse(ns.flbas, &lba_index);
> -		ms = ns.lbaf[lba_index].ms;
> -
>  		err = nvme_identify_ns_csi(dev_fd(dev), 1, 0, NVME_CSI_NVM, &nvm_ns);
>  		if (!err) {
>  			sts = nvm_ns.elbaf[lba_index] & NVME_NVM_ELBAF_STS_MASK;
> -- 
> 2.34.1
> ```

yalanliufio, 2023-07-26:
Could you let me how to get the fix code so I can test it? I am using nvme-cli 2.4 now, can I just add your patch on the nvme-cli 2.4?

igaw, 2023-07-27:
Yes, you can apply it directly or just wait a sec. I'll apply @keithbusch patch.

igaw, 2023-07-27:
@yalanliufio if you don't want to build your own version you can also use the AppImage which gets build from our CI build. Should be up to date in a few of minutes: https://monom.org/linux-nvme/upload/AppImage/nvme-cli-latest-x86_64.AppImage

yalanliufio, 2023-07-28:
Could you let me know what release version this fix will be?

igaw, 2023-07-28:
The next release will have the fix,this will be https://github.com/linux-nvme/nvme-cli/milestone/8

yalanliufio, 2023-07-28:
great, thank you!

yalanliufio, 2023-08-08:
I would like to reopen this issue, as I still have PRP Offset Invalid error. This time the error also happens on format 512B+ 8B, 512B is data size, 8B is metadata size, metdata Contiguous with LBA Data, Forming Extended LBA. The major difference of this error with the PRP Offset Invalid error above is write command using PACT (Protection Information Action) is set to 1. 

Please refer NVM Command Set Specification, Revision 1.0c
Figure 9: Protection Information Field Definition 

Steps to repeat the problem:

1. issue identify namespace data structure with namespace -1 (all namespace)
nvme id-ns /dev/nvme4 -n -1
lbaf  0 : ms:0   lbads:9  rp:0x1 (in use)
lbaf  1 : ms:0   lbads:12 rp:0 
lbaf  2 : ms:8   lbads:9  rp:0x3 
lbaf  3 : ms:8   lbads:12 rp:0x2 
lbaf  4 : ms:64  lbads:12 rp:0x3 

2. format namespace 1 to format 2, protection 1, and metadata transfer as extended LBA
nvme format /dev/nvme4 -n 1 -l 2 -p 0 -i 1 -m 1
You are about to format nvme4, namespace 0x1.
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ... 
Success formatting namespace:1

3.issue namespace data structure with namespace id 1
nvme id-ns /dev/nvme4 -n 1
nlbaf   : 4
flbas   : 0x12
mc      : 0x3
dpc     : 0x1f
dps     : 0x1
nmic    : 0
lbaf  0 : ms:0   lbads:9  rp:0x1 
lbaf  1 : ms:0   lbads:12 rp:0 
lbaf  2 : ms:8   lbads:9  rp:0x3 (in use)
lbaf  3 : ms:8   lbads:12 rp:0x2 
lbaf  4 : ms:64  lbads:12 rp:0x3 

4. issue write commad with PACT=1
nvme write /dev/nvme4n1 -s 0 -c 15 -z 8192 -d write_8192.data -p 8
Rounding data size to fit block count (8320 bytes)
NVMe status: PRP Offset Invalid: The Offset field for a PRP entry is invalid(0x13)

Per NVM Command Set Specification, Revision 1.0c
Figure 133: Write Command 16b Guard Protection Information Processing

[image]

c) MD=8, PI, PRACT=1: Metadata not resident in host buffer
in this case, since metadata is not resident in host buffer, controller expect transfer size 8192, but nvme-cli send 8320 bytes

Repeat previous steps with nvme cli 2.2.1, write command finish successfully and has no problem.

keithbusch, 2023-08-08:
The tool is trying to be too clever and missing the corner cases.

yalanliufio, 2023-08-08:
Yeah, it is a corner cases. It just happened at the boundary of PRP entry and PRP list. I wish NVM spec has a bit in identify to indicate whether it is OK to transfer extra data in PRP, unfortunately they don't.

igaw, 2023-09-20:
@yalanliufio any chance to submit a PR for this? I don't have any test setup for this, so I can not really test if I get it right or not.

yalanliufio, 2023-09-22:
@igaw I don't know how to to fix this, but I can help to do the test if someone can help on do the fix.

igaw, 2023-09-25:
Okay, I'll see if I can figure out the problem and post a test patch.

yalanliufio, 2023-10-05:
The simple fix would be just revert back to nvme cli version 2.2.1

igaw, 2023-10-11:
There are two things we should do:
- if the user provides the `-c` option actually use this value instead calculating the value
- fix the calculation

igaw, 2023-10-11:
@yalanliufio in case you can test the above PR please give it a whirl. Thanks!

PR #2085 (2023-10-11): nvme: do not include meta data for PRACT=1 and MD=8
No meta data is transfered for PRACT=1 and MD=8:
   5.2.2.1 Protection Information and Write Commands
   5.2.2.2 Protection Informatio and Read Commands

Fixes: #2016
closed 2023-10-11 by keithbusch