linux-nvme/nvme-cli #2021: Reopen #1934: create a namesapce with entering just MB or GB capacity

wxiong38, 2023-08-02:
With the latest nvme-cli and libnvme, I still saw the issue when trying to create a  ns with GB size.

# nvme create-ns /dev/nvme0 -C 800G -f 0
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)

# ./nvme -version
nvme version 2.5 (git 2.5)
libnvme version 1.5 (git 1.5-11-g9087f71)

Do I miss something here?

Thanks,
Wendy

ikegami-t, 2023-08-02:
Seems needed to specify the option below -S namespace size (NSZE) also but not only with the option -C namespace capacity (NCAP).
```
[--nsze-si=<nsze-si> | -S <nsze-si>]
```

igaw, 2023-08-08:
This taps into the same discussion we have in https://github.com/linux-nvme/nvme-cli/issues/1982

Maybe @maurizio-lombardi made some progress.

ikegami-t, 2023-08-09:
FYI: The create-ns command parameters for SI units were implemented as new arguments by the PR comment https://github.com/linux-nvme/nvme-cli/pull/1750#issuecomment-1368553697 below.
> I feel like we need entirely new arguments, and ensure they're mutually exclusive with the existing nsze/ncap args. The old ones follow the spec defined units, so I think it'd actually be more confusing to repurpose them to include si units.

maurizio-lombardi, 2023-08-09:
> This taps into the same discussion we have in #1982
> 
> Maybe @maurizio-lombardi made some progress.

Not yet but it definitely needs attention so I will start working on it again, as you said in this [https://github.com/linux-nvme/nvme-cli/issues/1982#issuecomment-1614750434](comment), this may break the existing command line but unfortunately I can't see a better way.

igaw, 2024-04-16:
This has been fixed but I forgot to close this issue. The parser understands 'G' etc.

closed 2024-04-16 by igaw