linux-nvme/nvme-cli #2029: nvme list outputs valid list but increments error count

Hr46ph, 2023-08-10:
Running `nvme list` as root outputs a valid list but also triggers an error in the log:

```
Error Information (NVMe Log 0x01, 16 of 64 entries)
Num   ErrCount  SQId   CmdId  Status  PELoc          LBA  NSID    VS  Message
  0       1069     0  0xa001  0x4004      -            0     1     -  Invalid Field in Command
```
Specifying output-format or verbose does not change this behavior.

Other `nvme` commands such as `nvme smart-log /dev/nvme0n1` do not trigger an error.

**Distribution: Arch Linux (fully updated)
Kernel: 6.1.39-3-lts
package version: nvme-cli 2.5-1
NVMe SSD: Samsung MZVLB256HAHQ**

igaw, 2023-08-11:
We issue `nvme_get_nsid` `nvme_identify_ns` and `nvme_identify_ns_descs` when executing `nvme list`. 
For `smart-log` we only issue a `nvme_get_log`. 

So it's not clear which one of these gets logged by the firmware.  Let's try to capture the command which triggers the log entry. Could you do this:

```shell
 # cd /sys/kernel/debug/tracing/
 # echo 'status!=0' > events/nvme/nvme_complete_rq/filter
 # echo 1 > events/nvme/nvme_complete_rq/enable
 # echo 1 > events/nvme/nvme_async_event/enable
 # echo 1 > tracing_on

# nvme list

 # echo 0 > tracing_on
 # cat trace > ~/nvme_complete_rq-trace.txt

# nvme error-log /dev/nvme0n1
 ```

and provide the logs? thanks

Hr46ph, 2023-08-11:
@igaw 

[Link to logs.](https://drive.proton.me/urls/TSJYHK45VG#cpTMYrPMhYzS)

Hr46ph, 2023-08-16:
> Oh, there is something else also using ftrace, thus it's a bit big. You might want to disable the other ftrace points first to reduce the log size. If this isn't problem with uploading, it's not a problem for me.

Can you help me do that? I literally copy/pasted your examples to get the logs. I have no idea how to find and disable other ftraces.

Hr46ph, 2023-08-16:
Uploading is not an issue and I have no idea how to disable other ftraces (sorry).

I can redo the logs its just that I'm confused as the command `echo 1 > events/nvme/enable` was in your original instructions.

Can you be more elaborate as to what you want me to do? I'll  be happy to do it.

Thanks!

igaw, 2023-08-17:
```shell
# cd /sys/kernel/debug/tracing/

# mkdir instances/nvme
# cd instances/nvme

# echo 1 > events/nvme/enable
# echo 1 > tracing_on

# nvme list

# echo 0 > tracing_on
# cat trace > ~/nvme_complete_rq-trace.txt

# nvme error-log /dev/nvme0n1
```

The log should have look similiar to this:

```
# cat ~/nvme_complete_rq-trace.txt 
# tracer: nop
#
# entries-in-buffer/entries-written: 201/201   #P:16
#
#                                _-----=> irqs-off/BH-disabled
#                               / _----=> need-resched
#                              | / _---=> hardirq/softirq
#                              || / _--=> preempt-depth
#                              ||| / _-=> migrate-disable
#                              |||| /     delay
#           TASK-PID     CPU#  |||||  TIMESTAMP  FUNCTION
#              | |         |   |||||     |         |
[...]
            nvme-16163   [012] .....  6381.126887: nvme_setup_cmd: nvme0: qid=0, cmdid=20, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=0, ctrlid=0)
          <idle>-0       [002] dNh1.  6381.132296: nvme_sq: nvme0: qid=0, head=30, tail=30
          <idle>-0       [002] .Ns1.  6381.132301: nvme_complete_rq: nvme0: qid=0, cmdid=20, res=0x0, retries=0, flags=0x2, status=0x0
            nvme-16163   [012] .....  6381.132330: nvme_setup_cmd: nvme0: qid=0, cmdid=21, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=3, ctrlid=0)
          <idle>-0       [002] dNh1.  6381.132365: nvme_sq: nvme0: qid=0, head=31, tail=31
          <idle>-0       [002] .Ns1.  6381.132366: nvme_complete_rq: nvme0: qid=0, cmdid=21, res=0x0, retries=0, flags=0x2, status=0x0
[...]
```

Looking at my logs, I see that we issue two `nvme_admin_identify` one with `cns=0` (Identify Namespace data structure) and `cns=3` (Namespace Identification Descriptor list).  Both of them are mandatory to be supported by the firmware (Figure 273: Identify – CNS Values, base spec)

Hr46ph, 2023-08-18:
Thanks! Much smaller file this time, is that due to that mkdir instances I presume?

[logfile](https://drive.proton.me/urls/YRR08N9DR8#FFtVTcCNvvRI)

igaw, 2023-08-18:
Yep, ftrace supports instances. I didn't know this either, so a clear win :)

Anyway, the interesting section from the log is:

```
            nvme-2688474 [019] ..... 154646.698628: nvme_setup_cmd: nvme0: qid=0, cmdid=12300, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=0, ctrlid=0)
          <idle>-0       [002] d.h1. 154646.706220: nvme_sq: nvme0: qid=0, head=24, tail=24
          <idle>-0       [002] ..s1. 154646.706228: nvme_complete_rq: nvme0: qid=0, cmdid=12300, res=0x0, retries=0, flags=0x2, status=0x0
            nvme-2688474 [019] ..... 154646.706258: nvme_setup_cmd: nvme0: qid=0, cmdid=12301, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=3, ctrlid=0)
          <idle>-0       [002] d.h1. 154646.706301: nvme_sq: nvme0: qid=0, head=25, tail=25
          <idle>-0       [002] ..s1. 154646.706303: nvme_complete_rq: nvme0: qid=0, cmdid=12301, res=0x0, retries=0, flags=0x2, status=0x2002
            nvme-2688474 [019] ..... 154646.706752: nvme_setup_cmd: nvme1: qid=0, cmdid=61467, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=0, ctrlid=0)
          <idle>-0       [017] d.h1. 154646.706823: nvme_sq: nvme1: qid=0, head=31, tail=31
          <idle>-0       [017] ..s1. 154646.706830: nvme_complete_rq: nvme1: qid=0, cmdid=61467, res=0x0, retries=0, flags=0x2, status=0x0
            nvme-2688474 [019] ..... 154646.706844: nvme_setup_cmd: nvme1: qid=0, cmdid=24, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=3, ctrlid=0)
          <idle>-0       [017] d.h1. 154646.706855: nvme_sq: nvme1: qid=0, head=0, tail=0
          <idle>-0       [017] ..s1. 154646.706856: nvme_complete_rq: nvme1: qid=0, cmdid=24, res=0x0, retries=0, flags=0x2, status=0x4002
            nvme-2688474 [019] ..... 154646.707150: nvme_setup_cmd: nvme2: qid=0, cmdid=28679, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=0, ctrlid=0)
          <idle>-0       [019] d.h1. 154646.707217: nvme_sq: nvme2: qid=0, head=31, tail=31
          <idle>-0       [019] ..s1. 154646.707219: nvme_complete_rq: nvme2: qid=0, cmdid=28679, res=0x0, retries=0, flags=0x2, status=0x0
            nvme-2688474 [019] ..... 154646.707225: nvme_setup_cmd: nvme2: qid=0, cmdid=32772, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=3, ctrlid=0)
          <idle>-0       [019] d.h1. 154646.707234: nvme_sq: nvme2: qid=0, head=0, tail=0
          <idle>-0       [019] ..s1. 154646.707234: nvme_complete_rq: nvme2: qid=0, cmdid=32772, res=0x0, retries=0, flags=0x2, status=0x4002
            nvme-2688474 [019] ..... 154646.707424: nvme_setup_cmd: nvme3: qid=0, cmdid=28700, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=0, ctrlid=0)
          <idle>-0       [001] d.h1. 154646.707494: nvme_sq: nvme3: qid=0, head=31, tail=31
          <idle>-0       [001] ..s1. 154646.707500: nvme_complete_rq: nvme3: qid=0, cmdid=28700, res=0x0, retries=0, flags=0x2, status=0x0
            nvme-2688474 [019] ..... 154646.707509: nvme_setup_cmd: nvme3: qid=0, cmdid=24605, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=3, ctrlid=0)
          <idle>-0       [001] d.h1. 154646.707520: nvme_sq: nvme3: qid=0, head=0, tail=0
          <idle>-0       [001] ..s1. 154646.707521: nvme_complete_rq: nvme3: qid=0, cmdid=24605, res=0x0, retries=0, flags=0x2, status=0x4002
            nvme-2688474 [019] ..... 154646.707712: nvme_setup_cmd: nvme4: qid=0, cmdid=20507, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=0, ctrlid=0)
          <idle>-0       [018] d.h1. 154646.707781: nvme_sq: nvme4: qid=0, head=31, tail=31
          <idle>-0       [018] ..s1. 154646.707788: nvme_complete_rq: nvme4: qid=0, cmdid=20507, res=0x0, retries=0, flags=0x2, status=0x0
            nvme-2688474 [019] ..... 154646.707797: nvme_setup_cmd: nvme4: qid=0, cmdid=32792, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=3, ctrlid=0)
          <idle>-0       [018] d.h1. 154646.707807: nvme_sq: nvme4: qid=0, head=0, tail=0
          <idle>-0       [018] ..s1. 154646.707809: nvme_complete_rq: nvme4: qid=0, cmdid=32792, res=0x0, retries=0, flags=0x2, status=0x4002
```

So the cns=3 (Namespace Identification Descriptor list) command is triggering the error.

Hr46ph, 2023-08-18:
So what happens next? Is it a bug and can it be fixed or is something wrong on my end?

keithbusch, 2023-08-18:
There's no bug, really. Some devices are just overly pedantic about logging harmless errors. In this case, the nvme-cli tool queried an optional identification and the device simply doesn't support that mode, which is fine: the tool doesn't need it to continue, but it is useful when available.

There is no protocol defined way for the host to know if the device supports it without trying it, so we try it and expect it might fail. The device in your case chooses to log this error, and it's rather unfortunate because many people find this behavior confusing.

Hr46ph, 2023-08-18:
Oke, let's close it then. Thanks both for looking into this and answering my questions :D.

igaw, 2023-08-19:
We started to add a quirk framework for libnvme-mi. I am wondering if we could make use of this also for nvme-cli. Thus I wanted to figure out where exactly nvme-cli/libnvme triggers the error.

Hr46ph, 2023-08-31:
@igaw Sorry I missed these notifications. I happened to stumble upon an old tab and it reloaded. Should I reopen it?

That command outputs a lot of errors while creating that tar. Is that expected?

```tar: /sys/devices/uncore_iio_1/power/runtime_active_time: File shrank by 4094 bytes; padding with zeros
tar: /sys/devices/uncore_iio_1/power/runtime_status: File shrank by 4084 bytes; padding with zeros
tar: /sys/devices/uncore_iio_1/power/autosuspend_delay_ms: Read error at byte 0, while reading 4096 bytes: Input/output error
tar: /sys/devices/uncore_iio_1/power/runtime_suspended_time: File shrank by 4094 bytes; padding with zeros
tar: /sys/devices/uncore_iio_1/power/control: File shrank by 4091 bytes; padding with zeros
tar: /sys/devices/uncore_iio_1/perf_event_mux_interval_ms: File shrank by 4094 bytes; padding with zeros
tar: /sys/devices/uncore_iio_1/type: File shrank by 4093 bytes; padding with zeros
tar: /sys/devices/uncore_iio_1/cpumask: File shrank by 4094 bytes; padding with zeros
tar: /sys/devices/uncore_iio_1/die0: File shrank by 4088 bytes; padding with zeros
tar: Exiting with failure status due to previous errors
```

igaw, 2023-09-01:
Yeah, tar is not so happy how sysfs is behaving. But this isn't a problem, tar should still create a proper tar archive with the content.

Let's look first at the tar file before reopening. It might just not help at all

Hr46ph, 2023-09-01:
Have a look if its useful, its here:
https://drive.proton.me/urls/FW4F4RNY8M#aruLD2xwovYd

let me know if you need anything else.

igaw, 2023-09-04:
D'oh, I didn't read the log correct. There are 5 nvme disk, hence the larger amount of `nvme_admin_identify` commands. So this part makes now sense to me.

Though what is really strange is, that you report that the error log contains status `0x4004` but he log says the failing command has a status of `0x2002`.

```
            nvme-2688474 [019] ..... 154646.706258: nvme_setup_cmd: nvme0: qid=0, cmdid=12301, nsid=1, flags=0x0, meta=0x0, cmd=(nvme_admin_identify cns=3, ctrlid=0)
          <idle>-0       [002] d.h1. 154646.706301: nvme_sq: nvme0: qid=0, head=25, tail=25
          <idle>-0       [002] ..s1. 154646.706303: nvme_complete_rq: nvme0: qid=0, cmdid=12301, res=0x0, retries=0, flags=0x2, status=0x2002
```

This doesn't really fit together.

keithbusch, 2023-09-04:
The error log entry is printed with the phase bit in the lower bit position, but the trace event shifts it out, hence the difference.

igaw, 2023-09-04:
Thanks for the explanation. So that means for this particular device we should not issue the  `Namespace Identification Descriptor list` command. Given we have another device which wants a quirk (avoid stack smashing thing) we might be able to address this issue as well. @keithbusch do you think this is a good idea?

keithbusch, 2023-09-04:
Sure, we have something similar in kernel because suspend/resume cycles would query the same identify and was blowing up the device's error logs. The kernel, though, just remembers the quirk at runtime only after seeing an error, and doesn't save across power cycles. For user space, I think we'd either need a quirk list in the binary or perhaps a vid:did list in some /etc/ file. I kinda like the file option since it can be modified automatically as programs execute and discover new devices needing the quirk rather than require a patch. We can also ship a default list of know devices in the source package too.

igaw, 2023-09-05:
Let's reopen this bug report and track the progress with it.

igaw, 2023-10-11:
We agreed during ALPSS to rely completely on `sysfs` for commands like `nvme list` etc. Thus we should do this here: https://github.com/linux-nvme/libnvme/issues/681

igaw, 2023-12-19:
With libnvme v1.7 and the upcoming kernel 6.8 this is fixed.

closed 2023-12-19 by igaw