linux-nvme/nvme-cli #2033: Endurance log missing Total Endurance Group Capacity (TEGCAP), Unallocated Endurance Group Capacity (UEGCAP)...

yalanliufio, 2023-08-16:
Please refer NVM Express Base Specification Revision 2.0c

Figure 217: Endurance Group Information Log Page

These fields are missing:
Bytes 01 Endurance Group Features (EGFEAT):
Bytes 07:06 Domain Identifier:
Bytes 175:160 Total Endurance Group Capacity (TEGCAP):
Bytes 191:176 Unallocated Endurance Group Capacity (UEGCAP):

Could you please take a look and add these fields so we can be ready to test nvme 2.0 SSD drive? Thanks

ikegami-t, 2023-08-18:
Could you please try to test the PR linux-nvme/libnvme#699 and #2035 changes?

yalanliufio, 2023-08-21:
I couldn't test at this moment as I don't have a drive support endurance group. Will do the test as soon as I have one. Thank you!

igaw, 2023-09-04:
Changes merged. Thanks @ikegami-t!

closed 2023-09-04 by igaw