linux-nvme/nvme-cli #2050: plugin is not displayed though I built it with json-c library

kyjin8, 2023-09-14:
Even though I built it with the json-c library, the plugin is not displayed.

```
meson setup --force-fallback-for=libnvme .build
meson compile -C .build
meson install -C .build
```

I already installed `libjson-c-dev` with `sudo apt install libjson-c-dev`, 
and confirmed that `get_option('json-c')` is enabled through debugging in `mseon.build`.

the result of `find / -name json-c`:
```
/usr/include/json-c
/usr/lib/x86_64-linux-gnu/cmake/json-c
```

When using `--force-fallback-for=libnvme,json-c`, the plugin was displayed. 
But I think it's normal that the build should work without adding json-c to force-fallback-for.
The contents below are my test environment and results. Please let me know if you need any more information.

```
```
> os : Ubuntu 22.04 LTS

the result of `meson setup --reconfigure --force-fallback-for=libnvme .build`:
```
The Meson build system
Version: 0.61.2
Source dir: /nvme-cli
Build dir: /nvme-cli/.build
Build type: native build
Project name: nvme-cli
Project version: 2.5
C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the host machine: cc ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Looking for a fallback subproject for the dependency libnvme because:
Use of fallback dependencies is forced.

Executing subproject libnvme 

libnvme| Project name: libnvme
libnvme| Project version: 1.5
libnvme| C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
libnvme| C linker for the host machine: cc ld.bfd 2.38
libnvme| C++ compiler for the host machine: c++ (gcc 11.4.0 "c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
libnvme| C++ linker for the host machine: c++ ld.bfd 2.38
libnvme| Did not find pkg-config by name 'pkg-config'
libnvme| Found Pkg-config: NO
libnvme| Did not find CMake 'cmake'
libnvme| Found CMake: NO
libnvme| Run-time dependency json-c found: NO (tried pkgconfig and cmake)
libnvme| Not looking for a fallback subproject for the dependency json-c because:
libnvme| Use of fallback dependencies is disabled.
libnvme| Run-time dependency openssl found: NO (tried pkgconfig and cmake)
libnvme| Not looking for a fallback subproject for the dependency openssl because:
libnvme| Use of fallback dependencies is disabled.
libnvme| Run-time dependency libkeyutils found: NO (tried pkgconfig and cmake)
libnvme| Checking if "__builtin_type_compatible_p" compiles: YES (cached)
libnvme| Checking if "typeof" compiles: YES (cached)
libnvme| Checking if "byteswap.h" compiles: YES (cached)
libnvme| Checking if "bswap64" : links: YES (cached)
libnvme| Checking if "statement-expr" compiles: YES (cached)
libnvme| Checking if "isblank" : links: YES (cached)
libnvme| Checking if "linux/mctp.h" compiles: YES (cached)
libnvme| Checking if "netdb" : links: YES (cached)
libnvme| Compiler for C supports function attribute fallthrough: YES (cached)
libnvme| Configuring libnvme.spec using configuration
[... 116 lines cut ...]
Installing /nvme-cli/subprojects/libnvme/src/libnvme-mi.h to /usr/local/include
Installing /nvme-cli/subprojects/libnvme/src/nvme/api-types.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/fabrics.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/filters.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/ioctl.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/linux.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/log.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/nbft.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/tree.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/types.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/util.h to /usr/local/include/nvme
Installing /nvme-cli/subprojects/libnvme/src/nvme/mi.h to /usr/local/include/nvme
Installing /nvme-cli/.build/meson-private/libnvme.pc to /usr/local/lib/x86_64-linux-gnu/pkgconfig
Installing /nvme-cli/.build/meson-private/libnvme-mi.pc to /usr/local/lib/x86_64-linux-gnu/pkgconfig
Installing /nvme-cli/completions/bash-nvme-completion.sh to /usr/local/share/bash-completion/completions
Installing /nvme-cli/completions/_nvme to /usr/local/share/zsh/site-functions
Installing /nvme-cli/.build/70-nvmf-autoconnect.conf to /usr/local/lib/dracut/dracut.conf.d
Installing /nvme-cli/.build/nvmefc-boot-connections.service to /usr/local/lib/systemd/system
Installing /nvme-cli/.build/nvmf-autoconnect.service to /usr/local/lib/systemd/system
Installing /nvme-cli/.build/nvmf-connect.target to /usr/local/lib/systemd/system
Installing /nvme-cli/.build/nvmf-connect@.service to /usr/local/lib/systemd/system
Installing /nvme-cli/.build/70-nvmf-autoconnect.rules to /usr/local/lib/udev/rules.d
Installing /nvme-cli/.build/71-nvmf-iopolicy-netapp.rules to /usr/local/lib/udev/rules.d
Installing /nvme-cli/.build/discovery.conf to /usr/local/etc/nvme
```

closed 2023-09-14 by igaw

kyjin8, 2023-09-15:
Oh it works now.. Thank you :)

kyjin8, 2023-09-15:
But I tried installing `cmake` and it didn't work. Could you tell me about the reasons for this?
(it works well with installing pkg-config)

This is the result of `meson --reconfigure --force-fallback-for=libnvme .build`:
```
The Meson build system
Version: 0.61.2
Source dir: /nvme-cli-carrera
Build dir: /nvme-cli-carrera/.build
Build type: native build
Project name: nvme-cli
Project version: 2.5
C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the host machine: cc ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Looking for a fallback subproject for the dependency libnvme because:
Use of fallback dependencies is forced.

Executing subproject libnvme 

libnvme| Project name: libnvme
libnvme| Project version: 1.5
libnvme| C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
libnvme| C linker for the host machine: cc ld.bfd 2.38
libnvme| C++ compiler for the host machine: c++ (gcc 11.4.0 "c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
libnvme| C++ linker for the host machine: c++ ld.bfd 2.38
libnvme| Did not find pkg-config by name 'pkg-config'
libnvme| Found Pkg-config: NO
libnvme| Found CMake: /usr/bin/cmake (3.22.1)
libnvme| Run-time dependency json-c found: NO (tried pkgconfig and cmake)
libnvme| Not looking for a fallback subproject for the dependency json-c because:
libnvme| Use of fallback dependencies is disabled.
libnvme| Run-time dependency openssl found: NO (tried pkgconfig and cmake)
libnvme| Not looking for a fallback subproject for the dependency openssl because:
libnvme| Use of fallback dependencies is disabled.
libnvme| Run-time dependency libkeyutils found: NO (tried pkgconfig and cmake)
libnvme| Checking if "__builtin_type_compatible_p" compiles: YES (cached)
libnvme| Checking if "typeof" compiles: YES (cached)
libnvme| Checking if "byteswap.h" compiles: YES (cached)
libnvme| Checking if "bswap64" : links: YES (cached)
libnvme| Checking if "statement-expr" compiles: YES (cached)
libnvme| Checking if "isblank" : links: YES (cached)
libnvme| Checking if "linux/mctp.h" compiles: YES (cached)
libnvme| Checking if "netdb" : links: YES (cached)
libnvme| Compiler for C supports function attribute fallthrough: YES (cached)
libnvme| Configuring libnvme.spec using configuration
libnvme| Configuring config.h using configuration
libnvme| Program python3 found: YES (/usr/bin/python3)
libnvme| WARNING: Python files installed by Meson might not be found by python interpreter.
libnvme| This warning can be avoided by setting "python.platlibdir" option.
libnvme| WARNING: Python files installed by Meson might not be found by python interpreter.
libnvme| This warning can be avoided by setting "python.purelibdir" option.
libnvme| Program swig found: NO
libnvme| Has header "Python.h" with dependency python: YES (cached)
libnvme| Configuring nbft-dump-diff.sh using configuration
libnvme| Configuring gen-nbft-diffs.sh using configuration
libnvme| Program diff found: YES (/usr/bin/diff)
libnvme| Configuring conf.py using configuration
libnvme| Configuring api.rst using configuration
libnvme| Configuring index.rst using configuration
libnvme| Configuring quickstart.rst using configuration
libnvme| Configuring installation.rst using configuration
libnvme| Configuring mi.rst using configuration
libnvme| Configuring config-schema.json using configuration
libnvme| Program /nvme-cli-carrera/subprojects/libnvme/doc/../scripts/kernel-doc found: YES (/nvme-cli-carrera/subprojects/libnvme/doc/../scripts/kernel-doc)
libnvme| Program /nvme-cli-carrera/subprojects/libnvme/doc/../scripts/kernel-doc-check found: YES (/nvme-cli-carrera/subprojects/libnvme/doc/../scripts/kernel-doc-check)
libnvme| Build targets in project: 23
libnvme| Subproject libnvme finished.

[... 56 lines cut ...]
    build location    : /nvme-cli-carrera/.build

  Dependencies
    json-c            : False
    libhugetlbfs      : False

  Configuration
    pdc enabled       : False

  Subprojects
    libnvme           : YES 2 warnings

  User defined options
    force_fallback_for: libnvme

Found ninja-1.10.1 at /usr/bin/ninja
```

and this is the result of `apt list --installed | grep cmake`
```
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

cmake-data/jammy-updates,now 3.22.1-1ubuntu1.22.04.1 all [installed,automatic]
cmake/jammy-updates,now 3.22.1-1ubuntu1.22.04.1 amd64 [installed]
```

igaw, 2023-09-15:
I think the problem is that the cmake method is not able to figure out the version of the library. At least this is the case on my system with

```meson
project('t')

json_c_dep = dependency('json-c',
                        version: '>=0.13',
                        required: true,
                        method: 'cmake')
```

```
Calling CMake (['/usr/bin/cmake']) in /home/wagi/tmp/.build/meson-private/cmake_json-c with:
  - "-DNAME=json-c"
  - "-DARCHS="
  - "-DVERSION="
  - "-DCOMPS="
  - "-DSTATIC=False"
  - "--trace-expand"
  - "--trace-format=json-v1"
  - "--no-warn-unused-cli"
  - "--trace-redirect=cmake_trace.txt"
  - "-DCMAKE_TOOLCHAIN_FILE=/home/wagi/tmp/.build/meson-private/cmake_json-c/CMakeMesonToolchainFile.cmake"
  - "."
Guessed CMake target 'json-c::json-c'
CMake TARGET:
  -- name:      json-c::json-c
  -- type:      SHARED
  -- imported:  True
  -- properties: {
      'INTERFACE_INCLUDE_DIRECTORIES': ['/usr/include', '/usr/include/json-c']
      'IMPORTED_CONFIGURATIONS': ['NONE']
      'IMPORTED_LOCATION_NONE': ['/usr/lib64/libjson-c.so.5.2.0']
      'IMPORTED_SONAME_NONE': ['libjson-c.so.5']
     }
  -- tline: CMake TRACE: /usr/lib64/cmake/json-c/json-c-targets.cmake:53 add_library(['json-c::json-c', 'SHARED', 'IMPORTED'])
Include Dirs:         ['/usr/include', '/usr/include/json-c']
Compiler Options:     []
Libraries:            ['/usr/lib64/libjson-c.so.5.2.0']
Dependency json-c found: NO unknown version, but need: ['>=0.13']
Dependency lookup for json-c with method 'cmake' failed: Unknown version, but need ['>=0.13'].
Run-time dependency json-c found: NO 
```

That just means that the supported way is to use `pkg-config` and not `cmake` on these distributions.

kyjin8, 2023-09-18:
I also set the same settings on my system and saw that the log came out similar to yours.
Then, I will configure it to build with pkg-config and use it. Thanks for the reply!