linux-nvme/nvme-cli #2053: Why can’t I find the KV command set in the nvme-cli-v2.5?

cljglll, 2023-09-20:
Is it not supported?

keithbusch, 2023-09-20:
No one contributed it. I am not aware of anyone who even has access to a production level implementation, but if someone does and wants to develop the '`kv`' command set extension, sounds good; key-value is an NVMe TWG supported command set, so it's welcome here even if it was a bad idea.

Linux kernel exports generic namespaces (ex, /dev/ng0n1), so driver kv support is in place to handle commands via the low level "`nvme io-passthru`' command, but a '`kv`' extension would certainly be more user friendly.

cljglll, 2023-09-20:
> No one contributed it. I am not aware of anyone who even has access to a production level implementation, but if someone does and wants to develop the '`kv`' command set extension, sounds good; [...]

So it is privately owned by the manufacturer of SSD. Thank you for your reply.

closed 2023-09-20 by cljglll