linux-nvme/nvme-cli #2061: weird encoding on output of `nvme smart-log`

solsticedhiver, 2023-09-24:
With nvme-cli 2.5 (on archlinux x86_64), the output of `nvme smart-log` in french locale is weird in a utf-8 terminal.

```
$ sudo nvme smart-log /dev/nvme0
Smart Log for NVME device:nvme0 namespace-id:ffffffff
critical_warning			: 0
temperature				: 24 °C (297 K)
available_spare				: 100%
available_spare_threshold		: 10%
percentage_used				: 10%
endurance group critical warning summary: 0
Data Units Read				: 244���99797 (12,54 TB)
Data Units Written			: 395���50889 (20,25 TB)
host_read_commands			: 4���949���14710
host_write_commands			: 7���078���73386
controller_busy_time			: 10380
power_cycles				: 2571
power_on_hours				: 14170
unsafe_shutdowns			: 172
media_errors				: 0
num_err_log_entries			: 0
Warning Temperature Time		: 0
Critical Composite Temperature Time	: 0
Thermal Management T1 Trans Count	: 0
Thermal Management T2 Trans Count	: 0
Thermal Management T1 Total Time	: 0
Thermal Management T2 Total Time	: 0

$ echo $LANG
fr_FR.UTF-8
```

the same output in C locale:
```
$ LANG=C sudo nvme smart-log /dev/nvme0
Smart Log for NVME device:nvme0 namespace-id:ffffffff
critical_warning			: 0
temperature				: 23 °C (296 K)
available_spare				: 100%
available_spare_threshold		: 10%
percentage_used				: 10%
endurance group critical warning summary: 0
Data Units Read				: 24499804 (12.54 TB)
Data Units Written			: 39550965 (20.25 TB)
host_read_commands			: 494914925
host_write_commands			: 707874428
controller_busy_time			: 10380
power_cycles				: 2571
power_on_hours				: 14170
unsafe_shutdowns			: 172
media_errors				: 0
num_err_log_entries			: 0
Warning Temperature Time		: 0
Critical Composite Temperature Time	: 0
Thermal Management T1 Trans Count	: 0
Thermal Management T2 Trans Count	: 0
Thermal Management T1 Total Time	: 0
Thermal Management T2 Total Time	: 0
```

igaw, 2023-09-25:
As far I understand the fr_FR.UTF-8 lcoale, it the thousand separate is a single space. What do you get as output for this:

`LC_ALL=fr_FR.UTF-8 awk 'BEGIN{printf "%s -> %\047i\n", ENVIRON["LC_ALL"], 13389}'`

?

solsticedhiver, 2023-09-25:
I got:
```
$ LC_ALL=fr_FR.UTF-8 awk 'BEGIN{printf "%s -> %\047i\n", ENVIRON["LC_ALL"], 13389}'
fr_FR.UTF-8 -> 13 389
```
python babel is quite good with those locale formatting, I think if you need it:

```python
>>> from babel.numbers import format_decimal
>>> format_decimal(12345.678, locale='de_DE')
'12.345,678'
>>> format_decimal(12345.678, locale='fr_FR')
'12\u202f345,678'
```

igaw, 2023-09-25:
Ah I see, so the problem is that in your locale settings, the separator is `\202f` which is a Narrow No-Break Space according:

  https://www.compart.com/en/unicode/U+202F 

I assume Python is updating the encoding because this seems to be the UTF-16 encoding. But this is besides the point. 

I suspect that `strlen' is not handling the UTF econding correctly. Need to check some docs on this topic.

https://github.com/linux-nvme/nvme-cli/blob/cbcd3187e0746e712429892daf5199de46944777/util/types.c#L72C1-L84
```c
        if (l10n) {
                sep = localeconv()->thousands_sep;
                len = strlen(sep);
        }

        /* terminate at the end, and build up from the ones */
        str[--idx] = '\0';

        do {
                if (len && !((sizeof(str) - idx) % (3 + len))) {
                        for (i = 0; i < len; i++)
                                str[--idx] = sep[i];
                }
```

solsticedhiver, 2023-09-25:
My C knowledge is limited but to use unicode or stuff like you need wchar_t ? and wcslen function to get the length ?

I can't help more than that

igaw, 2023-09-25:
In this case it's just the copy loop is not aware that it can be multiple characters.

PR #2062 (2023-09-25): Fix utf-8 encoded number strings
The separator char can be an utf-8 encoded symbol thus the length of the string is necessarly one.

Fix: #2061
closed 2023-09-25 by igaw