linux-nvme/nvme-cli #2063: How can I set PRP1 with admin-passthru command?

jnegara, 2023-09-25:
I'm trying to create a submission queue via the following admin-passthru command:

```
nvme admin-passthru /dev/nvme1n1 --show-command --opcode=1 --cdw10=$CDW10 --cdw11=$CDW11 --write --data-len=8 <<< 0xdeadbeefbabecafe
```

From the device logs, I see that CDW10, and CDW11 are being correctly populated.

PRP1 on the other hand... 

I've tried everything. I've tried sending the PRP as hex, decimal, and 8 byte binary representations. I've tried using the --input-file option. I've even tried sending it via metadata.

Still, I see PRP1 set to some non 0 value that is definitely not deadbeefbabecafe.

How does this command set PRP1? What is the expected usage?

Thanks!

keithbusch, 2023-09-25:
You can't set it from user space. You can only set the virtual address. The driver has to convert that to physical address(es) which becomes your PRP fields.

jnegara, 2023-09-25:
So I'm not supposed to pass in the payload of the PRP, but an address that points to the PRP? Is there any chance you could show me an example of what this looks like? Which option do I pass this address through?

keithbusch, 2023-09-25:
You don't deal with PRPs from user space at all. You provide virtual addresses, and you tell the kernel about it in the ioctl struct's 'addr'. In you example, you told the tool  to allocate 8 bytes and you filled those 8 bytes with that hex string. The tool will allocate an 8 bytes user space virtual buffer, and set the 'addr' field to that. The kernel will the turn that into a PRP later.

jnegara, 2023-09-25:
> In you example, you told the tool to allocate 8 bytes and you filled those 8 bytes with that hex string. The tool will allocate 8 bytes user space virtual buffer, and set the 'addr' field to that. 

I think this was my understanding. I'm not worried about sending the PRP itself, I just want to set the data that will populate the PRP. In the example I shared I would expect the driver to receive a submission queue entry in which PRP1 is set to `0xdeadbeefbabecafe` - is this expectation wrong? If so, how would I use the CLI to do this?

I want to issue a create I/O submission queue command, which requires that PRP1 is set to the base pointer of the new submission queue. I see that the driver is receiving correct values for all of the command dwords, but not the PRP.

keithbusch, 2023-09-25:
Your example should be filling the contents of the address with "0xdeadbe" because you requested 8 bytes. The rest of you string will be truncated. The value of PRP1 will be the physical address of the user space virtual address that was allocated to hold that truncated string.

You do not get to choose the value of PRP1. You can only set what contents PRP1 points to.

keithbusch, 2023-09-26:
Side note, not sure what your goal is with trying to create io queues from this utility. You won't be able to use it. The memory for the queue from this the command will be freed immediately after the command finishes.

closed 2023-12-08 by igaw