linux-nvme/nvme-cli #2068: Create-ns does not work

rsajwani, 2023-09-27:
I just recently start exploring Nvme. I setup a VM on MS azure with SSD (Nvme conroller).. Here are some details of the Vm

```
azureuser@vm-1:~$ nvme -version
nvme version 1.16 
```

```
azureuser@vm-1:~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0          7:0    0  63.5M  1 loop /snap/core20/2015
loop1          7:1    0  40.8M  1 loop /snap/snapd/20092
loop2          7:2    0 111.9M  1 loop /snap/lxd/24322
sda            8:0    0    75G  0 disk
└─sda1         8:1    0    75G  0 part /mnt
nvme0n1      259:0    0    30G  0 disk
├─nvme0n1p1  259:1    0  29.9G  0 part /
├─nvme0n1p14 259:2    0     4M  0 part
└─nvme0n1p15 259:3    0   106M  0 part /boot/efi
```

Then I tried to list

```
azureuser@vm-1:~$ sudo nvme -list
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          SN: 000001           MSFT NVMe Accelerator v1.0               1          32.21  GB /  32.21  GB    512   B +  0 B   v1.00000
```

```
azureuser@vm-1:~$ sudo nvme -list-ns /dev/nvme0n1
[   0]:0x1
```

```
azureuser@vm-1:~$ sudo nvme id-ctrl /dev/nvme0

cntlid    : 0
ver       : 0x10400
rtd3r     : 0x61a80
rtd3e     : 0x61a80
oaes      : 0x100
ctratt    : 0
rrls      : 0
cntrltype : 1
fguid     :
crdt1     : 0
```

But when I try to run create-ns or delete-ns , it give me error

```
azureuser@vm-1:~$ sudo nvme create-ns /dev/nvme0 -f 0 -s 2560 -c 2560
NVMe status: INVALID_OPCODE: The associated command opcode field is not valid(0x1)
```

```
azureuser@vm-1:~$ sudo nvme create-ns /dev/nvme0 -b 1 -s 2560 -c 2560
identify failed
NVMe status: INVALID_NS: The namespace or the format of that namespace is invalid(0xb)
```

Can anyone help , if I am doing something wrong ...

keithbusch, 2023-09-28:
Your device doesn't support dynamic namespace management. That's ok, very few devices support it.

rsajwani, 2023-09-28:
how do I find out if device does not support dynamic management. Is it a firmware level thing or something else ..

keithbusch, 2023-09-28:
It's a firmware feature. Support is reported in Identity Controller (id-ctrl) in the OACS field.

igaw, 2023-09-28:
Isn't namespace create/deletion done via of the Azure web interface? I doubt this is supported via the host.

rsajwani, 2023-09-28:
The device is oacs 2 ... seems like there is no support for create/deletion via azure web interface .. Thanks , for the responses ..

closed 2023-09-28 by rsajwani