linux-nvme/nvme-cli #2082: "nvme list-subsys -o json" command output provides special character in it.

Abhijit2211, 2023-10-10:
The "nvme list-subsys -o json" command provides the special character in the output on RHEL 8.6  

[root@svm ~]# nvme list-subsys -o json
--------------------------------------------------
json@{ "Subsystems" : [ { "Name" : "nvme-subsys0", "NQN" : "HPE MS10i-p Gen10+ Boot Controller ", "Paths" : [ { "Name" : "nvme0", "Transport" : "pmie", "Address" : "0000:11:00.0", "State" : "live" } ] } ] }
--------------------------------------------------
in the above output, an unexpected character has come after 'HPE MS10i-p Gen10+ Boot Controller'.

Que:
Can anyone please let me know why the special/unexpected character comes in the nvme output.

Abhijit2211, 2023-10-11:
I am using "nvme-cli-1.16-3.el8.x86_64" version.
Is there any issue with this version for the "nvme list-subsys -o json" command?

igaw, 2023-10-11:
IIRC, nvme-cli 1.16 lacks all the NQN whitespace cleanup fixes we have in the 2.x branch.

The version you are using is maintained by RH, thus you can report this bug to RH and see if they are pulling the fixes from upstream. 

BTW, you can try our latest greatest upstream code as AppImage:  https://oxygen.monom.org/linux-nvme/

```shell
$ wget https://monom.org/linux-nvme/upload/AppImage/nvme-cli-latest-x86_64.AppImage
$ chmod +x nvme-cli-latest-x86_64.AppImage
$ ./nvme-cli-latest-x86_64.AppImage list-subsys -o json
```

If it still see this problem, we have a bug in upstream too.

igaw, 2023-11-17:
No feedback. Closing.

closed 2023-11-17 by igaw