linux-nvme/nvme-cli #2100: Group for nvme access

m-kru, 2023-10-21:
Is there any group that user can be added to so that there is no requirement for sudo/root access?

ikegami-t, 2023-10-29:
Seems no group mentioned as tried to add root, bin and sys groups for my local ubuntu account but still failed to execute the nvme command without sudo and root access. This may be resolved by setting `/etc/sudoers` instead but not sure.

igaw, 2023-11-02:
Most of the commands will depend on `root` access on the `/dev/nvme*` devices. We are working on getting at least `nvme list` (the read operations) not depending on write access to the device nodes. But I'd say this tool is for `root` only (except the `nvme list` operation).

closed 2023-11-02 by igaw