linux-nvme/nvme-cli #2104: extra byte returned in stdout from set-feature

tkenbo, 2023-10-30:
We encountered this issue when we are sending set feature command with feature id 129 (host-id).
Not 100% occurrence but it happened frequently.
```
[root@de-r650-008 ~]# nvme set-feature /dev/nvme1 -f 129 -v 0 -d ~/host-id.dat | hexdump -C
00000000  73 65 74 2d 66 65 61 74  75 72 65 3a 30 78 38 31  |set-feature:0x81|
00000010  20 28 48 6f 73 74 20 49  64 65 6e 74 69 66 69 65  | (Host Identifie|
00000020  72 29 2c 20 76 61 6c 75  65 3a 30 30 30 30 30 30  |r), value:000000|
00000030  30 30 2c 20 63 64 77 31  32 3a 30 30 30 30 30 30  |00, cdw12:000000|
00000040  30 30 2c 20 73 61 76 65  3a 30 0a 20 20 20 20 20  |00, save:0.     |
00000050  20 20 30 20 20 31 20 20  32 20 20 33 20 20 34 20  |  0  1  2  3  4 |
00000060  20 35 20 20 36 20 20 37  20 20 38 20 20 39 20 20  | 5  6  7  8  9  |
00000070  61 20 20 62 20 20 63 20  20 64 20 20 65 20 20 66  |a  b  c  d  e  f|
00000080  0a 30 30 30 30 3a 20 66  66 20 66 66 20 66 66 20  |.0000: ff ff ff |
00000090  66 66 20 66 66 20 66 66  20 66 66 20 66 66 20 20  |ff ff ff ff ff  |
000000a0  20 20 20 20 20 20 20 20  20 20 20 20 20 20 20 20  |                |
000000b0  20 20 20 20 20 20 20 20  22 2e 2e 2e 2e 2e 2e 2e  |        ".......|
000000c0  2e a0 22 0a                                       |..".|
000000c4
```
In this case, host-id.dat contains only 8 bytes
```
[root@de-r650-008 ~]# hexdump host-id.dat
0000000 ffff ffff ffff ffff
0000008
```
So looking at the set-feature stdout, ascii side of data is printed starting at byte 0xb9 "2e 2e 2e 2e 2e 2e 2e 2e a0", one extra byte is added. 
Looking at set_feature function.  function d in nvme-print.c is called.
`d(buf, cfg.data_len, 16, 1);` 
I looked at the d function and found there might be an extra byte when len is not multiple of width.
When that happens, the following code is executed.
```
	if (!line_done) {
		unsigned b = width - (i % width);
		ascii[i % width + 1] = '\0';
		printf( " %*s \"%.*s\"",
				2 * b + b / group + (b % group ? 1 : 0), "",
				width, ascii);
	}
```
Here I wonder if `ascii[i % width + 1] = '\0'` shouldn't be `ascii[i % width] = '\0'`?
I tested with my custom build with the change.

kernel we encountered this issue: 
4.18.0-477.15.1.el8_8.x86_64
5.14.0-284.30.1.el9_2.x86_64
nvmecli: 2.2.1 (1.2)

Please advise.

ikegami-t, 2023-10-31:
Just created the PR #2107 to fix this issue.

igaw, 2023-11-02:
I've merged @ikegami-t #2107 PR.

closed 2023-11-02 by igaw

tkenbo, 2023-11-02:
thanks @ikegami-t @igaw for fixing it quickly.