linux-nvme/nvme-cli #2105: admin-passthru data_len: Invalid Argument

alexeyponomarev, 2023-10-30:
We're heavily utilizing the admin-passthru commands in our test framework including the "data_len" (-l) parameter calculating it as (NUMDU+NUMDL)*4 for our getLogPage requests via admin-passthru.

We previously used this only on the drives with MDTS=5 (128KB max transfer size) and we noticed that the data_len can be only <=MDTS, otherwise we'd get "Invalid Argument" from the tool:

```
// drive with MDTS=128K
nvme id-ctrl /dev/nvme0 | grep mdts
mdts      : 5
nvme show-regs /dev/nvme0 | grep "^cap"
cap     : 1000303c030fff

// getLogPage to LID 2 with 0x8000 NUMD, set data_len to 131072 (128KB) - passing
nvme admin-passthru --opcode 0x02 --cdw10=0x80000002  -l 131072 /dev/nvme0 -b -i /tmp/out.bin
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x4002)

// getLogPage to LID 2 with 0x8000 NUMD, set data_len to 132096 (129KB) - fail (expected, data_len > MDTS)
nvme admin-passthru --opcode 0x02 --cdw10=0x80000002  -l 132096 /dev/nvme0 -b -i /tmp/out.bin
passthru: Invalid argument
```

However with newer drives with MDTS=1MB we get Invalid Argument from the tool even on commands with data_len less than MDTS.

Issue 1 - the tool seems to _require_ libhugetlbfs when data_len >=512K, is this correct? There's no warning for the parameter limits in the man/help, but without libhugetlbfs any requests we make that are >=512K in data_len are failing with Invalid Argument
```
// drive with MDTS=1MB
nvme id-ctrl /dev/nvme1 | grep mdts
mdts      : 8
nvme show-regs /dev/nvme1 | grep "^cap"
cap     : f0003014023fff

// getLogPage to LID 2 with 0x20000 NUMD, set data_len to 524288 (512KB) - fail
nvme admin-passthru --opcode 0x02 --cdw10=0x2 --cdw11=0x2  -l 524288 /dev/nvme1 -b -i /tmp/out.bin
passthru: Invalid argument

mmap(NULL, 532480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fd3a8264000
ioctl(3, NVME_IOCTL_ADMIN_CMD, 0x7ffedf389260) = -1 EINVAL (Invalid argument)
write(2, "passthru: Invalid argument", 26passthru: Invalid argument) = 26
newfstatat(1, "", {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0), ...}, AT_EMPTY_PATH) = 0
write(1, "\n", 1
)                       = 1
munmap(0x7fd3a8264000, 532480)          = 0
```
Issue 2 - even after installing the necessary packets and rebuilding nvme-cli with libhugetlbfs support - there're certain values in 509KB-511KB range that are always failing (everything <=508KB and >=512KB is fine):

```
./.build/nvme admin-passthru --opcode 0x02 --cdw10=2  -l 519168 /dev/nvme1 -b -i /tmp/out.bin
Admin Command Get Log Page is Success and result: 0x00000000
./.build/nvme admin-passthru --opcode 0x02 --cdw10=2  -l 520192 /dev/nvme1 -b -i /tmp/out.bin
Admin Command Get Log Page is Success and result: 0x00000000
./.build/nvme admin-passthru --opcode 0x02 --cdw10=2  -l 521216 /dev/nvme1 -b -i /tmp/out.bin
passthru: Invalid argument
./.build/nvme admin-passthru --opcode 0x02 --cdw10=2  -l 522240 /dev/nvme1 -b -i /tmp/out.bin
passthru: Invalid argument
./.build/nvme admin-passthru --opcode 0x02 --cdw10=2  -l 523264 /dev/nvme1 -b -i /tmp/out.bin
passthru: Invalid argument
./.build/nvme admin-passthru --opcode 0x02 --cdw10=2  -l 524288 /dev/nvme1 -b -i /tmp/out.bin
Admin Command Get Log Page is Success and result: 0x00000000
./.build/nvme admin-passthru --opcode 0x02 --cdw10=2  -l 525312 /dev/nvme1 -b -i /tmp/out.bin
Admin Command Get Log Page is Success and result: 0x00000000
```

Environment:

- nvme-cli 2.6
- CentOS 9
- Kernel vanilla 6.1.47

Thank you in advance for your feedback here!

keithbusch, 2023-10-31:
Oo, you've stepped into a doozy ....

There are some kernel limitations for the driver that require it have no more than 128 physical segments (It was 127 segments in 6.1, but it's 128 today). On a system with 4k page size (all x86 platforms), if your user space buffer maps to perfectly discontinuous pages, that's 128 segments, and over the limit in 6.1.

By somewhat random chance, depending on the state of your system memory fragmentation, you very well may get some or all pages to be physically contiguous and stay under the segment limit for even larger transfers, but it's certainly not reliable, even when advising THP's.

Hugetlbfs is unfortunately the only way to guarantee we can allocate from user space within the kernel's required segment limits when you need that large of a buffer, so this tool tries to use that when available. Looking back, though, libhugetlbfs may not have been the best library dependency to introduce.

alexeyponomarev, 2023-11-01:
@keithbusch thank your for detailed explanation here, much appreciated.

So for "issue 1" we'll just use libhugetlbfs across all of our systems, thanks for confirming. Not sure if the tool can/need to warn the user that the command might fail otherwise if large buffer request is detected?

For "issue 2" above - 508-511KB buffer requests failing even with libhugetlbfs support: this is because of the 127 segments requirement in 6.1 vs. 128 segments now? I've rebuilt with "#define HUGE_MIN 0x7F000" instead of 0x80000 and all request sizes from 0 to MDTS are passing now.

igaw, 2023-11-02:
Not sure if my idea is clever or just plain stupid. But we could make the logic when to use `libhugetlbfs` a bit more dynamic or at least possible to stear from the command line.

keithbusch, 2023-11-02:
> For "issue 2" above - 509-511KB buffer requests failing even with libhugetlbfs support: this is because of the 127 segments requirement in 6.1 vs. 128 segments now? I've rebuilt with "#define HUGE_MIN 0x7F000" instead of 0x80000 and all request sizes from 0 to MDTS are passing now.

Yes, that's correct. We're off by one page on the hugepage threshold size for the worst case scenario.

keithbusch, 2023-11-02:
> Not sure if my idea is clever or just plain stupid. But we could make the logic when to use `libhugetlbfs` a bit more dynamic or at least possible to stear from the command line.

We should probably just remove the libhugetlbfs dependency and use something a little more common: mmap with MAP_ANONYMOUS and MAP_HUGETLB. If that fails, then just use madvise and hope for the best?

If you want to just lower the hugepage size threshold, that might be okay to. Not sure that we want to let this be user configurable per command, though.

igaw, 2023-11-02:
I am in favor to drop the libhugetlbfs dependency. And on second thought having size knobs as workaround is a bit silly.

PR #2128 (2023-12-01): nvme: replace libhugetlbfs with mmap and madvise
Instead depending on libhugetlbfs for large memory allocation mapping
just use mmap and madvise for trying to allocate continuous memory.

Only lightly tested. The small allocation works fine, though I am not so sure 
about the huge allocation part. @keithbusch is that what you had in mind?

Fixes: #2105
closed 2023-12-01 by igaw