linux-nvme/nvme-cli #2106: nvme id-ctrl output keys documentation

m-kru, 2023-10-31:
The `nvme id-ctrl` output has a very simple structure `key : value`. However, as the acronyms or abbreviations are used as the keys, sometimes it is really hard to get to know what the key really means. I know there is an extra `-H` flag that can be provided that makes the output more human-readable. However, a lot of keys are still enigmatic even with `-H`. `nvme help id-ctrl` neither provides any information on the keys meanings nor on where to find more information.  

Where can user find an information what the keys actually mean?

I think it would be also valuable to improve the `nvme help id-ctrl` output with one sentence pointing to the documentation of keys.

keithbusch, 2023-10-31:
The names of the structure fields come from the spec, available at nvmexpress.org.

m-kru, 2023-10-31:
Do you think thins information can be added to the `nvme help id-ctrl` output?

igaw, 2023-11-02:
Sure, this is possible, it's just a question who is willing to spend the time to add *and* maintain help texts. It's already difficult to keep up with the spec changes and get them added. I'll don't have the bandwidth to do something like this.

closed 2023-11-02 by igaw