linux-nvme/nvme-cli #2108: nvme/plugins: nvme-cli 1.16x version issue

jinhuahuang2021, 2023-11-02:
Does nvme-cli release new version after v1.16?
I found some Linux system (CentOS7, CentOS8...) still use nvme-cli v.1.16.
There is little hard to build nvme-cli 2.x on CentOS 7.

igaw, 2023-11-02:
There is not active development on the 1.x branch anymore. But if there are build problems etc, I don't see any problems to release a new version. But don't expect features. What is the problem?

(And what is the problem to build 2.x? See scripts/build.sh for different build configuration including samu/muon which avoids the meson python dependencies)

jinhuahuang2021, 2023-11-03:
Got, thank you.

ikegami-t, 2023-11-04:
Tried to install on CentOS 7 by muon but failed as below. Also to resolve the failure tried to install latest gcc but failed to configure since `isl` not installed and could not find to install it also.
[build-muon.log](https://github.com/linux-nvme/nvme-cli/files/13256699/build-muon.log)
```sh
[root@localhost nvme-cli]# ./scripts/build.sh -m muon 
which: no samu in (/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/root/bin)
Cloning into '/root/nvme-cli/.build-ci/build-tools/samurai'...
remote: Enumerating objects: 40, done.
...
~/nvme-cli/.build-ci/build-tools/samurai ~/nvme-cli
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o build.o build.c
...
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o util.o util.c
gcc  -o samu build.o deps.o env.o graph.o htab.o log.o parse.o samu.o scan.o tool.o tree.o util.o
~/nvme-cli
which: no muon in (/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/root/bin)
Cloning into '/root/nvme-cli/.build-ci/build-tools/muon'...
remote: Enumerating objects: 2270, done.
...
Resolving deltas: 100% (158/158), done.
~/nvme-cli/.build-ci/build-tools/muon ~/nvme-cli
+ dir=stage1
+ mkdir -p stage1
+ pkgconf_cmd=
+ command -v pkgconf
+ command -v pkg-config
+ pkgconf_cmd=pkg-config
+ '[' -n pkg-config ']'
+ pkg-config libpkgconf
+ pkgconf_src=libpkgconf_null.c
+ pkgconf_cflags=
+ pkgconf_libs=
+ gcc -Iinclude src/amalgam.c -o stage1/muon
In file included from src/amalgam.c:46:0:
src/functions/dependency.c: In function 'dep_pkgconfig_define':
src/functions/dependency.c:88:2: error: 'for' loop initial declarations are only allowed in C99 mode
  for (int64_t idx = 0; idx < arraylen; idx += 2) {
  ^
src/functions/dependency.c:88:2: note: use option -std=c99 or -std=gnu99 to compile your code
```

ikegami-t, 2023-11-04:
Failed on CentOS 8 also as below as `muon/subprojects/libcurl not found`.
[build-muon-centos-8.log](https://github.com/linux-nvme/nvme-cli/files/13256887/build-muon-centos-8.log)
```sh
[root@linux nvme-cli]# ./scripts/build.sh -m muon
/usr/bin/which: no samu in (/usr/share/Modules/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/root/bin)
Cloning into '/root/nvme-cli/.build-ci/build-tools/samurai'...
remote: Enumerating objects: 40, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (37/37), done.
remote: Total 40 (delta 0), reused 21 (delta 0), pack-reused 0
Receiving objects: 100% (40/40), 38.37 KiB | 413.00 KiB/s, done.
~/nvme-cli/.build-ci/build-tools/samurai ~/nvme-cli
gcc -O -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o build.o build.c
...
gcc -O -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o util.o util.c
...
+ gcc -Iinclude src/amalgam.c -o stage1/muon
detected compiler gcc '8.5.0' (['gcc']), linker ld.bfd
configuring 'muon', version: 0.2.0
c compiler: supports argument '-Wno-missing-braces': YES
...
c compiler: supports argument '-fstrict-aliasing': YES
configuring '/root/nvme-cli/.build-ci/build-tools/.build-muon/src/version.c'
warn libpkgconf not enabled
err project /root/nvme-cli/.build-ci/build-tools/muon/subprojects/libcurl not found
fallback ['libcurl'] failed for 'libcurl'
err required dependency ['libcurl'] not found
/root/nvme-cli/.build-ci/build-tools/muon/src/external/meson.build:28:22: error required dependency not found
 28 |     dep = dependency(name, required: get_option(name), kwargs: kwargs)
                           ^
/root/nvme-cli/.build-ci/build-tools/muon/src/external/meson.build:28:11: error in function dependency()
 28 |     dep = dependency(name, required: get_option(name), kwargs: kwargs)
                ^
/root/nvme-cli/.build-ci/build-tools/muon/src/meson.build:106:1: error in function subdir()
106 | subdir('external')
      ^
/root/nvme-cli/.build-ci/build-tools/muon/meson.build:99:1: error in function subdir()
 99 | subdir('src')
      ^
```

igaw, 2023-11-06:
Thansk @ikegami-t for testing the samu/build. Highly appreciated.

- CentOS 7:
> ```shell
> src/functions/dependency.c:88:2: error: 'for' loop initial declarations are only allowed in C99 mode
>   for (int64_t idx = 0; idx < arraylen; idx += 2) {
>   ^
> src/functions/dependency.c:88:2: note: use option -std=c99 or -std=gnu99 to compile your code
> ```
The compiler is supporting C99. I suspect we need explicitly add `-std=c99` to the CFLAGS.

- CentOS 8:

libcurl: libcurl is likely to be available but might be too old. The build script is using the wrap feature of the
meson build but we should create a build without using this feature at all. In an earlier version I had this scripted as well.
It looks like we should also have a build like this.

@djukyo:
Thanks for you valuable feedback. We should be able to address all your points so that the 2.x is also working in situation you described. 

What distribution do is completely out of the scope of the upstream project. Though I think all current distro ship the 2.x branch: https://repology.org/project/nvme-cli/badges
Distro released which are in maintenance mode usually do not update package versions at least not major version.

My idea is to have a script which setups a project which contains all the necessary sources to build (samu, muon, libnvme, nvme-cli) and another script (maybe extend the build.sh script?) which is able to build the sources  with a minimum depencency on `make` and `gcc`.

 If this is still not working we still have the option to bring back some Makefiles but I'd rather try first out if we can get away without going back to Makefile times.

ikegami-t, 2023-11-06:
Thansk for your advices. Just reconfirmed `build.sh -m muon` behavior. On `Ubuntu 23.10` it works by deleting three libraries as below but not only libcurl.
```diff
diff --git a/scripts/build.sh b/scripts/build.sh
index 44948356..fa04d05b 100755
--- a/scripts/build.sh
+++ b/scripts/build.sh
@@ -157,9 +157,6 @@ tools_build_muon() {
 
     CC="${CC}" ninja="${SAMU}" stage1/muon setup        \
         -Dprefix="${BUILDDIR}/build-tools"              \
-        -Dlibcurl=enabled                               \
-        -Dlibarchive=enabled                            \
-        -Dlibpkgconf=enabled                            \
         -Ddocs=disabled                                 \
         -Dsamurai=disabled                              \
         "${BUILDDIR}/build-tools/.build-muon"
```
On `CentOS 8` additionally deleted `unit/meson.build` all builds since failed to build them. And disabled ccan `bswap_64()` as below since duplicated with the `/usr/include/bits/byteswap.h` definition.
```diff
[root@linux nvme-cli]# git diff
diff --git a/ccan/ccan/endian/endian.h b/ccan/ccan/endian/endian.h
index 3753f490..78fa2099 100644
--- a/ccan/ccan/endian/endian.h
+++ b/ccan/ccan/endian/endian.h
@@ -87,6 +87,7 @@ static inline uint32_t bswap_32(uint32_t val)
 }
 #endif /* !HAVE_BYTESWAP_H */
 
+#if 0
 #if !HAVE_BSWAP_64
 /**
  * bswap_64 - reverse bytes in a uint64_t value.
@@ -102,6 +103,7 @@ static inline uint64_t bswap_64(uint64_t val)
        return BSWAP_64(val);
 }
 #endif
+#endif
```
And also executed gcc manually to build as below since failed to link by any linker option on the build script.
```sh
[root@linux .build-ci]# gcc -o nvme nvme.p/nbft.c.o nvme.p/fabrics.c.o nvme.p/nvme.c.o nvme.p/nvme-models.c.o nvme.p/nvme-print.c.o nvme.p/nvme-print-stdout.c.o nvme.p/nvme-print-binary.c.o nvme.p/nvme-rpmb.c.o nvme.p/nvme-wrap.c.o nvme.p/plugin.c.o nvme.p/libnvme-wrap.c.o nvme.p/ccan/ccan/hash/hash.c.o nvme.p/ccan/ccan/htable/htable.c.o nvme.p/ccan/ccan/ilog/ilog.c.o nvme.p/ccan/ccan/likely/likely.c.o nvme.p/ccan/ccan/list/list.c.o nvme.p/ccan/ccan/str/debug.c.o nvme.p/ccan/ccan/str/str.c.o nvme.p/ccan/ccan/strset/strset.c.o nvme.p/util/argconfig.c.o nvme.p/util/base64.c.o nvme.p/util/cleanup.c.o nvme.p/util/crc32.c.o nvme.p/util/suffix.c.o nvme.p/util/types.c.o -ldl subprojects/libnvme/src/libnvme.a subprojects/libnvme/ccan/libccan.a subprojects/libnvme/src/libnvme-mi.a 
```
Later will do confirm more and try `CentOS 7` also.

sc108-lee, 2023-11-06:
@djukyo make static
is available for 6
staic binary does not require any dependency.

igaw, 2023-11-07:
@ikegami-t thanks! We clearly have to consolidate the various build steps a bit better and document it how to use it (including the static build).

- If we drop the libcurl & co, we need to download and build libnvme in a separate step. There are a few problems to solve but it's possible. The initial build instruction was doing this. It tries to be smart and setup the subproject itself:
```yaml
  build-minimal:
    name: muon minimal
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v3 
      - name: build muon and samurai build-tool
        run: |
          mkdir build-tools
          cd build-tools
          git clone --depth 1 https://git.sr.ht/~lattis/muon
          cd muon

          export CC=gcc
          export ninja=build/samu

          ./tools/bootstrap_ninja.sh build
          ./bootstrap.sh build

          build/muon setup               \
              -Dlibcurl=disabled         \   
              -Dlibarchive=disabled      \    
              -Ddocs=disabled            \
              -Dsamurai=disabled         \
              build
          build/samu -C build
          build/muon -C build test
      - name: fetch libnvme
        run: |
          cd subprojects
          git clone https://github.com/linux-nvme/libnvme.git
          libnvme_ref=$(sed -n "s/revision = \([0-9a-z]\+\)/\1/p" libnvme.wrap)
          git -C libnvme checkout $libnvme_ref
      - name: build
        run: |
          export PATH=$(pwd)/build-tools/muon/build:$PATH

          muon setup                     \
              -Dlibnvme:python=false     \
              -Dlibnvme:json-c=disabled  \
              -Djson-c=disabled          \
              build
          samu -C build
          muon -C build test  
```
- bswap64: something is not working with our feature test in this case:
```meson
conf.set10(
    'HAVE_BSWAP_64',
    cc.links(
        '''#include <byteswap.h>
            int main(void) {
                return bswap_64(0);
            }
        ''',
        name: 'bswap64'
    ),
    description: 'Is bswap_64 available?'
)
```

jinhuahuang2021, 2023-11-07:
In my place, if using build.sh on CentOS 7. 

[root@bogon nvme-cli]# ./scripts/build.sh -m muon
which: no samu in (/usr/lib/jvm/jdk-11-oracle-x64/bin:.:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/root/bin)
Cloning into '/root/huangjinhua/nvme-cli/.build-ci/build-tools/samurai'...
remote: Enumerating objects: 40, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (37/37), done.
remote: Total 40 (delta 0), reused 21 (delta 0), pack-reused 0
Unpacking objects: 100% (40/40), done.
~/huangjinhua/nvme-cli/.build-ci/build-tools/samurai ~/huangjinhua/nvme-cli
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o build.o build.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o deps.o deps.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o env.o env.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o graph.o graph.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o htab.o htab.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o log.o log.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o parse.o parse.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o samu.o samu.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o scan.o scan.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o tool.o tool.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o tree.o tree.c
gcc  -std=c99 -Wall -Wextra -Wshadow -Wmissing-prototypes -Wpedantic -Wno-unused-parameter -c -o util.o util.c
gcc  -o samu build.o deps.o env.o graph.o htab.o log.o parse.o samu.o scan.o tool.o tree.o util.o
~/huangjinhua/nvme-cli
which: no muon in (/usr/lib/jvm/jdk-11-oracle-x64/bin:.:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/root/bin)
Cloning into '/root/huangjinhua/nvme-cli/.build-ci/build-tools/muon'...
fatal: unable to access 'https://git.sr.ht/~lattis/muon/': The certificate issuer's certificate has expired.  Check your system date and time.

igaw, 2023-11-07:
> In my place, if using build.sh on CentOS 7.

The script fails because a certificate is outdated

> which: no muon in (/usr/lib/jvm/jdk-11-oracle-x64/bin:.:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/root/bin) Cloning into '/root/huangjinhua/nvme-cli/.build-ci/build-tools/muon'... fatal: [...]

This is the place where it fails

```sh
    mkdir -p "${BUILDDIR}"/build-tools
    git clone --depth 1 https://git.sr.ht/~lattis/muon \
        "${BUILDDIR}/build-tools/muon"
    pushd "${BUILDDIR}/build-tools/muon" || exit 1
```

You could try to change the clone step to `git -c http.sslVerify=false clone` which should tell git to ignore the certificate check:

```sh
    mkdir -p "${BUILDDIR}"/build-tools
    git -c http.sslVerify=false clone --depth 1 https://git.sr.ht/~lattis/muon \
        "${BUILDDIR}/build-tools/muon"
    pushd "${BUILDDIR}/build-tools/muon" || exit 1
```

ikegami-t, 2023-11-07:
> * `unit/meson.build` Can you post the output? Depending on the problem, we might just disable it if we detect the situation.

Attach the output [unit-meson.log](https://github.com/linux-nvme/nvme-cli/files/13281268/unit-meson.log) and failed as below so looks a linker error possibly related to the linker failuer that avoided by the manual gcc executing I did.
```sh
[68/69] linking unit/test-uint128
samu: job failed with status 1: gcc  -o unit/test-argconfig-parse unit/test-argconfig-parse.p/test-argconfig-parse.c.o unit/test-argconfig-parse.p/util/argconfig.c.o unit/test-argconfig-parse.p/util/suffix.c.o -Wl,--as-needed -Wl,--no-undefined -static -Wl,--start-group subprojects/libnvme/src/libnvme.a subprojects/libnvme/ccan/libccan.a -Wl,--end-group
/usr/bin/ld: cannot find -lc
collect2: error: ld returned 1 exit status
```

martin-belanger, 2023-11-07:
Is it possible that the static libraries are missing? You could try installing them: `yum install glibc-static`.

ikegami-t, 2023-11-07:
Seems glibc-static not supported on `CentOS 8` as below (Also seems not supported on `CentOS 7` but supported on `CentOS 6.4` as Google searched). By the way installed glibc-devel to make sure but still the liker error caused.
```sh
[root@linux nvme-cli]# yum install glibc-static
Last metadata expiration check: 0:05:51 ago on Tue Nov  7 09:56:56 2023.
No match for argument: glibc-static
Error: Unable to find a match: glibc-static
```

martin-belanger, 2023-11-07:
I did some Google-ing and found the following where they explain how to add the static libraries. Hope it works for you.

https://stackoverflow.com/questions/60238534/add-libraries-for-static-compilation-centos-8

ikegami-t, 2023-11-07:
On `CentOS 7` the C99 error is resolved by adding `-std=c99` to the CFLAGS as suggeted as below.
```diff
[root@localhost nvme-cli]# git diff
diff --git a/scripts/build.sh b/scripts/build.sh
index 4494835..ec350b7 100755
--- a/scripts/build.sh
+++ b/scripts/build.sh
@@ -153,7 +153,7 @@ tools_build_muon() {
         "${BUILDDIR}/build-tools/muon"
     pushd "${BUILDDIR}/build-tools/muon" || exit 1
 
-    CC="${CC}" ninja="${SAMU}" ./bootstrap.sh stage1
+    CC="${CC}" CFLAGS="${CFLAGS} -std=c99" ninja="${SAMU}" ./bootstrap.sh stage1
 
     CC="${CC}" ninja="${SAMU}" stage1/muon setup        \
         -Dprefix="${BUILDDIR}/build-tools"              \
```
But still the error [centos-7-git-error.log](https://github.com/linux-nvme/nvme-cli/files/13282136/centos-7-git-error.log) caused and checking it to resolve.
```sh
+ gcc -std=c99 -Iinclude src/amalgam.c -o stage1/muon
detected compiler gcc '4.8.5' (['gcc']), linker ld.bfd
configuring 'muon', version: 0.2.0
/root/nvme-cli/.build-ci/build-tools/muon/meson.build:25:33: error command failed: 'fatal: Needed a single revision
'
 25 |     git_rev_parse = run_command(git, 'rev-parse', '--short', '@', check: true)
                                      ^
/root/nvme-cli/.build-ci/build-tools/muon/meson.build:25:21: error in function run_command()
 25 |     git_rev_parse = run_command(git, 'rev-parse', '--short', '@', check: true)
                          ^
```

ikegami-t, 2023-11-07:
> I did some Google-ing and found the following where they explain how to add the static libraries. Hope it works for you.
> 
> https://stackoverflow.com/questions/60238534/add-libraries-for-static-compilation-centos-8

Thank you so much! The linker errors resolved by installing glibc-static and the nvme binary created and works.
```
  217  dnf config-manager --set-enabled powertools
  218  dnf search glibc
  219  dnf provides */libc.a
  220  dnf install glibc-static
  221  scripts/build.sh -m muon
```

ikegami-t, 2023-11-07:
> * If we drop the libcurl & co, we need to download and build libnvme in a separate step. There are a few problems to solve but it's possible. The initial build instruction was doing this. It tries to be smart and setup the subproject itself:

Could you please let me know the reason to be needed the librareis below to build to make sure? (Since seems not used by nvme-cli and libnvme codes so seems also not necessary to remain to me.)
1. libcurl
2. libarchive
3. libpkgconf

jinhuahuang2021, 2023-11-08:
@igaw, any big difference between 'meson build' and 'script/build.sh -muon'? It seems 'meson build' is a simple way.
I tested with 'meson build' on CentOS7/8, all are OK.

igaw, 2023-11-08:
@ikegami-t
- `std=c99`
Good catch. Let's add this to the script.
- `git` failure
We should obviously check if git is available instead just blindly try to use it. 
- muon's library dependency
The reason it works for you is, because you already have libnvme checkout in `subprojects`. If you build from scratch the build would fail. Thus we can get rid of them but we need to fetch/build libnvme by hand.

@jinhuahuang2021 
If you have meson available (and it's not way too old) it's the simplest way to build the project. The `muon` approach is a fallback in case you don't have `meson` available, e.g. embedded systems tend to stay away from Python. Thus we are trying to figure out how to build in such a scenarios.

jinhuahuang2021, 2023-11-08:
It's great.
Using 'make static' works well. (Compile at Ubuntu22.04, and works well at CentOS7).

[image]

ikegami-t, 2023-11-08:
The muon build on `CentOS 8` without the libnvme checkout by dropping the libraries are not failed and created the nvme binary working as the log [centos-8-build-muon.log](https://github.com/linux-nvme/nvme-cli/files/13299293/centos-8-build-muon.log). So can we delete the libraries from the build? By the way the warn and err messages are output by the build but not stopped and finished by creating the nvme binary.
```sh
[root@linux nvme-cli-2]# ./scripts/build.sh -m muon
...
c compiler: supports argument '-fstrict-aliasing': YES
configuring '/root/nvme-cli-2/.build-ci/build-tools/.build-muon/src/version.c'
warn libpkgconf not enabled
warn dependency ['libcurl'] not found
warn libpkgconf not enabled
warn dependency ['libarchive'] not found
warn libpkgconf not enabled
fallback ['pkgconf', 'dep_libpkgconf'] failed for 'libpkgconf'
fallback ['pkgconf', 'dep_libpkgconf'] failed for 'libpkgconf'
warn dependency ['libpkgconf'] not found
warn libpkgconf not enabled
[bestline] entering subproject 'bestline'
[bestline] detected compiler gcc '8.5.0' (['gcc']), linker ld.bfd
[bestline] configuring 'bestline', version: undefined
found dependency 'bestline' (declared dependency) version undefined static
warn libpkgconf not enabled
warn dependency ['tracy'] not found
warn ninja/samu not found, skipping project tests 

err command not found
err error writing compile_commands.json
summary:
- muon 0.2.0
      libcurl: false
      libarchive: false
      libpkgconf: false
      samurai: false
      bestline: true
setup complete
...
[105/106] linking subprojects/libnvme/test/test-cpp
[106/106] linking nvme
err command not found
err command not found
warn failed to run ninja
running tests for project 'nvme-cli'
5/5 f: 0 (1) [=================================================================]
err command not found
warn failed to run ninja
running tests for project 'libnvme'
15/15 f: 0 (1) [===============================================================]
finished 20 tests, 2 expected fail, 0 fail, 0 skipped
```

jinhuahuang2021, 2023-11-09:
@igaw 
Some Linux distributions are still using version 1.16. 
Can nvme-cli create its own 2.x repo (such as a yum repository) to make it convenient for these Linux distributions to install the 2.x version without downloading and compiling the code?

igaw, 2023-11-10:
@jinhuahuang2021 I don't mind if someone is willing to step up and setup build jobs which create rpm/dpkg etc for various distributions. But I don't want to spend my time on something like this.

There is an alternative in this case. Just use the [AppImages](https://appimage.org/) we provide:

https://monom.org/linux-nvme/upload/AppImage/nvme-cli-latest-x86_64.AppImage

This should work even on older distros. I know I need to document this a bit better.

jinhuahuang2021, 2023-11-10:
@igaw Great.
Is there a list of your images in your monom.org? If I look for 'nvme-cli-latest-aarch64.AppImage'?

igaw, 2023-11-10:
Yep, this is my own server which just host the latest greatest version as AppImage. I don't safe any older versions at this point. Maybe the latest release could be kept around. Though I don't want to spend again a lot of time to figure out how to do this. So if someone wants to figure this out, please go ahead and send a PR for it.

igaw, 2023-11-16:
I've merged a couple of patches to improve the `./script/build.sh -m muon` step. The necessary changes for CentOS 7 and 8 from @ikegami-t are in. And I made it a bit less stupid so that `samurai` and `muon` are not build every time. As far I can tell, the dependency for this build are:
 - `make`
 - `gcc`
 - `git`

The dependency on `git` is just for the project setup. E.g. we could split the script into a download and build
step. So that it's possible to build the whole thing offline.

> @igaw Great. Is there a list of your images in your monom.org? 

There is only the latest x86_64 AppImage currently.

> If I look for 'nvme-cli-latest-aarch64.AppImage'?

Unfortunately, cross builds are difficult. I've just figured out how to cross build the CI builds. According the AppImage documentation it's possible but I fear this is not straight forward and needs a bit of time to get done. Anyone interested to figure this out?

closed 2023-11-17 by igaw

ikegami-t, 2023-11-17:
Thanks for your support. Just confirmed the latest `scripts/build.sh -m muon` build works to create the nvme binary on both Ubuntu 23.10, CentOS 8 and CentOS 7.
1. Note: But still for the CentOS 7 the muon itself needed to be fixed for the compile error so now waiting the muon patch below applied. (Locally fixed the muon itself build error on CentOS 7 to check the muon build.)
<https://lists.sr.ht/~lattis/muon/patches/46641>
2. FYI: The muon patch below was already applied for the CentOS 7.
<https://lists.sr.ht/~lattis/muon/patches/46549>

Also on the CentOS 8 and CentOS 7 the `scripts/build.sh -m muon` build works but still the error and warning messages etc. output as below. (Log file: [build-muon-centos-7.log](https://github.com/linux-nvme/nvme-cli/files/13393540/build-muon-centos-7.log))
```sh
[root@localhost nvme-cli]# ./scripts/build.sh -m muon
...
[libnvme] warn TODO: install user/group mode
...
configuring '/root/nvme-cli/.build-ci/71-nvmf-netapp.rules'

err command not found
err error writing compile_commands.json
summary:
...
[61/106] compiling c unit/test-suffix-binary-parse.p/util/suffix.c.o
../subprojects/libnvme/test/test.c: In function 'test_ctrl':
../subprojects/libnvme/test/test.c:46:9: warning: missing braces around initializer [-Wmissing-braces]
  struct nvme_ns_list ns_list = { 0 };
         ^
...
[85/106] linking subprojects/libnvme/test/test-mi-mctp
../subprojects/libnvme/test/mi.c: In function 'test_admin_invalid_formats':
../subprojects/libnvme/test/mi.c:582:2: warning: missing braces around initializer [-Wmissing-braces]
  } req = { 0 };
  ^
...
[106/106] linking nvme
err command not found
err command not found
warn failed to run ninja
running tests for project 'nvme-cli'
5/5 f: 1 (1) [====================================================================================================================================================================================================================]
err command not found
warn failed to run ninja
running tests for project 'libnvme'
15/15 f: 0 (1) [==================================================================================================================================================================================================================]
finished 20 tests, 2 expected fail, 1 fail, 0 skipped
[fail ]   0.01s, uint128
stdout: 'ERROR: printing {000003e8.00000000.00000000.00000000}, got '1 000', expected '1 000'
'
```
By the way `glibc-static` required to instal as below for the muon build so should we describe this by `README.md`?
1. CentOS 8
```
  217  dnf config-manager --set-enabled powertools
  218  dnf search glibc
  219  dnf provides */libc.a
  220  dnf install glibc-static
  221  scripts/build.sh -m muon
```
2. CentOS 7
```
  320  yum install glibc-static
```

martin-belanger, 2023-11-17:
>Just confirmed the latest scripts/build.sh -m muon build works to create the nvme binary on both Ubuntu 23.10, CentOS 8 and CentOS 7.

@ikegami-t - I'm just curious. I'm also using Ubuntu 23.10 and I'm able to compile everything using the standard `meson` build without any issue. Have you tried building with the following packages installed? 

```
sudo apt install -y meson uuid-dev libjson-c-dev swig cmake libsystemd-dev python3-dev libssl-dev
```

ikegami-t, 2023-11-17:
Sorry probably my comment was not clear. Yes on Ubuntu 23.10 the `meson` also works but I have just ~~tryied~~ **tried** the `muon` build tested on Ubuntu 23.10 also to compare the `muon` build on CentOS 8 and CentOS 7. (Since still the error and warning messages are output on CentOS 8 and CentOS 7 although the nvme binary created correctly. No error and waning message on Ubunt 23.10 `muon` buld now.) Thansk for your advice.

igaw, 2024-11-29:
BTW, I've started to create a wiki with aim to collect all the trick and tips around nvme-cli. Feel free to update/extend/fix it: https://github.com/linux-nvme/nvme-cli/wiki

ikegami-t, 2024-11-29:
Seems related to the muon changes below but not sure at this moment.
https://git.sr.ht/~lattis/muon/commit/6ec469bb#bootstrap.sh

PR #2129 (2023-11-17): doc: add additional build instructions
Add more information how to build the project in different scenarios (older distros, static builds etc.).

Also mention where to download the test AppImage.

Fixes: #2108