linux-nvme/nvme-cli #2110: Add `--human` command line option to `self-test-log`

igaw, 2023-11-02:
The `-H` (aka `--human`) option is not supported (yet) by the `self-test-log` command.

_Originally posted by @igaw in https://github.com/linux-nvme/nvme-cli/issues/1640#issuecomment-1790985016_

ikegami-t, 2023-11-03:
Just FYI: Current the output on my local envitonment.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme self-test-log /dev/nvme1n1 
Device Self Test Log for NVME device:nvme1n1
Current operation  : 0
Current Completion : 0%
Self Test Result[0]:
  Operation Result             : 0
  Self Test Code               : 1
  Valid Diagnostic Information : 0
  Power on hours (POH)         : 0x395
  Vendor Specific              : 0 0
Self Test Result[1]:
  Operation Result             : 0xf
Self Test Result[2]:
  Operation Result             : 0xf
...
```

ikegami-t, 2023-11-05:
Instead of the `--human` command line option the `--verbose` command line option is supported as below but now the command debug output is also displayed by the option as implemented the output by the commit cf77b4a8.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme self-test-log /dev/nvme1 -h
Usage: nvme self-test-log <device> [OPTIONS]

Retrieve the self-test log for the given device and given test
(or optionally a namespace) in either decoded format (default) or binary.

Options:
  [  --verbose, -v ]                    --- Increase output verbosity
  [  --output-format=<FMT>, -o <FMT> ]  --- Output format: normal|json|binary
  [  --dst-entries=<NUM>, -e <NUM> ]    --- Indicate how many DST log entries
                                            to be retrieved,
                                            by default all the 20 entries
                                            will be retrieved
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme self-test-log /dev/nvme1 -v
opcode       : 02
flags        : 00
rsvd1        : 0000
nsid         : ffffffff
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000234
metadata_len : 00000000
addr         : 55cba2626000
metadata     : 0
cdw10        : 008c0006
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 429 us
Device Self Test Log for NVME device:nvme1
Current operation  : 0
Current Completion : 0%
Self Test Result[0]:
  Operation Result             : 0 Operation completed without error
  Self Test Code               : 1 Short device self-test operation
  Valid Diagnostic Information : 0
  Power on hours (POH)         : 0x395
  Vendor Specific              : 0 0
Self Test Result[1]:
  Operation Result             : 0xf Entry not used (does not contain a result)
Self Test Result[2]:
  Operation Result             : 0xf Entry not used (does not contain a result)
...
```

igaw, 2023-11-07:
Alright, then we don't need to do anything :)

closed 2023-11-07 by igaw