linux-nvme/nvme-cli #2134: gen-tls-key fails on rhel 9.2

decoyindus, 2023-11-23:
Trying to use in-kernel TLS using nvme-cli, referring the following article:
https://lwn.net/Articles/942817/

Exported a nvme target with TCP transport, nqn 'testnqn' with localhost '127.0.0.1' as port (Ref. https://www.mankier.com/8/nvmetcli):

modprobe nvmet-tcp
nvmetcli restore test.json
modprobe nvme-tcp

test.json:
{
  "ports": [
    {
      "addr": {
        "adrfam": "ipv4",
        "traddr": "127.0.0.1",
        "treq": "not specified",
        "trsvcid": "4420",
        "trtype": "tcp"
      },
      "ana_groups": [
        {
          "ana": {
            "state": "optimized"
          },
          "grpid": 1
        }
      ],
      "param": {
        "inline_data_size": "16384",
        "pi_enable": "0"
      },
      "portid": 1,
      "referrals": [],
      "subsystems": [
        "testnqn"
      ]
    }
  ],
  "subsystems": [
    {
      "allowed_hosts": [],
      "attr": {
        "allow_any_host": "1",
        "cntlid_max": "65519",
        "cntlid_min": "1",
        "model": "Linux",
        "pi_enable": "0",
        "qid_max": "128",
        "serial": "8d56b289fe7f0a73eff8",
        "version": "1.3"
      },
      "namespaces": [
        {
          "ana_grpid": 1,
          "device": {
            "nguid": "00000000-0000-0000-0000-000000000000",
            "path": "/dev/sdb",
            "uuid": "45446286-372f-43e8-bfde-024ca4ef7bf1"
          },
          "enable": 1,
          "nsid": 10
        }
      ],
      "nqn": "testnqn"
    }
  ]
}

Normally, nvme connect is successful.

Command: nvme connect -t tcp -n testnqn -a 127.0.0.1 -s 4420
dmesg log:
[83521.042982] nvmet: creating nvm controller 1 for subsystem testnqn for NQN hostnqn.
[83521.043228] nvme nvme0: creating 6 I/O queues.
[83521.044897] nvme nvme0: mapped 6/0/0 default/read/poll queues.
[83521.047239] nvme nvme0: new ctrl: NQN "testnqn", addr 127.0.0.1:4420

While using 'nvme gen-tls-key' with nvme-cli to generate key required for TLS handshake, getting this error:

Failed to insert key, error 126

CONFIGURATION DETAILS:

KERNEL VERSION: 5.14.0-284.30.1.el9_2.x86_64
RHEL VERSION: 9.2 (Plow)
NVMET VERSION: 0.7, RELEASE 3.el9
NVME-CLI RPM VERSION: nvme-cli-2.4-2.el9.x86_64
LIBNVME RPM VERISON: libnvme-1.4-4.el9.x86_64

nvme gen-tls-key --subsysnqn=testnqn

Expected Behaviour:
Should print the key in PSK Interchange format,
'NVMeTLSkey-1:01:<base64 encoded data>:'
 
Actual Result:
nvme gen-tls-key --subsysnqn=testnqn
Failed to insert key, error 126

Result of 'echo $?' : 1
echo $?
1

decoyindus, 2023-11-23:
Also, as mentioned in the article https://lwn.net/Articles/942817/ I'm not able to set the attribute addr_tsas for the localhost port Ive created, as there is no such attribute supported. Is there some configuration that I need to change, or any missing packages/libraries that I need to install?

hreinecke, 2023-11-24:
When you specify '--subsysnqn' for nvme gen-tls-key the resulting key will be inserted into the '.nvme' keyring. That's currently not well documented, but will be cleaned up with PR#2130.

hreinecke, 2023-11-24:
Please check if the '.nvme' keyring is available in /proc/keys. If not than your kernel probably doesn't have TLS encryption for NVMe enabled.

decoyindus, 2023-11-24:
Thanks for replying @hreinecke , is there any configuration changes that I can perform to enable TLS encryption?

decoyindus, 2023-11-27:
**Update:** As stated above, the .nvme keyring is not present in /proc/keys. But on performing **nvme connect** with --tls option succeeds. How does it work? 
#nvme connect -t tcp -n testnqn -a 127.0.0.1 -s 4420 --tls -v
nvme0: testnqn connected

AntonHPE, 2024-02-17:
I assume you are using gnutls, not ktls.

igaw, 2024-07-02:
All the patches are now upstream and all project are updated. In this process a bunch of bugs have been fixed. I suggest you try again with latest upstream version. Should work now (famous last words)

closed 2024-07-02 by igaw