linux-nvme/nvme-cli #2136: create-ns fail on DC SN840 with NS_INSUFFICIENT_CAPACITY

evhv, 2023-11-25:
Good afternoon, just went through quest of bricking and unbricking the drive.

The hardware in question is:
WD Ultrastar DC SN840
WUS4BA138DSP3X1
Firwmare: R2210000

In odrer to create several new namespaces I had to delete the default one as it took up all available space, detach and delete-ns worked without any issues, however when I tried to create the new one I was met with NS_INSUFFICIENT_CAPACITY error. 
Creating namespace with the same size as was the default one lead to the same issue.
unvmcap   : 3840755982336

<details>
<summary>Output of create-ns</summary>

```shell
root@ender:~# nvme create-ns /dev/nvme1 -s 3840755982336 -c 3840755982336 -f 2 -m 0 -d 0
NVMe status: NS_INSUFFICIENT_CAPACITY: Creating the namespace requires more free space than is currently available. The Command Specific Information field of the Error Information Log specifies the total amount of NVM capacity required to create the namespace in bytes(0x6115)
``` 

</details>
<details>
<summary>Output of wdc id-ctrl</summary>

```shell
root@ender:~# nvme wdc id-ctrl /dev/nvme1
NVME Identify Controller:
vid       : 0x1b96
ssvid     : 0x1b96
sn        : ********          
mn        : WUS4BA138DSP3X1                         
fr        : R2210000
rab       : 4
ieee      : 0014ee
cmic      : 0
mdts      : 0
cntlid    : 0
ver       : 0x10300
rtd3r     : 0x1312d00
rtd3e     : 0
oaes      : 0x300
ctratt    : 0x1
rrls      : 0
cntrltype : 0
fguid     : 
crdt1     : 0
crdt2     : 0
crdt3     : 0
oacs      : 0x5f
acl       : 3
aerl      : 15
frmw      : 0x19
lpa       : 0xf
elpe      : 255
npss      : 13
avscc     : 0x1
apsta     : 0
wctemp    : 349
cctemp    : 353
mtfa      : 47
hmpre     : 0
hmmin     : 0
tnvmcap   : 3840755982336
unvmcap   : 3840755982336
rpmbs     : 0
edstt     : 30
dsto      : 1
fwug      : 255
kas       : 0
hctma     : 0
mntmt     : 0
mxtmt     : 0
sanicap   : 0x60000002
hmminds   : 0
hmmaxd    : 0
nsetidmax : 0
endgidmax : 0
anatt     : 0
anacap    : 0
anagrpmax : 0
nanagrpid : 0
pels      : 0
sqes      : 0x66
cqes      : 0x44
maxcmd    : 256
nn        : 128
oncs      : 0xfe
fuses     : 0
fna       : 0x2
vwc       : 0
awun      : 0
awupf     : 0
nvscc     : 1
nwpc      : 0
acwu      : 0
sgls      : 0x70001
mnan      : 0
subnqn    : nqn.2018-01.com.wdc:NGUID:0014EE81000BAB220000000000000000
ioccsz    : 0
iorcsz    : 0
icdoff    : 0
ctrattr   : 0
msdbd     : 0
ps    0 : mp:25.00W operational enlat:50 exlat:50 rrt:0 rrl:0
          rwt:0 rwl:0 idle_power:- active_power:-
ps    1 : mp:24.00W operational enlat:50 exlat:50 rrt:1 rrl:1
          rwt:1 rwl:1 idle_power:- active_power:-
ps    2 : mp:23.00W operational enlat:50 exlat:50 rrt:2 rrl:2
          rwt:2 rwl:2 idle_power:- active_power:-
ps    3 : mp:22.00W operational enlat:50 exlat:50 rrt:3 rrl:3
          rwt:3 rwl:3 idle_power:- active_power:-
ps    4 : mp:21.00W operational enlat:50 exlat:50 rrt:4 rrl:4
          rwt:4 rwl:4 idle_power:- active_power:-
ps    5 : mp:20.00W operational enlat:50 exlat:50 rrt:5 rrl:5
          rwt:5 rwl:5 idle_power:- active_power:-
ps    6 : mp:19.00W operational enlat:50 exlat:50 rrt:6 rrl:6
          rwt:6 rwl:6 idle_power:- active_power:-
ps    7 : mp:18.00W operational enlat:50 exlat:50 rrt:7 rrl:7
          rwt:7 rwl:7 idle_power:- active_power:-
ps    8 : mp:17.00W operational enlat:50 exlat:50 rrt:8 rrl:8
          rwt:8 rwl:8 idle_power:- active_power:-
ps    9 : mp:16.00W operational enlat:50 exlat:50 rrt:9 rrl:9
          rwt:9 rwl:9 idle_power:- active_power:-
ps   10 : mp:15.00W operational enlat:50 exlat:50 rrt:10 rrl:10
          rwt:10 rwl:10 idle_power:- active_power:-
ps   11 : mp:14.00W operational enlat:50 exlat:50 rrt:11 rrl:11
          rwt:11 rwl:11 idle_power:- active_power:-
ps   12 : mp:13.00W operational enlat:50 exlat:50 rrt:12 rrl:12
          rwt:12 rwl:12 idle_power:- active_power:-
ps   13 : mp:12.00W operational enlat:50 exlat:50 rrt:13 rrl:13
          rwt:13 rwl:13 idle_power:- active_power:-
wdc vsn : NULL
``` 
</details>

After some time in denial as I couldn't get the official dm-cli, I managed to create a small namespace with nvme-cli and then used 
```shell
nvme wdc drive-resize
``` 
which brought me back to where I started, 3.84TB drive with one ns, also tried work around using resize-namespace however it does not work on SN840.

Out of experience does it seem like the issue lies within nvme-cli handling the create-ns on these drives, firmware or dm-cli is the only way to use multiple namespaces?

All the best,
Ev

keithbusch, 2023-11-25:
The size for a create-ns command is in blocks (typically 512 or 4096 bytes per block), but the unvmcap is reported in bytes. You need to convert the available byte capacity to a block count.

evhv, 2023-11-25:
Well, the issue was between pc and a chair, what an oversight on my part, thank you very much @keithbusch for pointing at blocks vs bytes.
It works, though there is increase in smart err log every time I execute it, guess just some commands not understood properly by the controller.

Thanks again and all the best
Ev

closed 2023-11-25 by evhv

keithbusch, 2023-11-25:
In hindsight, the command parameters are too closely aligned with the specification wording. It's more natural to think of capacity in GB rather than blocks (which itself has an indirect meaning here since you have to consult another identification to even know how many bytes a block has for a specific format index...), but we can't change what the existing parameters mean now.