linux-nvme/nvme-cli #2170: Removing unactivated firmware from slot

obswwf, 2023-12-22:
After running `nvme fw-commit -s 2 -a 0 /dev/nvme0` and `nvme fw-log /dev/nvme0` I realized that the new firmware version is identical to the old one in `frs1`. Is there a way to remove the new firmware from `frs2`?

ikegami-t, 2023-12-27:
Seems needed to replace the frs2 firmware with the new version firmware if you have instead of the removing since seems not able to just remove the frs2 firmware.

keithbusch, 2023-12-27:
There is no commit action to evict an image from a slot. You have to overwrite it with something else.

obswwf, 2023-12-27:
Are there plans to implement this feature in a future release?

ikegami-t, 2023-12-28:
As far as confirmed NVMe TPs and OCP specification seems no plan for the feature.

igaw, 2024-01-18:
If the nvme committee defines this feature we can implement it. Until then, nothing we can do. So you need to talk to them first.

closed 2024-01-18 by igaw