linux-nvme/nvme-cli #2172: If the Output format option is used in the id-ctrl command and the json parameter is passed, there will be no output.

BaojianXu, 2023-12-25:
nvme-cli version 2.7.1 and 2.6 
`nvme --version
nvme version 2.7.1 (git 2.7.1)
libnvme version 1.7 (git 1.7)
`
nvme id-ctrl /dev/nvme0 --output-format=normal
NVME Identify Controller:
vid       : 0x1c5f
......
but 
nvme id-ctrl /dev/nvme0 --output-format=json
there will be nothing.

ikegami-t, 2023-12-25:
Seems json-c output format disabled as described by https://github.com/linux-nvme/nvme-cli/blob/master/README.md#nvme-cli-dependencies. Also the issue #2050 closed may be related to this issue. Anyway on my local environment the json output works correctly with the version.

BaojianXu, 2023-12-26:
> Seems json-c output format disabled as described by https://github.com/linux-nvme/nvme-cli/blob/master/README.md#nvme-cli-dependencies. Also the issue #2050 closed may be related to this issue. Anyway on my local environment the json output works correctly with the version.

Thank you for your help. I installed libjson-c-dev and then rebuilt it. json-c is works.
`sudo apt-get install libjson-c-dev`
`meson setup .build`

closed 2023-12-26 by BaojianXu