linux-nvme/nvme-cli #2173: nvme list shows incorrect usage/size and format

mxiong90, 2023-12-26:
./nvme-cli version 2.7 or higher shows incorrect usage and format. size is  7.68 TB, but shows 61.45TB and format show 512B with older nvm-cli, but sees 4KiB.

`# ./nvme version
nvme version 2.7.1 (git 2.7.1)
libnvme version 1.7 (git 1.7)

# ./nvme list
[700147.853053][T3178441] block nvme0n1: No UUID available providing old NGUID
[700147.864077][T3178441] block nvme1n1: No UUID available providing old NGUID
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            S791NG0W700267       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q
/dev/nvme1n1          /dev/ng1n1            S791NG0W700262       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q
 `

v2.6
`

# ./nvme version
nvme version 2.6 (git 2.6)
libnvme version 1.6 (git 1.6)

# ./nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            S791NG0W700267       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /   7.68  TB    512   B +  0 B   GDC6602Q
/dev/nvme1n1          /dev/ng1n1            S791NG0W700262       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /   7.68  TB    512   B +  0 B   GDC6602Q
`

ikegami-t, 2023-12-27:
Seems caused by the PR linux-nvme/libnvme#754 changes https://github.com/linux-nvme/libnvme/commit/7959f52960fd1964a75823f1f75f02a91a8304ac so can you test the issue behavior by reverting the commit? (Since the issue not caused on my local environment as the nvme list works correctly with the latest version.) Note: The PR includes the changes https://github.com/linux-nvme/libnvme/commit/52242431977d1a80695765038df6911a4648fdb2 also so may needed to revert the commit also but not sure.

mxiong90, 2024-01-06:
was able to duplicate using nvme-cli v2.6 with libnvme v1.7 and libnvme v1.7.1 

./nvme version
nvme version 2.6 (git 2.6)
libnvme version 1.7 (git 1.7)

./nvme list
[ 2135.952268][T23119] block nvme0n1: No UUID available providing old NGUID
[ 2135.960255][T23119] block nvme1n1: No UUID available providing old NGUID
[ 2135.968229][T23119] block nvme2n1: No UUID available providing old NGUID
[ 2135.976210][T23119] block nvme3n1: No UUID available providing old NGUID
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            S791NG0W700245       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q
/dev/nvme1n1          /dev/ng1n1            S791NG0W700251       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q
/dev/nvme2n1          /dev/ng2n1            S791NG0WB00094       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q
/dev/nvme3n1          /dev/ng3n1            S791NG0W700250       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q

./nvme version
nvme version 2.6 (git 2.6)
libnvme version 1.7.1 (git 1.7.1)

./nvme list
[ 1678.289917][T22288] block nvme0n1: No UUID available providing old NGUID
[ 1678.297927][T22288] block nvme1n1: No UUID available providing old NGUID
[ 1678.305914][T22288] block nvme2n1: No UUID available providing old NGUID
[ 1678.313897][T22288] block nvme3n1: No UUID available providing old NGUID
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            S791NG0W700245       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q
/dev/nvme1n1          /dev/ng1n1            S791NG0W700251       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q
/dev/nvme2n1          /dev/ng2n1            S791NG0WB00094       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q
/dev/nvme3n1          /dev/ng3n1            S791NG0W700250       SAMSUNG MZTL27T6HBLA-00A07               0x1          0.00   B /  61.45  TB      4 KiB +  0 B   GDC6602Q

mxiong90, 2024-01-18:
Thanks, below is output. 

>cat /sys/class/nvme/nvme0/nvme0n1/queue/physical_block_size
```4096```

and
>nvme id-ns /dev/nvme0n1
```
NVME Identify Namespace 1:
nsze    : 0x37e3e92b0
ncap    : 0x37e3e92b0
nuse    : 0
nsfeat  : 0x1a
nlbaf   : 1
flbas   : 0
mc      : 0
dpc     : 0
dps     : 0
nmic    : 0
rescap  : 0
fpi     : 0x80
dlfeat  : 9
nawun   : 1023
nawupf  : 7
nacwu   : 0
nabsn   : 1023
nabo    : 0
nabspf  : 7
noiob   : 0
nvmcap  : 7681501126656
npwg    : 255
npwa    : 7
npdg    : 255
npda    : 7
nows    : 255
mssrl   : 0
mcl     : 0
msrc    : 0
nulbaf  : 0
anagrpid: 0
nsattr  : 0
nvmsetid: 0
endgid  : 0
nguid   : 37393130577002450025384700000001
eui64   : 0000000000000000
lbaf  0 : ms:0   lbads:9  rp:0 (in use)
lbaf  1 : ms:0   lbads:12 rp:0
```

keithbusch, 2024-01-18:
Wait, why would we check physical? The logical is the one to use.

mxiong90, 2024-01-18:
adding logical block size, output shows 512
```cat /sys/class/nvme/nvme0/nvme0n1/queue/logical_block_size
512                                          
```

igaw, 2024-01-19:
> Wait, why would we check physical? The logical is the one to use.

My bad, I got this wrong. Let me fix this.

igaw, 2024-01-19:
Should work again as expected. Thanks for the report!

closed 2024-01-19 by igaw