linux-nvme/nvme-cli #2185: Failed to write to /dev/nvme-fabrics: Invalid argument

AmanAaroh, 2024-01-17:
$ sudo nvme connect -t tcp -n nvmet-test -a 192.168.1.18 -s 4420
Failed to write to /dev/nvme-fabrics: Invalid argument

I am trying to connect a Host with a target but I am getting an invalid argument error but by using the same command and same steps i have already connected other target to the host . 
Currently the target is an odroid m1 board which is a single board computer and has arm processor in it .
I have already checked the steps and the commands and its all correct.

here are few information regarding that:
Target side:
odroid@pc5:~$ dmesg |grep "nvmet_tcp"
[89558.086560] nvmet_tcp: enabling port 1 (192.168.1.14:4420)

odroid@pc5:~$ sudo modprobe nvme
odroid@pc5:~$ sudo modprobe nvme-tcp

odroid@pc5:~$ sudo nvme discover -t tcp -a 192.168.1.14 -s 4420

Discovery Log Number of Records 1, Generation counter 2
=====Discovery Log Entry 0======
trtype:  tcp
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified, sq flow control disable supported
portid:  1
trsvcid: 4420
subnqn:  nvmet-test
traddr:  192.168.1.14
sectype: none

odroid@pc5:~$ sudo nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     2M392L12AJNC         XPG GAMMIX S5                            1         256.06  GB / 256.06  GB    512   B +  0 B   VC0S036H
/dev/nvme1n1     21f3951cde232fdb     Linux                                    1         256.06  GB / 256.06  GB    512   B +  0 B   5.10.0-o

odroid@pc5:~$ dmesg | grep nvme
[    2.989724] nvme 0002:21:00.0: assign IRQ: got 80
[    2.990187] nvme nvme0: pci function 0002:21:00.0
[    2.990289] nvme 0002:21:00.0: enabling device (0000 -> 0002)
[    2.990327] nvme 0002:21:00.0: enabling bus mastering
[    2.990378] nvme 0002:21:00.0: saving config space at offset 0x0 (reading 0x57661cc1)
[    2.990394] nvme 0002:21:00.0: saving config space at offset 0x4 (reading 0x100006)
[    2.990409] nvme 0002:21:00.0: saving config space at offset 0x8 (reading 0x1080201)
[    2.990424] nvme 0002:21:00.0: saving config space at offset 0xc (reading 0x0)
[    2.990438] nvme 0002:21:00.0: saving config space at offset 0x10 (reading 0xf0200004)
[    2.990453] nvme 0002:21:00.0: saving config space at offset 0x14 (reading 0x0)
[    2.990467] nvme 0002:21:00.0: saving config space at offset 0x18 (reading 0x0)
[    2.990481] nvme 0002:21:00.0: saving config space at offset 0x1c (reading 0x0)
[    2.990495] nvme 0002:21:00.0: saving config space at offset 0x20 (reading 0x0)
[    2.990509] nvme 0002:21:00.0: saving config space at offset 0x24 (reading 0xf0204000)
[    2.990524] nvme 0002:21:00.0: saving config space at offset 0x28 (reading 0x0)
[    2.990538] nvme 0002:21:00.0: saving config space at offset 0x2c (reading 0x57661cc1)
[    2.990553] nvme 0002:21:00.0: saving config space at offset 0x30 (reading 0x0)
[    2.990567] nvme 0002:21:00.0: saving config space at offset 0x34 (reading 0x40)
[    2.990581] nvme 0002:21:00.0: saving config space at offset 0x38 (reading 0x0)
[    2.990595] nvme 0002:21:00.0: saving config space at offset 0x3c (reading 0x150)
[    3.142754] nvme nvme0: allocated 64 MiB host memory buffer.
[    3.177953] nvme nvme0: 1/0/0 default/read/poll queues
[    3.242141] cpu cpu0: pvtm = 85070, from nvmem
[    3.257291] rockchip,bus bus-npu: pvtm = 85070, from nvmem
[    3.332276] mpp_rkvenc fdf40000.rkvenc: pvtm = 85070, from nvmem
[    3.335440] mpp_rkvdec2 fdf80200.rkvdec: pvtm = 85070, from nvmem
[    3.338614] mali fde60000.gpu: pvtm = 85070, from nvmem
[89555.786545] nvmet: adding nsid 1 to subsystem nvmet-test
[89558.086560] nvmet_tcp: enabling port 1 (192.168.1.14:4420)
[89632.064569] nvmet: creating controller 1 for subsystem nqn.2014-08.org.nvmexpress.discovery for NQN nqn.2014-08.org.nvmexpress:uuid:a9d198af-04bf-42c3-824c-816839453ce8.
[89632.066143] nvme nvme1: new ctrl: NQN "nqn.2014-08.org.nvmexpress.discovery", addr 192.168.1.14:4420
[89632.067364] nvme nvme1: Removing ctrl: NQN "nqn.2014-08.org.nvmexpress.discovery"
[89649.524928] nvme nvme1: failed to connect socket: -110
[89662.796796] nvmet: creating controller 1 for subsystem nvmet-test for NQN nqn.2014-08.org.nvmexpress:uuid:a9d198af-04bf-42c3-824c-816839453ce8.
[89662.798551] nvme nvme1: creating 4 I/O queues.
[89662.800889] nvme nvme1: mapped 4/0/0 default/read/poll queues.
[89662.803442] nvme nvme1: new ctrl: NQN "nvmet-test", addr 192.168.1.14:4420
[89705.279274] nvmet: creating controller 2 for subsystem nvmet-test for NQN nqn.2014-08.org.nvmexpress:uuid:7b269131-a71f-459c-8426-95048e4111d2.

On the Host side :

aarohnvme@nvme:~$ sudo nvme connect -t tcp -n nvmet-test -a 192.168.1.14 -s 4420
Failed to write to /dev/nvme-fabrics: Invalid argument

aarohnvme@nvme:~$ sudo dmesg | grep nvme
[106106.392853] nvme nvme1: Invalid MNAN value 1024
[106420.817422] nvme nvme1: Invalid MNAN value 1024
[106423.198971] nvme nvme1: Invalid MNAN value 1024
[106517.291096] nvme nvme1: Invalid MNAN value 1024

igaw, 2024-01-18:
`[106106.392853] nvme nvme1: Invalid MNAN value 1024`

The problem is that the target side does not set the mnan value correctly. This check was added to the kernel couple of  releases back. You can either use an older kernel on the host side or update to a more recent kernel on the target side.

You need this fix on the target side: https://github.com/torvalds/linux/commit/3c3ee16532c1

closed 2024-01-18 by igaw

AmanAaroh, 2024-01-19:
> `[106106.392853] nvme nvme1: Invalid MNAN value 1024`
> 
> The problem is that the target side does not set the `nn` value correctly. This check was added to the kernel couple of releases back. You can either use an older kernel on the host side or update to a more recent kernel on the target side.
> 
> You need this fix on the target side: [torvalds/linux@3c3ee16532c1](https://github.com/torvalds/linux/commit/3c3ee16532c1)

Thank you So much it worked