linux-nvme/nvme-cli #2189: Prometheus exporter for SMART and OCP C0 Log Page

jmhands, 2024-01-21:
Is there any roadmap for native integration for a Prometheus exporter? I saw some changes coming in json, it would be good to align any exporters on a specific format. My suggestion would be to track drives by  "sn" with info on "mn" and "fw" from `sudo nvme id-ctrl /dev/nvme1n1 -o json`  then have a standard option for exporting statistics to Prometheus from `sudo nvme smart-log /dev/nvme1n1 -o json` and the OCP log page C0.

A small issue is the C0 log page isn't available in the older releases, but this should be the most helpful log along with the normal smart-log to be able to calculate WAF for workloads with `"Physical media units written"`. Other data in the OCP log will be useful for tracking fleet health across many NVMe SSDs.

works with the latest app image provided
`sudo ./nvme-cli-latest-x86_64.AppImage ocp smart-add-log /dev/nvme0n1 -o json`

jmhands, 2024-01-22:
I wrote a sample exporter with chatgpt (not very good but works)
https://github.com/jmhands/nvme_exporter/tree/main
There are some 3rd party attempts but one coming from nvme-cli directly I could make some standard grafana dashboards. The use case I have in mind is graphing WAF over time for various workloads and monitoring the health of SSD fleet.

igaw, 2024-01-23:
Thanks for the python code, helps to understand what your really need as input for the integration. 

I think we should first define a schema. Could you append here a complete JSON formatted one?

jmhands, 2024-01-23:
Sure, for the POC I just had pretty much everything that wasn't static be a `gauge` type but possible we want some of the items in the SMART log as a `counter`, such as errors or things that reset to zero after reboot. I think it would be fairly straightforward, use exactly the same names as nvme-cli and NVM Express use for NVMe SMART Log. For OCP log page, you can see the spec here https://www.opencompute.org/documents/datacenter-nvme-ssd-specification-v2-5-pdf section 4.8.6 for C0 log page. Most SSD vendors will now support this on data center SSD, and nvme-cli ocp plug-in already parses it just fine. For SSD that doesn't support it, just export the smart-log. Some of the other exporters tried getting SSD info from `nvme list` but better just to get the model, serial, and firmware from nvme id-ctrl or you could get from libnvme directly with the identify command.

jmhands, 2024-02-10:
> There are several github projects out there based on the parsing of the `nvme-cli` command output and I have been using for some time the [Node textfile exporter](https://github.com/prometheus/node_exporter/blob/master/README.md#textfile-collector). I have recently [updated it](https://github.com/prometheus-community/node-exporter-textfile-collector-scripts/pull/204) with parallel execution and included a Grafana dashboard. This has its own limitations.
> 
> I think that to avoid yet another exporter that is not aligned neither with the NVMe specifications nor with Prometheus:
> 
> * NVMe CLI has a golang library for reporting the smart information
> * The NVMe Prometheus exporter is developed under the Node exporter project, for example added as `nvme-smart` [collector](https://github.com/prometheus/node_exporter/blob/master/README.md?plain=1#L142).
> 
> To calculate things like WAF, that do not have directly an opcode, it could be possible to use Prometheus recording rules.

Node exporter is great and that would be a good place to add. It would be nice to have the flexibility to enable more logs pages (like the OCP one I mentioned) for predictive failure and health monitoring for large deployments, but even the base NVMe smart-log in node exporter would be awesome. I found the smart exporter for smartmontools doesn't do SAS properly, for example, hence why I want to do right from nvme-cli instead of smartmontools.

dswarbrick, 2024-02-15:
> The NVMe Prometheus exporter is developed under the Node exporter project, for example added as `nvme-smart` [collector](https://github.com/prometheus/node_exporter/blob/master/README.md?plain=1#L142).

Just to avoid any confusion, the existing nvme collector in node_exporter exposes information gleaned from sysfs, i.e., that which is exposed by the kernel.

It is against [node_exporter policy](https://github.com/prometheus/node_exporter/blob/master/CONTRIBUTING.md#collector-implementation-guidelines) to call external binaries. The textfile collector mechanism is exempt from this, since the apps which generate the textfile metrics are not called by node_exporter itself.

dobbi84, 2024-02-15:
> It is against [node_exporter policy](https://github.com/prometheus/node_exporter/blob/master/CONTRIBUTING.md#collector-implementation-guidelines) to call external binaries. The textfile collector [...]

I had this suspect. Probably the exporter should have some sort of alignment with the `nvme-cli` release, i wonder if it should be an exporter or a script for the textfile extension.

igaw, 2024-08-01:
I understand there is no need to add a new output command as the node exporter is not going to call any binaries anyway. So good to close? Or do I miss something?

dobbi84, 2024-08-01:
> I understand there is no need to add a new output command as the node exporter is not going to call any binaries anyway. So good to close? Or do I miss something?

I think that @jmhands had a wish list in his comment https://github.com/linux-nvme/nvme-cli/issues/2189#issuecomment-1907058641

igaw, 2024-08-01:
I am confused. I understood that the POC in the comment is parsing outputs from `nvme list`, `nvme smart-log` and `nvme ocp smart-add-log` and the wish was one command which will provide all the the infos needed for the logger in a single command. But the node exporter project doesn't want to depend on a binary. So there is 'no need' for a super smart log command.

Or is there still a general interest for getting such a command for non node exporter setups? If so, I'd say we should define the expected output. I don't know what is needed here, so I need this input first. I think it's possible to implement it without too much hustle (yeah, I know famous last words...).

dobbi84, 2024-08-01:
> I am confused. I understood that the POC in the comment is parsing outputs from `nvme list`, `nvme smart-log` and `nvme ocp smart-add-log` and the wish was one command which will provide all the the [...]

We can still parse the output through another exporter like the one created by @jmhands.

igaw, 2024-08-02:
After a bit of pondering, I would like to avoid mixing the different commands output. The `id-ctrl` issues the corresponding NVMe command and that is what is printed out. The same is true for `smart-log`. These are the low level commands from the spec and I'd like to keep it these this way.

If you want a single command which gives you all the necessary information in one go I am fine by introducing e a new command for this (plugin?). And this means we are back to my original question. Do we need this and what would the output look like.

sbates130272, 2024-11-07:
@jmhands @igaw where did we land on this? @jmhands are you still considering working up a PR for this? Or did you find an existing solution out there that works that we could maybe reference in documenation and close this PR?

igaw, 2024-11-08:
From my last comment, I didn't like the idea to change the existing low level commands as they are matching with the spec. 

I don't mind to introduce a user friendly version which provides the summary. As I don't know what you expect, I'd like to hear what is the expected output? Can you come up with something?

sbates130272, 2024-11-08:
@igaw I did some digging and I don't *think* there is anything needed from nvme-cli. I think what we want to do is update the [nvme textfile collector][ref-nvme-collect] to issue the command to obtain the ocp log page and format that as needed. I see that on Ubuntu 22.04 the nvme collector is now on by default. I am not sure if the same is true for other modern distros.

Let me do some more digging. @jmhands do you have anything to add here?

[ref-nvme-collect]: https://github.com/prometheus-community/node-exporter-textfile-collector-scripts/blob/master/nvme_metrics.py

dobbi84, 2025-03-14:
Hi all, i have included the OCP metrics in this [nvme exporter](https://github.com/E4-Computer-Engineering/nvme-exporter). it is a fork of https://fritchie.github.io/nvme_exporter/ but since the original owner was unresponsive to PRs it became another repo. 

Initially, I have tried to go with the textfile exporter but i found it easier and more efficient to use a dedicated exporter. The project includes few Prometheus rules and Grafana visualizations for WAF, current DWPD and estimated remaining life based on [SNIA endurance doc](https://www.snia.org/sites/default/files/SSSI/NVMe_SAS_SATA_Endurance_White_Paper.pdf). We aim to include among others, carbon intensity related metrics.

I'd like to track compatibility between the exporter, the NVMe CLI, the OCP and the NVMe express specifications.

Any help and contribution is more than welcome.

igaw, 2025-03-14:
Nice! I wonder if it wouldn't be better to track for feature instead of nvme-cli versions. And just ignore it if something is missing in the output. Otherwise you need to update the exporter for each nvme-cli release. Distros might patch nvme-cli but not updating their versions. So you can't really rely on the nvme-cli version string to actually be the thing you tested. 

BTW, feel free to add something to the nvme-cli wiki: https://github.com/linux-nvme/nvme-cli/wiki if you think it's worthwhile to add some info here.

dobbi84, 2025-03-17:
That's a good idea @igaw i will make the exporter ignore or log is something is missing. I will make a PR for the Wiki.

Do you know if there is a way from the NVMe CLI to obtain the cell type (example TLC or QLC) and determine the drive form factor?

igaw, 2025-03-17:
wiki: I think you should be able to change it directly. If not let me know. In this case I didn't configure it correctly.

cell type: Sorry, no idea abouthe cell type. Maybe somehow encoded in the endurance group info? Also no clue on the formfactor. I suppose, it would be necessary to create a database where these things are tracked.

hreinecke, 2025-07-25:
I am really not sure. @dswarbrick says that calling external binaries from an exporter is not allowed, yet that is what the mentioned exporter does. Any reason why it's not using libnvme directly?

hreinecke, 2025-07-28:
> [@hreinecke](https://github.com/hreinecke) Only Prometheus node_exporter's policy forbids calling external binaries. Third-party exporters can do whatever they like.

But still. Calling 'nvme' as an external program is horrible; that's _precisely_ why we did libnvme in the first place. And you can link to C libraries in go, so why don't you link against libnvme directly for the prometheus exporter?

dobbi84, 2025-07-28:
> But still. Calling 'nvme' as an external program is horrible; that's _precisely_ why we did libnvme in the first place. And you can link to C libraries in go, so why don't you link against libnvme [...]

I found the exporter understandable enough to extend it with OCP metrics, for the time being it serves its purpose albeit in a horrible way. The next effort will be moving it to libnvme, thanks for the suggestion.

dswarbrick, 2025-07-28:
> But still. Calling 'nvme' as an external program is horrible; that's _precisely_ why we did libnvme in the first place. And you can link to C libraries in go, so why don't you link against libnvme directly for the prometheus exporter?

Calling `nvme` externally is likely so that `sudo` can be used, allowing the exporter to run as an unprivileged user. Linking to libnvme and accessing the block devices directly from Go (or any language) would likely mean that the exporter would have to run as root. Despite Go being a memory-safe language, I think there is still a certain stigma to running http (or any network-enabled) services as root, due to RCE exploits in the past (e.g. way back before it became standard to drop privileges after binding to privileged ports). Go does not support the ability to drop privileges, and such a technique would likely not be applicable anyway, since it would need to repeatedly open and close the NVMe block devices.

igaw, 2025-08-22:
Do you mean something like

```bash
   while true; do 
     sudo nvme smart-log /dev/nvme1n1 -o json
     sleep 10
   done
```

or do you mean `monitor` is doing more work? I don't mind adding something like this if there is an user for it. Though it might get a bit tricky with libnvme in a long running process. There might be some dragons.

dkruces, 2025-08-26:
> Do you mean something like
> 
>    while true; do 
>      sudo nvme smart-log /dev/nvme1n1 -o json
>      sleep 10
>    done
> or do you mean `monitor` is doing more work? I don't mind adding something like this if there is an user for it. 

Exactly. But built-in nvme-cli, e.g.:

```sh
sudo nvme smart-log /dev/nvme1n1 --output-format=json --daemon=500 --output-file=output.json
```

Where daemon and output-file arguments can be something like:

```sh
Usage: nvme smart-log <device> [OPTIONS]

Retrieve SMART log for the given device (or optionally a namespace) in either
decoded format (default) or binary.

Options:
  [  --verbose, -v ]                    --- Increase output verbosity
  [  --output-format=<FMT>, -o <FMT> ]  --- Output format: normal|json|binary
  [  --output-file=<FILE>, -f <FILE> ]  --- Output file instead of stdout
  [  --namespace-id=<NUM>, -n <NUM> ]   --- (optional) desired namespace
  [  --raw-binary, -b ]                 --- output in binary format
  [  --human-readable, -H ]             --- show info in readable format
  [  --timeout=<NUM>, -t <NUM> ]        --- timeout value, in milliseconds
  [  --daemon=<NUM>, -t <NUM> ]        --- query value, in milliseconds
```

And I guess a timestamp to the JSON file to know when the query was performed.

I'm adding support for a monitor like this into [kdevops](https://github.com/linux-kdevops/kdevops). We currently have monitor configuration support here: https://github.com/linux-kdevops/kdevops/blob/main/kconfigs/monitors/Kconfig. I think avoiding the wrapper path would be desirable although it's likely to be our first PoC.

> Though it might get a bit tricky with libnvme in a long running process. There might be some dragons.

I might be missing some context. Could you elaborate on the potential issues you see with libnvme in long-running processes?

dwsuse, 2026-08-18:
Would something like #3865 actually be useful in this context?

dobbi84, 2026-08-20:
> Would something like [#3865](https://github.com/linux-nvme/nvme-cli/pull/3865) actually be useful in this context?

Not completely, I think it still miss the `ocp smart-add-log` metrics which are dependent on the NVMe drive to support them.

dwsuse, 2026-08-20:
oops, I thought it was `ocp_get_smart_extended_log` what was missing. So you only need `smart-add-log` here? (why the heck has OCP so many smart log pages?!)

Or do you want them both? Don't mind adding more if this helps.

dobbi84, 2026-08-20:
> eed `smart-add-log` here? (why the heck has OCP so many smart log pages?!)

ideally both in the same command output, can be nested structure but one command makes things easier instead of two.
Tbh the metrics useful for the WAF scope is the `Physical media units written` but if it doesn't cost much to output them all... i guess it depends on the OCP specifications

dwsuse, 2026-08-20:
yep, it's not a problem to hook up more stuff into this command. I'd say it's better to append instead of nesting, so it we don't have dependencies between the commands (what if one is not implemented?).  

Let me know what else you need here. But does the command name 'smart' still make sense then?

dobbi84, 2026-08-21:
I think that `monitor` parameter will help avoiding 2 commands, we will have the root problem i presume... so having those metrics readable by non root user via `sysfs` would completely close the loop.

Regarding naming, that is not my call to make but i think both [nvme express](https://nvmexpress.org/specifications/)  and [ocp](https://www.opencompute.org/documents/datacenter-nvme-ssd-specification-v2-0r21-pdf)  refer to SMART. The latter is outdated but i can't find version 2.7 (latest). In our specific case, the metrics come from "SMART / Heath Information Extended (Log Identifier C0h) Requirements" chapter.

dwsuse, 2026-08-21:
Adding sysfs entry for NVMe smart has a chance to be accepted, but for OCP I don't see any chance to get it ever get accepted. This means, the permission problem will not go away. Unless it would be possible to come up with 'read' only ioclt.  Not sure if this is feasible.

So this new command would just reduce the work of calling two commands to one. So it's not a big win afterall.

igaw, 2026-09-24:
I don't see anything left to do here. The idea with #3865 isn't really a win. Thus let's close this one. If you disagree, please reopen and state what we can do here. Thanks!

closed 2026-09-24 by igaw