linux-nvme/nvme-cli #2192: set feature c5 fails on nvme-cli 1.13

dandan00, 2024-01-24:
set feature 0xc5 fails on nvme-cli  1.13, but it's ok on nvme-cli 1.12

 nvme set-feature /dev/nvme4n1 -f 0xc5 -s -l 512 -d test.bin
NVMe status: INVALID_NS: The namespace or the format of that namespace is invalid(0xb)

nvme -version
nvme version 1.13

 nvme set-feature /dev/nvme4n1 -f 0xc5 -s -l 512 -d test.bin
set-feature:c5 (Unknown), value:00000000
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: c8 00 00 00 00 00 00 00 08 00 00 00 0a 00 00 00 "................"

nvme -version
nvme version 1.12

I want to know if there are any operations on namespace between these two versions?

ikegami-t, 2024-01-30:
The nvme-cli version 1.13 use the device file /dev/nvme4n1 namespace ID: 1 if the namespace-id option is not set but not used by the version 1.12 as the default namespace ID: 0. So can you try to set the namespace-id option as 0 by the version 1.13?

ikegami-t, 2024-01-31:
Yes right and sorry just rechecked the issue seems fixed by the PR #1838 changes before. By the way if you changed the device to `nvme4` from `nvme4n1` probably the command failed by any error to get NSID from the device when the namespace-id option set 0. (Sorry I can not test by the version now since the build failed by compile errors but if needed try to fix the errors later to test.)
```diff
- nvme set-feature /dev/nvme4n1 -n 0 -f 0xc5 -s -l 512 -d test.bin
+ nvme set-feature /dev/nvme4 -n 0 -f 0xc5 -s -l 512 -d test.bin
```
So as you mentioned probably the version is not able to use with the namespace ID: 0.

dandan00, 2024-02-01:
Thanks.
I looked at the PR #1838.
I will update the version to fix this problem.

closed 2024-04-16 by igaw