linux-nvme/nvme-cli #2196: Flush command with all namespace id 0xFFFFFFFF returns Invalid argument.

yalanliufio, 2024-01-29:
Hello 

My SSD drive does support flush command with all namespace id, but I can't send flush command with all namespace id, it always failed with error: Invalid argument no matter I use the nvme flush command or nvme io passthrough command.

nvme flush /dev/nvme2 -n -1
flush: Invalid argument

nvme io-passthru /dev/nvme2 -o 0 -n -1
passthru: Invalid argument

The nvme version in my server:
nvme --version
nvme version 2.2.1 (git 2.2.1+)
libnvme version 1.2 (git 1.2)

Does this mean nvme cli not support flush with all namespace id?  or is it a kernel problem?  below is the OS info I am using:

cat /etc/os-release output:
NAME="Rocky Linux"
VERSION="8.8 (Green Obsidian)"
ID="rocky"
ID_LIKE="rhel centos fedora"
VERSION_ID="8.8"
PLATFORM_ID="platform:el8"
PRETTY_NAME="Rocky Linux 8.8 (Green Obsidian)"
ANSI_COLOR="0;32"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:rocky:rocky:8:GA"
HOME_URL="https://rockylinux.org/"
BUG_REPORT_URL="https://bugs.rockylinux.org/"
SUPPORT_END="2029-05-31"
ROCKY_SUPPORT_PRODUCT="Rocky-Linux-8"
ROCKY_SUPPORT_PRODUCT_VERSION="8.8"
REDHAT_SUPPORT_PRODUCT="Rocky Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="8.8"

uname -r output:
4.18.0-477.21.1.el8_8.x86_64

keithbusch, 2024-01-29:
Does 'dmesg' say something like a "NVME_IOCTL_IO_CMD not supported" message?

yalanliufio, 2024-01-30:
Good question: Just tried and yes, it has this information:
[Mon Jan 29 21:15:23 2024] nvme nvme2: using deprecated NVME_IOCTL_IO_CMD ioctl on the char device!
[Mon Jan 29 21:15:23 2024] nvme nvme2: nvme: nsid (4294967295) in cmd does not match nsid (1)of namespace
[Mon Jan 29 21:15:36 2024] nvme nvme2: using deprecated NVME_IOCTL_IO_CMD ioctl on the char device!
[Mon Jan 29 21:15:36 2024] nvme nvme2: nvme: nsid (4294967295) in cmd does not match nsid (1)of namespace

keithbusch, 2024-01-30:
The kernel is preventing you from doing that for security reasons. Tooling can't get around that.

closed 2024-01-30 by igaw

yalanliufio, 2024-01-30:
Thank you very much, it is very helpful.