linux-nvme/nvme-cli #2203: potential use after free

igaw, 2024-02-02:
CodeQL complains about a UAF:

```
Memory may have been previously freed by  

ccan/ccan/strset/strset.c:196
			}

			/* Sew empty string back so remaining logic works */
			free(n->u.n);
			n->u.s = empty_str;
			break;
		}

```

https://github.com/linux-nvme/nvme-cli/blob/79abafe689432647b97a441b34714ab2b2708d4d/ccan/ccan/strset/strset.c#L211-L211

igaw, 2024-07-12:
Maybe we should just merge your fix. Upstream is not really responding. We did it the same for the include/define fix for bigendian too.

ikegami-t, 2024-07-13:
Thank you. Noted.

ikegami-t, 2024-07-24:
So can we close the issue now?

igaw, 2024-07-25:
Let's keep it open to track upstream. I still hope it gets eventually addressed.

ikegami-t, 2024-07-25:
I see. Thank you.

igaw, 2026-08-18:
Upstream is not responding, let's close it then.

closed 2026-08-18 by igaw