linux-nvme/nvme-cli #2206: the calculation of logical_block_size when PRACT=1 and MD=8 may wrong.

Bean-Li, 2024-02-05:
https://github.com/linux-nvme/nvme-cli/blob/79abafe689432647b97a441b34714ab2b2708d4d/nvme.c#L7242

cfg.prinfo's value is 0x0~0xf, it is the the [10, 13] bits of control. 

in kernel   include/linux/nvme.h
```
enum {
    NVME_RW_LR          = 1 << 15,
    NVME_RW_FUA         = 1 << 14,
    NVME_RW_APPEND_PIREMAP      = 1 << 9,
    NVME_RW_DSM_FREQ_UNSPEC     = 0, 
    NVME_RW_DSM_FREQ_TYPICAL    = 1, 
    NVME_RW_DSM_FREQ_RARE       = 2, 
    NVME_RW_DSM_FREQ_READS      = 3, 
    NVME_RW_DSM_FREQ_WRITES     = 4, 
    NVME_RW_DSM_FREQ_RW     = 5, 
    NVME_RW_DSM_FREQ_ONCE       = 6, 
    NVME_RW_DSM_FREQ_PREFETCH   = 7, 
    NVME_RW_DSM_FREQ_TEMP       = 8, 
    NVME_RW_DSM_LATENCY_NONE    = 0 << 4,                                                                                                                                   
    NVME_RW_DSM_LATENCY_IDLE    = 1 << 4,
    NVME_RW_DSM_LATENCY_NORM    = 2 << 4,
    NVME_RW_DSM_LATENCY_LOW     = 3 << 4,
    NVME_RW_DSM_SEQ_REQ     = 1 << 6,
    NVME_RW_DSM_COMPRESSED      = 1 << 7,
    NVME_RW_PRINFO_PRCHK_REF    = 1 << 10,
    NVME_RW_PRINFO_PRCHK_APP    = 1 << 11,
    NVME_RW_PRINFO_PRCHK_GUARD  = 1 << 12,
    NVME_RW_PRINFO_PRACT        = 1 << 13,
    NVME_RW_DTYPE_STREAMS       = 1 << 4,
};

```
NVME_RW_PRINFO_PRACT                           is the 13th bits (0 based) of control
NVME_RW_PRINFO_PRCHK_GUARD             is the 12th bits (0 based) of control

[image]

these code in kernel's drivers/nvme/host/ioctl.c 's  nvme_submit_io function.
```
    if ((io.control & NVME_RW_PRINFO_PRACT) &&
        ns->ms == sizeof(struct t10_pi_tuple)) {
        /*  
         * Protection information is stripped/inserted by the
         * controller.
         */
        if (nvme_to_user_ptr(io.metadata))
            return -EINVAL;
        meta_len = 0;                                                                                                                                                       
        metadata = NULL;
    } else {
        meta_len = (io.nblocks + 1) * ns->ms;
        metadata = nvme_to_user_ptr(io.metadata);
    }   

    if (ns->features & NVME_NS_EXT_LBAS) {
        length += meta_len;
        meta_len = 0;
    } else if (meta_len) {
        if ((io.metadata & 3) || !io.metadata)
            return -EINVAL;
    } 
```

so I think these code may be wrong. 

```
+       if (ns->flbas & NVME_NS_FLBAS_META_EXT) {
+               /*
+                * No meta data is transfered for PRACT=1 and MD=8:
+                *   5.2.2.1 Protection Information and Write Commands
+                *   5.2.2.2 Protection Informatio and Read Commands
+                */
+               if (!(cfg.prinfo == 0x1 && ms == 8))
+                       logical_block_size += ms;
+       }
```

should modified to follow.

```
    if (ns->flbas & NVME_NS_FLBAS_META_EXT) {
      /*
         * No meta data is transferred for PRACT=1 and MD=8:
         *   5.2.2.1 Protection Information and Write Commands
         *   5.2.2.2 Protection Information and Read Commands
         */
        if (!((cfg.prinfo & 0x8) != 0 && ms == 8))
            logical_block_size += ms;
    }

```

Is there something wrong with my understanding?

igaw, 2024-02-05:
Indeed, it looks like I mixed up `pract` with `prinfo`. So your suggested changes looks good to me. Would you mind to send a proper PR for this?

@keithbusch do you agree?

Bean-Li, 2024-02-21:
already fix in commit :  https://github.com/linux-nvme/nvme-cli/commit/f1fde3f82874a8bc82f06a8ba128f9bcbd2f534b .
so close this issue. 

```
nvme: do not include meta data for PRACT=1 and MD=8 (version 2)
No meta data is transferred for PRACT=1 and MD=8:
  5.2.2.1 Protection Information and Write Commands
  5.2.2.2 Protection Informatio and Read Commands

commit https://github.com/linux-nvme/nvme-cli/commit/8dcac74470ec02870477e6881766fd74a24738fe (nvme: do not include meta data for PRACT=1 and MD=8)
the condition in if sentence is wrong, mixed up pract with prinfo.

```

closed 2024-02-21 by Bean-Li